aws/amazon-freertos #2286: cmake build error

ivopashov, 2020-07-16:
**Describe the bug**
I am trying to build FreeRTOS on a esp32 devkit-c device but hit a build error. Here is what i do:
* I setup the toolchain reading [here](https://docs.espressif.com/projects/esp-idf/en/v3.3/get-started-cmake/macos-setup.html). 
```
➜  FreeRTOS 2 xtensa-esp32-elf-gcc --version
xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
```
I guess `xtensa-esp32-elf-gcc -> 5.2.0` is compatible with `esp-idf -> 3.3` as this is what espressif provide for that version.

* I have cmake and it looks to be compatible version (> `3.13`):
```
➜  FreeRTOS 2 cmake -version                                                                                                         
cmake version 3.17.3
```
* I go the quick connect path and download the freeRTOS for the esp-32-devkitc
* I navigate to the root download folder (`FreeRTOS 2` in my case) and fire `cmake -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 -S . -B my-build-directory`
* It throws a cmake error
```
-- Checking Python dependencies...
Python requirements from /Users/ivopashov/Downloads/FreeRTOS 2/vendors/espressif/esp-idf/requirements.txt are satisfied.
-- Building for target esp32
CMake Error at vendors/espressif/esp-idf/tools/cmake/scripts/expand_requirements.cmake:144 (list):
  list GET given empty list
Call Stack (most recent call first):
  vendors/espressif/esp-idf/tools/cmake/scripts/expand_requirements.cmake:199 (filter_components_list)

CMake Error at vendors/espressif/esp-idf/tools/cmake/idf_functions.cmake:338 (message):
  Failed to expand component requirements
Call Stack (most recent call first):
  vendors/espressif/boards/esp32/CMakeLists.txt:433 (idf_import_components)
  CMakeLists.txt:49 (include)

-- Configuring incomplete, errors occurred!
See also "/Users/ivopashov/repos/freertos-build/CMakeFiles/CMakeOutput.log".
See also "/Users/ivopashov/repos/freertos-build/CMakeFiles/CMakeError.log".
```
and the `CMakeError.log` output
```
Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: /Users/ivopashov/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc 
Build flags: -mlongcalls
Id flags:  

The output was:
1
/Users/ivopashov/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find crt1-sim.o: No such file or directory
/Users/ivopashov/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find _vectors.o: No such file or directory
/Users/ivopashov/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lsim
/Users/ivopashov/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lhandlers-sim
/Users/ivopashov/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lhal
collect2: error: ld returned 1 exit status

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: /Users/ivopashov/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ 
Build flags: -mlongcalls
Id flags:  

The output was:
1
/Users/ivopashov/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find crt1-sim.o: No such file or directory
/Users/ivopashov/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find _vectors.o: No such file or directory
/Users/ivopashov/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lsim
/Users/ivopashov/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lhandlers-sim
/Users/ivopashov/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lhal
collect2: error: ld returned 1 exit status
```

Something leads me to believe it is some version mismatch but this environment is quite new for me and am having issue to grasp what the real problem is.

I am using these [aws docs](https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html).

**System information**
- Which hardware board or part numbers? - espressif esp32-wroom-32d
- IDE used - VS code
- Operating System MacOS
- Version of FreeRTOS (run ``git describe --tags`` to find it)
- Project Demo
- If your project is a Custom Application, please add the relevant code snippet in the section `Code to reproduce the bug`. 

**Expected behavior**
I expect error free build.

leegeth, 2020-07-22:
@ivopashov ,

Thanks for reaching out to us and apologies for the delay.  We wanted to follow up with some clarifying questions to help you better.

1. From the logs that you shared, the build failed while trying to run cmake. One possibility that can lead to this issue is that the submodules are not available. Can you please run the command from root folder of amazon-freertos `git submodule update --init --recursive` to make sure you have all the submodules downloaded and available. After this can you please retry build?
2. Can you please let us know the software version or the branch that you are using to perform the ESP build?

Please get back to us with the requested details so that we can help you better.

lundinc2, 2020-07-22:
Hello @ivopashov,

I am able to reproduce this issue when the environment variable IDF_PATH has a value. Can you use the following command to confirm IDF_PATH is left undefined?
```
printenv printenv IDF_PATH
```

If this command returns a value please enter `unset IDF_PATH` and try to build again in a clean build folder.  

Thanks

lundinc2, 2020-07-28:
Hello @ivopashov,

I am going to resolve this issue with the hope that it was resolved. Please re-open this issue in case you are having any more trouble building Espressif with CMake

ivopashov, 2020-07-30:
Sorry for not replying in timely manner. @leegeth, i am using the quick connect flow so i am not downloading the repo from Github . @lundinc2 i know about this variable and it does not have value.
```
➜  FreeRTOS 2 echo $IDF_PATH                                                                          

```

However, i managed to build the thing from the repository. I am experiencing issues with the QUICK CONNECT flow only.

lundinc2, 2020-07-30:
@ivopashov can you use the repository in the meantime? I believe there is some issue with the quick connect build on Espressif.

ivopashov, 2020-07-31:
@lundinc2 Yes, I am good using the repo. Yhat's the path I took.

closed 2020-08-03 by dan4thewin

lundinc2, 2020-08-03:
@ivopashov can you post what version of FreeRTOS you downloaded from the console, and what the configuration was?

Thanks

tgsong, 2020-08-04:
@ivopashov One thing the call out, the path cannot contain spaces, otherwise I espressif build will fail. Try to remove the space in your path, `Users/ivopashov/Downloads/FreeRTOS 2`, and see if it works.

ivopashov, 2020-08-06:
@tgsong yep, that's the one. Spaces break the build. 
/cc @lundinc2