aws/amazon-freertos #2287: Demo bootloader build failure, microchip aws_bootloader

ic101, 2020-07-16:
I am trying to evaluate aws free rtos offerings, and am having issues building the pic32mz_ef_curiosity aws_bootloader demo (which is causing the aws_demos to fail building).

I have gone through the aws tutorial to generate security certificates for the demo applications, and placed the file: aws_ota_codesigner_certificate.pem into the codesigner_cert_utility folder. 

I believe the aws_ota_codesigner_certificate.pem certificate is in the correct format:
-----BEGIN CERTIFICATE-----
...
... numbers i don't care to share ...
...
-----END CERTIFICATE-----

There is a pre-build python script that is supposed to generate a header file. It's executed with this line from the IDE:
python ../../../../../vendors/microchip/boards/curiosity_pic32mzef/bootloader/bootloader/utility/codesigner_cert_utility/codesigner_cert_utility.py ../../../../../vendors/microchip/boards/curiosity_pic32mzef/bootloader//bootloader/utility/codesigner_cert_utility/aws_ota_codesigner_certificate.pem ../../../../../vendors/microchip/boards/curiosity_pic32mzef/bootloader//bootloader/include/aws_boot_codesigner_public_key.h

I tried replacing the two // with / thinking python may handle it funny, but no luck.

There are multiple issues at build time:

```
"User defined pre-build step: [python ../../../../../vendors/microchip/boards/curiosity_pic32mzef/bootloader/bootloader/utility/codesigner_cert_utility/codesigner_cert_utility.py ../../../../../vendors/microchip/boards/curiosity_pic32mzef/bootloader/bootloader/utility/codesigner_cert_utility/aws_ota_codesigner_certificate.pem ../../../../../vendors/microchip/boards/curiosity_pic32mzef/bootloader/bootloader/include/aws_boot_codesigner_public_key.h]"
############################################################ WARNING ############################################################
# No certificate present in ../../../../../vendors/microchip/boards/curiosity_pic32mzef/bootloader/bootloader/utility/codesigner_cert_utility/aws_ota_codesigner_certificate.pem
############################################################ WARNING ############################################################
"--------------------------------------"
make  -f nbproject/Makefile-pic32mz_ef_curiosity.mk dist/pic32mz_ef_curiosity/production/bootloader.production.hex
make[2]: Entering directory 'C:/Users/ian/Desktop/Open_Source_PIC32/amazon-freertos-release/projects/microchip/curiosity_pic32mzef/mplab/bootloader'
make[2]: *** No rule to make target '../../../../../vendors/microchip/boards/curiosity_pic32mzef/bootloader/common/microchip_code/framework/system/devcon/src/sys_devcon_cache_pic32mz.S', needed by 'build/pic32mz_ef_curiosity/production/_ext/1201805982/sys_devcon_cache_pic32mz.o'.  Stop.
make[1]: *** [nbproject/Makefile-pic32mz_ef_curiosity.mk:105: .build-conf] Error 2
```

The python script is throwing an exception with the file present:

```
    try:
        # Load the pem encoded certificate file
        with open(sys.argv[1], "r") as cert_file:
            certx509 = OpenSSL.crypto.load_certificate(OpenSSL.crypto.FILETYPE_PEM, cert_file.read().encode('ASCII'))
    except:
        # Any exceptions with opening the file or loading the file prints this warning.
        print("############################################################ WARNING ############################################################")
        print("# No certificate present in " + sys.argv[1])
        print("############################################################ WARNING ############################################################")
        sys.exit(0)
```

I have checked the file name and path and they appear correct. Perhaps I am having a version problem when python calls OpenSSL.crypto.load_certificate? Or maybe the key is incorrectly formatted?

I would like to help get this demo working, but I am not very good with python. 

There also seems to be an assembly issue that I will try to bring up with Microchip.

Development Environment:
Windows 10, Mplab X v5.35, XC32 2.20, Python 3.8.4

ic101, 2020-07-17:
Perhaps my  issues with the bootloader is the first problem of many.

After removing the bootloader to try working with the aws_demos (with no bootloader), it appears the IDE cannot find FreeRTOS.h. That's bad. It seems some other vendors are packaging this file. Microchip harmony 2.0.5 packaged this in a third party folder that is not present in the aws freertos vendor source. Perhaps I'm looking at a demo that is a few years old and no longer supported? or got left behind by changes to the aws freertos source?

Microchip seems to have moved on to Harmony 3

Is anyone else working with the pic32 mz ef curiosity board?

leegeth, 2020-07-22:
@ic101 ,

Thanks for bringing this issue to our attention. Please find a few suggestions that may help you with the issue that you are facing.

1. As understood from the comment, "I have gone through the aws tutorial to generate security certificates for the demo applications, and placed the file: aws_ota_codesigner_certificate.pem into the codesigner_cert_utility folder.", you may be trying to configure the OTA demo for pic32. Is that right? If you just want the default MQTT demo to be built and run, this configuration may not be required. You would be able to follow the instructions in [here](https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_mch.html).  Please let us know if you want to try the OTA demo itself.
2. One other thing that I noticed from the log snapshot you attached is, the path length may be long for mplab to successfully build a project in Windows. You will be able to verify this by trying to build the project after moving the root folder of amazon-freertos to shorter path(for example directly under C:\).

ic101, 2020-07-22:
@leegeth,

Thank you for your response. Shortening the file path has got me building.

The instructions in the aws link are indeed what I have been following.

The project found at:
projects/microchip/curiosity_pic32mzef/mplab/aws_demos
... seems setup to build the bootloader and combine the demo and bootloader binaries (unified hex files) for the pic32.

the bootloader build process is what calls the python script.

For me, OTA doesn't really matter at the moment, I'm mostly in 'Hello World' mode here. My first thought was to build however it was setup by default. I guess if a new user doesn't attempt an OTA update, they should be ok.

closed 2020-07-22 by ic101