aws/amazon-freertos #2292: Reconnection to WiFi and restoring MQTT connection after light sleep ESP32

EtienneMdv, 2020-07-17:
Hi,

I noticed that I wasn't able to neither receive nor publish MQTT messages after reconnecting my ESP32 board to the WiFi after going on a light sleep.

It seems like it is a socket issue.

Here's the error : `12227 [iot_thread] [ERROR][NET][122270] Error -27648 while sending data.`

I would be very grateful for your help!

Best,
Etienne

leegeth, 2020-07-21:
@EtienneMdv ,

Thanks for bringing this issue to our attention. We would need a few more details to better understand the issue that you are facing.
1. Are you reconnecting to the MQTT broker endpoint after the light sleep? If that is the case, have you tried to send an MQTT connect packet again to the broker?
2. If a connection is maintained to the MQTT broker endpoint during the light sleep, how long was the sleep for? If you are using AWS IoT MQTT broker, monitoring life-cycle messages on certain MQTT topics would help to understand if the MQTT connection was disconnected or not. Refer to the documentation about that [here](https://docs.aws.amazon.com/iot/latest/developerguide/life-cycle-events.html).
Also, analyzing the packets exchanged in a TCP dump would help to understand the disconnect reason. Can you please send us a TCP dump if possible?
3. Also, please let us know the software version that you are using.

dan4thewin, 2020-08-03:
Closing this, given its age.  If you still have a problem, please open another issue.

closed 2020-08-03 by dan4thewin