aws/amazon-freertos #2307: Dependency in the lwIP iot_secure_sockets.c file to the Wi-Fi library

vvelciu, 2020-07-23:
There is a dependency in the lwIP iot_secure_sockets.c file to the Wi-Fi library. According to the qualification guide, it looks like the Wi-Fi library is not a requirement, as long as the device can connect to the AWS Cloud across a secure ethernet connection. 
 
[image]

Is this call to WIFI_GetHostIP intended?

lundinc2, 2020-07-23:
Hello @vvelciu,

Thank you for reporting this issue, I have begun investigating.

gkwicker, 2020-08-13:
Hi @vvelciu, thank you for bringing this to our attention.  I have merged a fix (see PR #2359) for it.  Please let us know if there is anything else we can help with on this issue.  Thanks again!

closed 2020-08-15 by yanjos-dev