aws/amazon-freertos #2310: PIC32MZEF mplab project doesn't compile

DexterHaslem, 2020-07-24:
**Describe the bug**
A fresh checkout of the repo doing the following:

` git clone https://github.com/aws/amazon-freertos.git --recurse-submodules`

The submodules are pulled and look ok.

Opening the `projects\microchip\curiosity_pic32mzef\mplab\aws_demos` project like usual results in a bunch of build errors from mbed TLS it appears:

```
"C:\Program Files\Microchip\xc32\v2.41\bin\xc32-gcc.exe"    -g -x c -c -mprocessor=32MZ2048EFM100  -ffunction-sections -O1 -DMBEDTLS_CONFIG_FILE=\\\"aws_mbedtls_config.h\\\" -DCONFIG_MEDTLS_USE_AFR_MEMORY -D__free_rtos__ -I"../../../../../freertos_kernel/include" -I"../../../../../libraries/c_sdk/standard/common/include" -I"../../../../../libraries/freertos_plus/standard/crypto/include" -I"../../../../../libraries/freertos_plus/standard/tls/include" -I"../../../../../libraries/freertos_plus/standard/freertos_plus_tcp/include" -I"../../../../../libraries/freertos_plus/standard/freertos_plus_tcp/source/portable/Compiler/GCC" -I"../../../../../libraries/abstractions/wifi/include" -I"../../../../../libraries/abstractions/pkcs11/include" -I"../../../../../vendors/microchip/boards/curiosity_pic32mzef/aws_demos/application_code/microchip_code" -I"../../../../../freertos_kernel/portable/MPLAB/PIC32MZ" -I"../../../../../vendors/microchip/harmony/v2.05/framework" -I"../../../../../vendors/microchip/harmony/v2.05/bsp" -I"../../../../../demos/include" -I"../../../../../vendors/microchip/boards/curiosity_pic32mzef/aws_demos/config_files" -I"../../../../../demos/network_manager" -I"../../../../../libraries/c_sdk/standard/common/include/private" -I"../../../../../libraries/abstractions/platform/include" -I"../../../../../libraries/abstractions/platform/freertos/include" -I"../../../../../libraries/abstractions/secure_sockets/include" -I"../../../../../libraries/freertos_plus/standard/freertos_plus_tcp/test" -I"../../../../../vendors/microchip/harmony/v2.05/framework/driver/wifi/wilc1000/wireless_driver_extension" -I"../../../../../vendors/microchip/harmony/v2.05/framework/driver/wifi/wilc1000/wireless_driver_extension/driver/include" -I"../../../../../libraries/freertos_plus/standard/pkcs11/include" -I"../../../../../libraries/freertos_plus/standard/utils/include" -I"../../../../../demos/dev_mode_key_provisioning/include" -I"../../../../../vendors/microchip/harmony/v2.05/framework/system/common" -I"../../../../../vendors/microchip/harmony/v2.05/framework/driver/wifi/wilc1000/include" -I"../../../../../vendors/microchip/harmony/v2.05/framework/driver/wifi/wilc1000/wireless_driver_extension/common/include" -I"../../../../../vendors/microchip/harmony/v2.05/framework/driver/wifi/wilc1000/wireless_driver_extension/driver/source" -I"../../../../../vendors/microchip/harmony/v2.05/framework/driver/wifi/wilc1000/wireless_driver/include" -I"../../../../../libraries/c_sdk/aws/defender/include" -I"../../../../../libraries/c_sdk/standard/mqtt/include" -I"../../../../../libraries/c_sdk/standard/serializer/include" -I"../../../../../libraries/c_sdk/aws/shadow/include" -I"../../../../../libraries/c_sdk/standard/https/include" -I"../../../../../libraries/freertos_plus/aws/greengrass/include" -I"../../../../../libraries/freertos_plus/aws/ota/src" -I"../../../../../libraries/freertos_plus/aws/ota/include" -I"../../../../../libraries/3rdparty/mbedtls/include" -I"../../../../../libraries/abstractions/posix/include" -I"../../../../../vendors/microchip/boards/curiosity_pic32mzef/ports/posix" -I"../../../../../libraries/freertos_plus/standard/freertos_plus_posix/include" -I"../../../../../libraries/3rdparty/jsmn" -I"../../../../../libraries/3rdparty/pkcs11" -I"../../../../../libraries/3rdparty/tinycbor/src" -I"../../../../../libraries/3rdparty/tinycrypt/asn1" -I"../../../../../libraries/3rdparty/tinycrypt/lib/include" -I"../../../../../libraries/abstractions/pkcs11/mbedtls" -I"../../../../../libraries/3rdparty/mbedtls/include/mbedtls" -I"../../../../../libraries/3rdparty/mbedtls_utils" -I"../../../../../libraries/3rdparty/mbedtls_config" -I"../../../../../libraries/3rdparty/http_parser" -MMD -MF "build/pic32mz_ef_curiosity/production/_ext/1982230249/iot_logging_task_dynamic_buffers.o.d" -o build/pic32mz_ef_curiosity/production/_ext/1982230249/iot_logging_task_dynamic_buffers.o ../../../../../libraries/c_sdk/standard/common/logging/iot_logging_task_dynamic_buffers.c    -DXPRJ_pic32mz_ef_curiosity=pic32mz_ef_curiosity  -no-legacy-libc    -mnewlib-libc -std=gnu99 -fgnu89-inline -mdfp="C:/Program Files (x86)/Microchip/MPLABX/v5.40/packs/Microchip/PIC32MZ-EF_DFP/1.2.51"  
In file included from ../../../../../demos/dev_mode_key_provisioning/src/aws_dev_mode_key_provisioning.c:61:0:
../../../../../libraries/3rdparty/mbedtls/include/mbedtls/pk.h:58:10: error: #include expects "FILENAME" or <FILENAME>
 #include MBEDTLS_CONFIG_FILE
          ^
In file included from ../../../../../libraries/3rdparty/mbedtls/include/mbedtls/pk.h:61:0,
                 from ../../../../../demos/dev_mode_key_provisioning/src/aws_dev_mode_key_provisioning.c:61:
../../../../../libraries/3rdparty/mbedtls/include/mbedtls/md.h:62:10: error: #include expects "FILENAME" or <FILENAME>
 #include MBEDTLS_CONFIG_FILE
          ^
In file included from ../../../../../demos/dev_mode_key_provisioning/src/aws_dev_mode_key_provisioning.c:62:0:
../../../../../libraries/3rdparty/mbedtls/include/mbedtls/oid.h:57:10: error: #include expects "FILENAME" or <FILENAME>
 #include MBEDTLS_CONFIG_FILE
          ^
In file included from ../../../../../libraries/3rdparty/mbedtls/include/mbedtls/oid.h:60:0,
                 from ../../../../../demos/dev_mode_key_provisioning/src/aws_dev_mode_key_provisioning.c:62:
../../../../../libraries/3rdparty/mbedtls/include/mbedtls/asn1.h:57:10: error: #include expects "FILENAME" or <FILENAME>
 #include MBEDTLS_CONFIG_FILE
          ^
../../../../../demos/dev_mode_key_provisioning/src/aws_dev_mode_key_provisioning.c: In function 'prvProvisionPrivateECKey':
../../../../../demos/dev_mode_key_provisioning/src/aws_dev_mode_key_provisioning.c:135:5: error: unknown type name 'mbedtls_ecp_keypair'
     mbedtls_ecp_keypair * pxKeyPair = ( mbedtls_ecp_keypair * ) pxMbedPkContext->pk_ctx;
     ^
../../../../../demos/dev_mode_key_provisioning/src/aws_dev_mode_key_provisioning.c:135:41: error: 'mbedtls_ecp_keypair' undeclared (first use in this function)
     mbedtls_ecp_keypair * pxKeyPair = ( mbedtls_ecp_keypair * ) pxMbedPkContext->pk_ctx;
                                         ^
../../../../../demos/dev_mode_key_provisioning/src/aws_dev_mode_key_provisioning.c:135:41: note: each undeclared identifier is reported only once for each function it appears in
../../../../../demos/dev_mode_key_provisioning/src/aws_dev_mode_key_provisioning.c:135:63: error: expected expression before ')' token
     mbedtls_ecp_keypair * pxKeyPair = ( mbedtls_ecp_keypair * ) pxMbedPkContext->pk_ctx;
                                                               ^
../../../../../demos/dev_mode_key_provisioning/src/aws_dev_mode_key_provisioning.c:151:9: warning: implicit declaration of function 'mbedtls_mpi_write_binary' [-Wimplicit-function-declaration]
         lMbedResult = mbedtls_mpi_write_binary( &( pxKeyPair->d ), pxD, EC_D_LENGTH );
         ^
../../../../../demos/dev_mode_key_provisioning/src/aws_dev_mode_key_provisioning.c:151:61: error: request for member 'd' in something not a structure or union
         lMbedResult = mbedtls_mpi_write_binary( &( pxKeyPair->d ), pxD, EC_D_LENGTH );
                                                             ^
../../../../../demos/dev_mode_key_provisioning/src/aws_dev_mode_key_provisioning.c:162:22: error: request for member 'grp' in something not a structure or union
         if( pxKeyPair->grp.id == MBEDTLS_ECP_DP_SECP256R1 )
                      ^
../../../../../demos/dev_mode_key_provisioning/src/aws_dev_mode_key_provisioning.c:162:34: error: 'MBEDTLS_ECP_DP_SECP256R1' undeclared (first use in this function)
         if( pxKeyPair->grp.id == MBEDTLS_ECP_DP_SECP256R1 )
                                  ^
../../../../../demos/dev_mode_key_provisioning/src/aws_dev_mode_key_provisioning.c: In function 'prvProvisionPrivateRSAKey':
../../../../../demos/dev_mode_key_provisioning/src/aws_dev_mode_key_provisioning.c:220:5: error: unknown type name 'mbedtls_rsa_context'
     mbedtls_rsa_context * xRsaContext = pxMbedPkContext->pk_ctx;
     ^
[... cut ...]
make[2]: *** [build/pic32mz_ef_curiosity/production/_ext/2144440666/aws_dev_mode_key_provisioning.o] Error 255
make[2]: *** Waiting for unfinished jobs....
nbproject/Makefile-pic32mz_ef_curiosity.mk:2299: recipe for target 'build/pic32mz_ef_curiosity/production/_ext/2144440666/aws_dev_mode_key_provisioning.o' failed

... 

BUILD FAILED (exit value 2, total time: 6s)

```

This is on XC32 2.40 and MPLAB v5.41. Something with `MBEDTLS_CONFIG_FILE`.
This is on master at `0dc240d472065b21`. Thanks for any clarification

lundinc2, 2020-07-24:
Hello @DexterHaslem thanks for bringing this our attention. I'll take a look.

lundinc2, 2020-07-24:
I was unable to reproduce this on both MacOS and Windows. Looking at your log, your output shows
```
-DMBEDTLS_CONFIG_FILE=\\\"aws_mbedtls_config.h\\\"
```
but my log on Windows shows
```
-DMBEDTLS_CONFIG_FILE=\"aws_mbedtls_config.h\"
```

Can you try removing these extra backslashes?

### In mplab:
1. Right click `aws_demos` in the projects tab
1. Click `properties`
1. Under `Conf` > `XC32` click `xc32-gcc`
1. Click `Generated Command Line`
1. In the text boxy modify `-DMBEDTLS_CONFIG_FILE=\\\"aws_mbedtls_config.h\\\"` to `-DMBEDTLS_CONFIG_FILE=\"aws_mbedtls_config.h\"`

### Modify XML directly
1. Open https://github.com/aws/amazon-freertos/blob/9e376b1ec3e108bb20d5ebff1b9f4720b020b5d7/projects/microchip/curiosity_pic32mzef/mplab/aws_demos/nbproject/configurations.xml#L988
1. Modify `-DMBEDTLS_CONFIG_FILE=\\\"aws_mbedtls_config.h\\\"` to `-DMBEDTLS_CONFIG_FILE=\"aws_mbedtls_config.h\"`

DexterHaslem, 2020-07-25:
Thanks for spotting the fix. for some reason opening the project always re-mangles that preprocessor define for some reason, I am trying to narrow down if it's a particular MPLABX version

lundinc2, 2020-07-27:
@DexterHaslem this is the output of my MPLAB X version. I'll try and see if I can grab the version from the Windows PC I used as well.

```
Product Version: MPLAB X IDE v5.25
Java: 1.8.0_181; Java HotSpot(TM) 64-Bit Server VM 25.181-b13
Runtime: Java(TM) SE Runtime Environment 1.8.0_181-b13
System: Mac OS X version 10.14.6 running on x86_64; UTF-8; en_EN (mplab)
```

lundinc2, 2020-07-27:
@DexterHaslem do you mind providing what version of MPLAB X you are using?

lundinc2, 2020-07-27:
I have verified that currently our tests are performed on MPLAB X version 5.30 on Windows.

DexterHaslem, 2020-07-28:
XC32 2.40 and MPLAB v5.41. I got sidetracked, two different machines on that latest version, only one was exhibiting the issue causing massive confusion. I will go back to older version

DexterHaslem, 2020-07-28:
OK, mplabx v5.40+ seems to exhibit the issue on all three platforms. I went back to older version, thanks

closed 2020-07-28 by DexterHaslem