aws/amazon-freertos #2319: Linking CXX executable aws_demos.elf FAILED: aws_demos.elf

dipen-1337lab, 2020-07-28:
**Describe the bug**
The AFR Version 202007.00 generates a linking error
Linking CXX executable aws_demos.elf
FAILED: aws_demos.elf

As can be understood from the console output provide below, the linking error is because of the step number [875/904], which generates an "unrecognized arguments" error.
ldgen.py: error: unrecognized arguments: || (set FAIL_LINE=3& goto :ABORT)

**System information**
- Hardware board: ESP32-DevKitC
- IDE used: N/A
- Operating System [Windows 10 x64 -- Version: 1909 -- OS Build: 18363.959]
- Version of FreeRTOS: AFR Version 202007.00 downloaded from console.aws.amazon.com
- Project: OTA Demo

**Expected behavior**
The aws_demos.bin firmware image should be generated successfully for the OTA Update demo application on AFR Version 202007.00 on similar lines, as generated on AFR Version 202002.00

**Screenshots or console output**
...
... [Partial log for the error generated]
...
[49/55] Linking C static library esp-idf\main\libmain.a
[50/55] Linking C static library esp-idf\bootloader_support\libbootloader_support.a
[51/55] Generating idf_component_main.sections_info
[52/55] Generating idf_component_bootloader_support.sections_info
[53/55] Building C object CMakeFiles/bootloader.dir/dummy_main_src.c.obj
[54/55] Linking C executable bootloader.elf
[55/55] Generating bootloader.bin
esptool.py v2.8
[875/904] Generating esp32.project.ld
usage: ldgen.py [-h] [--input INPUT] [--fragments FRAGMENTS [FRAGMENTS ...]]
                [--sections SECTIONS] [--output OUTPUT] [--config CONFIG]
                [--kconfig KCONFIG] [--env NAME=VAL]
**ldgen.py: error: unrecognized arguments: || (set FAIL_LINE=3& goto :ABORT)**
[898/904] Building C object CMakeFiles/aws_demos.dir/demos/ble/aws_ble_gatt_server_demo.c.obj
../demos/ble/aws_ble_gatt_server_demo.c:275:19: warning: 'vGattDemoSvcHook' defined but not used [-Wunused-function]
 static BaseType_t vGattDemoSvcHook( void )
                   ^
[901/904] Building C object CMakeFiles/aws_demos.dir/demos/ota/aws_iot_ota_update_demo.c.obj
../demos/ota/aws_iot_ota_update_demo.c: In function '_establishMqttConnection':
../demos/ota/aws_iot_ota_update_demo.c:259:10: warning: unused variable 'pClientIdentifierBuffer' [-Wunused-variable]
     char pClientIdentifierBuffer[ OTA_DEMO_CLIENT_IDENTIFIER_MAX_LENGTH ] = { 0 };
          ^
../demos/ota/aws_iot_ota_update_demo.c:258:26: warning: unused variable 'willInfo' [-Wunused-variable]
     IotMqttPublishInfo_t willInfo = IOT_MQTT_PUBLISH_INFO_INITIALIZER;
                          ^
[903/904] Linking CXX executable aws_demos.elf
**FAILED: aws_demos.elf**
cmd.exe /C "cd . && C:\PROGRA~1\ESPRES~1\ESP-ID~1\tools\bin\XT1FE9~1.EXE  -mlongcalls  -nostdlib   -Wl,--gc-sections -Wl,--cref -Wl,--Map=aws_demos.map -Wl,--undefined=uxTopUsedPriority @CMakeFiles\aws_demos.rsp  -o aws_demos.elf  && cd ."
c:/progra~1/espres~1/esp-id~1/tools/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot open linker script file D:/afr-ota-1595935233664/FreeRTOS/build/esp-idf/esp32/esp32.project.ld: Invalid argument
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

D:\afr-ota-1595935233664\FreeRTOS\build>

**To reproduce**
Steps to reproduce the behaviour:

1. Login and download FreeRTOS from the FreeRTOS console.
2. Create a configuration for a ESP32-DevKitC. On the Configure FreeRTOS Software configuration properties page, under Libraries, choose OTA Updates. Under Demo projects, choose to enable the OTA demo.
3. Download and unzip to extract the folder contents.
4. Open Command Prompt with Administrator privilege.
5. Change the current working directory path to ..\FreeRTOS\
6. Run the following commands. Observe the error generated at the third step.
- cmake -DVENDOR=espressif -DBOARD=esp32_devkitc -DCOMPILER=xtensa-esp32 -DDEBUG=1 -GNinja -S . -B build
- cd build
- ninja

**Code to reproduce the bug**
N/A

**Additional context**
1. The FreeRTOS Version 202002.00 successfully compiles and links, to generate the aws_demos.bin file with OTA Update selected as the demo code.

Thank you!
Dipen

dan4thewin, 2020-07-29:
Thank you for this report.  We'll commence investigating.

dan4thewin, 2020-08-03:
Hi, @dipen-1337lab .  I've found other issue reports possibly related to this one.  If you pull the software from the github repository directly, are you able to successfully link the demo?

dipen-1337lab, 2020-08-04:
Namaste @dan4thewin, 

Below are the steps I performed to check if I am able to successfully link the demo.

Step 1 - Clone the https://github.com/aws/amazon-freertos master repository.

Step 2 - Change the below line of code in the aws_demo_config.h file. File path on my local machine is D:\amazon-freertos\vendors\espressif\boards\esp32\aws_demos\config_files

// #define CONFIG_MQTT_DEMO_ENABLED --> commented this line of code for building the default MQTT demo
#define CONFIG_OTA_UPDATE_DEMO_ENABLED --> added this line of code for building the OTA demo

Step 3 - Run the below commands on the command prompt.
a) cmake -DVENDOR=espressif -DBOARD=esp32_devkitc -DCOMPILER=xtensa-esp32 -DDEBUG=1 -GNinja -S . -B build
b) cd build
c) ninja

**The demo project was linked successfully and generated the aws_demos.bin file.**

Some console log for your reference is available below.

`D:\amazon-freertos>cmake -DVENDOR=espressif -DBOARD=esp32_devkitc -DCOMPILER=xtensa-esp32 -DDEBUG=1 -GNinja -S . -B build
-- The C compiler identification is GNU 5.2.0
-- The CXX compiler identification is GNU 5.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Program Files/Espressif/ESP-IDF Tools/tools/bin/xtensa-esp32-elf-gcc.exe
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.28.0.windows.1")
-- Submodule update
-- Checking Python dependencies...
Python requirements from D:/amazon-freertos/vendors/espressif/esp-idf\requirements.txt are satisfied.
-- IDF_TARGET not set, using default target: esp32
-- Building for target esp32

...
...
...

[70/80] Performing build step for 'bootloader'
ninja: no work to do.
[78/78] Generating aws_demos.bin
esptool.py v2.8

D:\amazon-freertos\build>`

Thanks | Regards,
Dipen

dan4thewin, 2020-08-05:
Thank you, Dipen.  I've alerted the team responsible for the FreeRTOS console, and they will pursue the matter further.  They did mention that a space within a path could cause a build to fail.  After you unzipped your download, did any of the paths contain spaces?

dipen-1337lab, 2020-08-05:
Namaste @dan4thewin,

I downloaded the Amazon FreeRTOS code today again (just now) and compiled the code to generate the binary file. Interestingly, the issue is not observed anymore and **aws_demos.bin** was generated successfully. Looks like it has been resolved.

With regards to a possible space within the path, below are my observations.

**Attempt 1 - Folder path: D:\folder name with space\freertos-1596609209069\FreeRTOS>**
**Observation: The space in the path seems to generate a failure for the cmake command.**
**Console Log**
D:\folder name with space\freertos-1596609209069\FreeRTOS>cmake -DVENDOR=espressif -DBOARD=esp32_devkitc -DCOMPILER=xtensa-esp32 -DDEBUG=1 -GNinja -S . -B build
-- The C compiler identification is GNU 5.2.0
-- The CXX compiler identification is GNU 5.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Program Files/Espressif/ESP-IDF Tools/tools/bin/xtensa-esp32-elf-gcc.exe
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.28.0.windows.1")
-- Checking Python dependencies...
Python requirements from D:/folder name with space/freertos-1596609209069/FreeRTOS/vendors/espressif/esp-idf\requirements.txt are satisfied.
-- IDF_TARGET not set, using default target: esp32
-- Building for target esp32
CMake Error at vendors/espressif/esp-idf/tools/cmake/scripts/expand_requirements.cmake:144 (list):
  list GET given empty list
Call Stack (most recent call first):
  vendors/espressif/esp-idf/tools/cmake/scripts/expand_requirements.cmake:199 (filter_components_list)

CMake Error at vendors/espressif/esp-idf/tools/cmake/idf_functions.cmake:339 (message):
  Failed to expand component requirements
Call Stack (most recent call first):
  vendors/espressif/boards/esp32/CMakeLists.txt:496 (idf_import_components)
  CMakeLists.txt:70 (include)

-- Configuring incomplete, errors occurred!
See also "D:/folder name with space/freertos-1596609209069/FreeRTOS/build/CMakeFiles/CMakeOutput.log".
See also "D:/folder name with space/freertos-1596609209069/FreeRTOS/build/CMakeFiles/CMakeError.log".

D:\folder name with space\freertos-1596609209069\FreeRTOS>

Log files are provided below.
[CMakeError.log](https://github.com/aws/amazon-freertos/files/5026687/CMakeError.log)
[CMakeOutput.log](https://github.com/aws/amazon-freertos/files/5026688/CMakeOutput.log)

**Attempt 2 - Folder path: D:\folderNameWithoutSpace\freertos-1596609209069\FreeRTOS>**
**Observation: The binary image is generated successfully for a folder path without spaces.**
**Console Log**
D:\folderNameWithoutSpace\freertos-1596609209069\FreeRTOS>cmake -DVENDOR=espressif -DBOARD=esp32_devkitc -DCOMPILER=xtensa-esp32 -DDEBUG=1 -GNinja -S . -B build
-- The C compiler identification is GNU 5.2.0
-- The CXX compiler identification is GNU 5.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Program Files/Espressif/ESP-IDF Tools/tools/bin/xtensa-esp32-elf-gcc.exe
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.28.0.windows.1")
-- Checking Python dependencies...
Python requirements from D:/folderNameWithoutSpace/freertos-1596609209069/FreeRTOS/vendors/espressif/esp-idf\requirements.txt are satisfied.
-- IDF_TARGET not set, using default target: esp32
-- Building for target esp32
...
... [ Skipping the complete log ]
...
[904/904] Generating aws_demos.bin
esptool.py v2.8

D:\folderNameWithoutSpace\freertos-1596609209069\FreeRTOS\build>

**Attempt 3 - Folder path: D:\gitHub\..\longFolderPath\..\FreeRTOS\build>**
**Observation: The binary image is generated successfully inside a folder with a total path length of 119 characters including backslashes and, the path does not contain spaces.**
**Remark: This is the path on my local system for project development. Hence, sharing this Attempt 3 observation additionally.**
**Console Log**
D:\gitHub\..longFolderPath\..\FreeRTOS>cmake -DVENDOR=espressif -DBOARD=esp32_devkitc -DCOMPILER=xtensa-esp32 -DDEBUG=1 -GNinja -S . -B build
-- The C compiler identification is GNU 5.2.0
-- The CXX compiler identification is GNU 5.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Program Files/Espressif/ESP-IDF Tools/tools/bin/xtensa-esp32-elf-gcc.exe
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.28.0.windows.1")
-- Checking Python dependencies...
Python requirements from D:/gitHub/../longFolderPath/../FreeRTOS/vendors/espressif/esp-idf\requirements.txt are satisfied.
-- IDF_TARGET not set, using default target: esp32
-- Building for target esp32
...
... [ Skipping the complete log ]
...

[904/904] Generating aws_demos.bin
esptool.py v2.8

D:\gitHub\..\longFolderPath\..\FreeRTOS\build>

Hope this helps! We may close this issue now.

Thanks | Regards,
Dipen

closed 2020-08-09 by yourslab