aws/amazon-freertos #2322: Error -2180 in mbedtls_x509_crt_parse function - MBEDTLS_ERR_X509_INVALID_FORMAT

Darsh-Dev, 2020-07-29:
Hi,
I  am working on porting AWS freeRTOS to a new platform I referred freertos programming guide to do that.

Now my issue is, I m facing certificate parsing error in mbedtls_x509_crt_parse() function call while performing TLS_Connect() in iot_tls.c file , Mbedtls version string is "mbed TLS 2.16.0".

I have attached my mbedtls/config.h here for reference and after debugging I enabled MBEDTLS_X509_ALLOW_EXTENSIONS_NON_V3 and MBEDTLS_X509_ALLOW_UNSUPPORTED_CRITICAL_EXTENSION macro but still no luck.

I have created a certificate from the one-click method from AWS console and then used tools/certificate_configuration/CertificateConfigurator.html tool to get freeRTOS supported aws_clientcredential_keys.h file and replace in my code.

Also if I apply Linux parse command with OpenSSL,

`openssl x509 -in x.y.z-certificate.pem.crt -text -noout`

it's working and showing output as an attached file.

is there anything I m missing here?

Any guidance will be appreciable.

[openssl_output.txt](https://github.com/aws/amazon-freertos/files/4992538/openssl_output.txt)
[config.txt](https://github.com/aws/amazon-freertos/files/4992547/config.txt)

Thanks

lundinc2, 2020-07-29:
Hello @Darsh-Dev,

Can you identify where the MBEDTLS_ERR_X509_INVALID_FORMAT error is occurring?

dan4thewin, 2020-07-30:
Hi, @Darsh-Dev ,
As another data point, you can try your aws_clientcredential_keys.h file with one of the demos using the windows simulator to see if the credentials permit a successful connection there.

lundinc2, 2020-08-03:
Hi @Darsh-Dev can I am going to close this issue as a duplicate of https://github.com/Infineon/amazon-freertos/issues/5. Please re-open this should you need further assistance.

Thanks!

closed 2020-08-03 by lundinc2