aws/amazon-freertos #2327: Unable to connect to GGC in docker using the GG discovery demo

OZsGit, 2020-08-03:
**Describe the bug**
Running the GG discovery demo, ESP32 fails to connect to the Greengrass core if the GGC is running in a docker container. If GGC is running on the system not in a docker container it is able to connect.

**System information**
- Which hardware board or part numbers?: ESP32-WROVER-KIT
- IDE used: N/A
- Operating System: Windows 10 to flash the ESP32/Ubuntu 18.04.4 LTS running Docker 19.03.12 running Greengrass
- Version of FreeRTOS: 202007.00-96-g5c345b395
- Project: Greengrass discovery demo 

**Expected behavior**
The demo is expected to successfully discover the GGC, connect, and run the remaining demo tasks.

**Screenshots or console output**
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x1e (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:7184
ho 0 tail 12 room 4
load:0x40078000,len:12652
load:0x40080400,len:6708
entry 0x40080778
I (76) boot: Chip Revision: 1
I (76) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot: ESP-IDF v3.3.2-235-ga3ce2f400 2nd stage bootloader
I (41) boot: compile time 15:41:08
I (41) boot: Enabling RNG early entropy source...
I (47) boot: SPI Speed      : 40MHz
I (51) boot: SPI Mode       : DIO
I (55) boot: SPI Flash Size : 4MB
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00010000 00006000
I (78) boot:  1 otadata          OTA data         01 00 00016000 00002000
I (85) boot:  2 phy_init         RF data          01 01 00018000 00001000
I (93) boot:  3 ota_0            OTA app          00 10 00020000 00177000
I (100) boot:  4 ota_1            OTA app          00 11 001a0000 00177000
I (108) boot:  5 storage          WiFi data        01 02 00317000 00010000
I (115) boot: End of partition table
I (120) boot: ota rollback check done
I (124) boot_comm: chip revision: 1, min. application chip revision: 0
I (131) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x279c0 (16                                                                                      2240) map
I (197) esp_image: segment 1: paddr=0x000479e8 vaddr=0x3ffbdb60 size=0x037d8 ( 1                                                                                      4296) load
I (203) esp_image: segment 2: paddr=0x0004b1c8 vaddr=0x40080000 size=0x00400 (                                                                                        1024) load
I (204) esp_image: segment 3: paddr=0x0004b5d0 vaddr=0x40080400 size=0x04a40 ( 1                                                                                      9008) load
I (221) esp_image: segment 4: paddr=0x00050018 vaddr=0x400d0018 size=0xb3e80 (73                                                                                      6896) map
I (480) esp_image: segment 5: paddr=0x00103ea0 vaddr=0x40084e40 size=0x16694 ( 9                                                                                      1796) load
I (534) boot: Loaded app from partition at offset 0x20000
I (534) boot: Disabling RNG early entropy source...
I (535) cpu_start: Pro cpu up.
I (538) cpu_start: Application information:
I (543) cpu_start: Project name:     esp-idf
I (548) cpu_start: App version:      202007.00-96-g5c345b395
I (555) cpu_start: Compile time:     Aug  4 2020 09:38:52
I (561) cpu_start: ELF file SHA256:  dcec516b658ca364...
I (567) cpu_start: ESP-IDF:          v3.3.2-235-ga3ce2f400
I (573) cpu_start: Single core mode
I (577) heap_init: Initializing. RAM available for dynamic allocation:
I (584) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (590) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (596) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (602) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (609) heap_init: At 3FFCC920 len 000136E0 (77 KiB): DRAM
I (615) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
I (621) heap_init: At 40078000 len 00008000 (32 KiB): IRAM
I (627) heap_init: At 4009B4D4 len 00004B2C (18 KiB): IRAM
[... 65 lines cut ...]
18 2650 [iot_thread] [INFO ][MQTT][26500] MQTT library cleanup done.
I (26887) wifi:flush txq
I (26887) wifi:stop sw txq
I (26897) wifi:lmac stop hw txq
19 2650 [iot_thread] [INFO ][DEMO][26500] Cleaned up MQTT library.
20 2650 [iot_thread] [INFO ][DEMO][26500] memory_metrics::freertos_heap::before::bytes::194116
I (26907) wifi:Deinit lldesc rx mblock:10
21 2650 [iot_thread] [INFO ][DEMO][26500] memory_metrics::freertos_heap::after::bytes::96872
22 2652 [iot_thread] [INFO ][DEMO][26520] memory_metrics::demo_task_stack::before::bytes::8920
23 2652 [iot_thread] [INFO ][DEMO][26520] memory_metrics::demo_task_stack::after::bytes::4952
24 2652 [iot_thread] [ERROR][DEMO][26520] Error running demo.
25 2662 [iot_thread] [INFO ][INIT][26620] SDK cleanup done.
26 2662 [iot_thread] [INFO ][DEMO][26620] -------DEMO FINISHED-------

**To reproduce**
Steps to reproduce the behavior:

1. Setup a GGC container image in a Docker container
2. Configure the GG group from the AWS GG as required for containerized GG images.
3. Modify aws_clientcredential.h and aws_clientcredential_keys.h with necessary details and flash the ESP32 with the AFR GG discovery demo
4. Add the ESP32 IoT device as a device to the GG group
5. Configure a subscription in the GG group with the ESP32 Thing as the publisher and the GGC device as the subscriber
6. Start Docker on the target machine. Start Greengrass in a Docker container
7. Deploy the GG group to the containerized Docker image
8. Run the AFR GG discovery demo

dan4thewin, 2020-08-05:
Hi @OZsGit .  Could you provided a packet capture of each scenario?

OZsGit, 2020-08-05:
> Hi @OZsGit . Could you provided a packet capture of each scenario?

I'm sorry, how do I capture packets? What packets should I try to capture?

yourslab, 2020-08-05:
> > Hi @OZsGit . Could you provided a packet capture of each scenario?
> 
> I'm sorry, how do I capture packets? What packets should I try to capture?

You may use Wireshark, tcpdump, or a similar utility.

Also, it would be helpful to show the Dockerfile or Docker Image that you use. Since SOCKETS_Connect is failing, it could be that one of the ports are not forwarded correctly.

OZsGit, 2020-08-06:
Edited the original post with Wireshark capture data and corresponding ESP32 device serial outputs for both scenarios. I'm not sure how to show the Dockerfile or Docker Image, as I simply followed the [instructions from AWS](https://docs.aws.amazon.com/greengrass/latest/developerguide/run-gg-in-docker-container.html) to pull the GGC Docker container image. I've also added the output of inspecting this image with docker image inspect [repository name].

yourslab, 2020-08-06:
Thank you very much for the comprehensive report. 

It seems that port 443 of the GGC is not accessible. If you used the following command to run the GGC using Docker:
```bash
docker run --rm --init -it --name aws-iot-greengrass --entrypoint /greengrass-entrypoint.sh -v c:/Users/%USERNAME%/Downloads/certs:/greengrass/certs -v c:/Users/%USERNAME%/Downloads/config:/greengrass/config -p 8883:8883 216483018798.dkr.ecr.us-west-2.amazonaws.com/aws-iot-greengrass:latest
```

You must change it to the following in order to forward ports listed in the AWS IoT Greengrass discovery demo application under Inbound Rules:
```bash
docker run --rm --init -it --name aws-iot-greengrass --entrypoint /greengrass-entrypoint.sh -v c:/Users/%USERNAME%/Downloads/certs:/greengrass/certs -v c:/Users/%USERNAME%/Downloads/config:/greengrass/config -p 8883:8883 -p 443:443 -p 80:80 216483018798.dkr.ecr.us-west-2.amazonaws.com/aws-iot-greengrass:latest
```

Essentially, we need to add `-p 443:443 -p 80:80` so that the ports needed by the discovery demo application are reachable.

Please feel free to reach out if you encounter more issues. Thanks again for your patience.

OZsGit, 2020-08-06:
Thanks @yourslab, I tried this and it doesn't appear to have worked. It fails in what appears to be the same way as before, at least according to the serial output. I've updated the original post with both serial output and packet capture for this scenario.

yourslab, 2020-08-07:
I can see from the packet capture that your ESP32 is not able to discover the GGC. It appears that the Docker container is advertising its own IP address rather than the IP address of the host system (Ubuntu 18.04) to AWS IoT. 

Please go to `AWS IoT` -> `Greengrass` -> `Cores` -> [`Name of Core`] -> `Connectivity`. 

Under Connectivity, you should add port 8883 and the IP address of the host system (Ubuntu 18.04) where the Docker container is running. The IP address should be in the form of 192.168.0.?.

OZsGit, 2020-08-08:
I checked and indeed the core endpoint doesn't show the IP address of the host system. The following two endpoints are shown:
[image]
Presumably there are two because I deployed the same group to GG on both the host system and in Docker. What should I do to address this?

yourslab, 2020-08-08:
Please try to add the IP address of the host system in the form of 192.168.0.* and port 8883.

OZsGit, 2020-08-08:
Adding the core endpoint has enabled the GG discovery demo to discovery the GG core and complete the demo tasks. The steps taken to resolve this were:
1. Start Docker and launch Greengrass in a container using
```shell
docker run --rm --init -it --name aws-iot-greengrass --entrypoint /greengrass-entrypoint.sh -v c:/Users/%USERNAME%/Downloads/certs:/greengrass/certs -v c:/Users/%USERNAME%/Downloads/config:/greengrass/config -p 8883:8883 -p 443:443 -p 80:80 216483018798.dkr.ecr.us-west-2.amazonaws.com/aws-iot-greengrass:latest
```
2. Deploy the Greengrass group to the containerized core from AWS
3. From `AWS IoT` -> `Greengrass` -> `Cores` -> [`Name of Core`] -> `Connectivity` manually add the local IP address of the host system with port 8883. 
4. Launch the FreeRTOS GG discovery demo on the ESP32.

Strangely if I tried to add the additional connection details _before_ deploying the GG group, after the deployment the connection details would be reset and the additional 192.168.0.? IP details would no longer be present. Starting the FreeRTOS demo at this point would again result in failure. This necessitated modifying the connection details after the deployment. Why would this be the case and how can I prevent the connection details from resetting? This may not be the appropriate thread or location for this discussion.

yourslab, 2020-08-08:
Ahh that is a good point @OZsGit. I will forward that to the Greengrass Core team. I will close this issue since it has been resolved, but if you encounter any blocker again, please feel free to reach out.

Thank you again for being very descriptive about the steps you've taken to resolve this problem!

closed 2020-08-08 by yourslab