aws/amazon-freertos #2342: Experimenting with CBMC with and without patches

rvantonder, 2020-08-07:
Hi 👋 . I chatted a bit with @danielsn and I'm interested in experimenting with some of the patch preprocessing used with this project and `cbmc`, similar to the ideas of the [existing patches](https://github.com/aws/amazon-freertos/tree/master/tools/cbmc/patches). I was able to get up and running with the `ctest` command in the [README](https://github.com/aws/amazon-freertos/tree/master/tools/cbmc#cmake-based-build). 

The tests seem flaky (I'm on Mac OS X). Either two or four tests sometimes fail, but I'll just ignore this unless this is a real problem. For reference:

```
The following tests FAILED:
         12 - IotHttpsClient_Disconnect-report (Failed)
         28 - IotHttpsClient_SendSync-report (Failed)
```

```
The following tests FAILED:
          8 - IotHttpsClient_Connect-report (Failed)
         12 - IotHttpsClient_Disconnect-report (Failed)
         20 - IotHttpsClient_ReadHeader-report (Failed)
         28 - IotHttpsClient_SendSync-report (Failed)
```

Couple of questions:

I'd basically like to, for example, add my own patch and see whether that changes the behavior of cbmc. To do that, it'd be helpful to understand how the current patches are applied to some analysis pipeline. I looked at some of the scripts, makefiles, and patches, but it's not clear to me how they fit together. It'd be great if there's a simple rundown of:

- What command to enter that uses the patches for some analysis (does `ctest` do this, or is there another command?)

- The scripts and steps of what the above command invokes (so that I can understand how they fit together, and modify parts of it). I'm interested in adding or removing patches, understanding how they affect the current analysis, and what the analysis target is (set of programs or subparts that are analyzed). 

Thanks!

yourslab, 2020-08-09:
Hi @rvantonder. Great questions. 

The patches are just applied using `git apply [name].patch` by [patch.py](https://github.com/aws/amazon-freertos/blob/master/tools/cbmc/patches/patch.py). These git patches are simply applied to the original amazon-freertos codebase, so you should also be able to achieve the same effect by modifying the original codebase.

rvantonder, 2020-08-13:
Thanks a lot for the response @yourslab. OK, I went back and figured out that `patch.py` needs to be run inside the directory where it lives and can see that the patches apply to the project. A last couple of questions:

1. After running `./patch.py`, what is the typical command to run CBMC on amazon-freertos on the changed files? Is it only the `ctest --build-and-test ...` in the README ?

2. Is it expected that some of the tests (that I mentioned above) fail for the `ctest --build-and-test ...` command?

3.  My understanding is that some of the patches were introduced to also overcome some (possibly performance related) issue in CBMC. I'm thinking specifically about `0002-Change-FreeRTOS_IP_Private.h-union-to-struct.patch`. This one mentions:

> This patch should be removed soon: It is for a CBMC issue being fixed now.

I'd be very interested if there's a reference handy to the fix in CBMC so I can understand that context, if you have that handy?

AniruddhaKanhere, 2020-08-13:
Hello @rvantonder,

If you do not want to set-up everything manually, you can run `python3 prepare.py` (only Python version 3 and above works in this case) in the [proofs directory](https://github.com/aws/amazon-freertos/tree/master/tools/cbmc/proofs). The script will apply the patches for you and set-up the Makefiles based on the configuration in `Makefile.json` in each proof subdirectory. See steps [here](https://github.com/aws/amazon-freertos/tree/master/tools/cbmc#setting-up-the-proofs).

To add to what @yourslab said, the patches are used only to make the source accessible by cbmc infrastructure. It removes `static`, `volatile` etc. keywords in the source code. But except that, cbmc does not _require_ patches. If you modify certain patches, my guess would be that certain proofs might fail since they will not be able to access the static functions under test. In some cases, the proofs might work correctly. It depends on what changes the modified patches make.

To answer your questions:
1. If you have run prepare.py without any failures, then you just need to go to the directory of each proof and run `make cbmc`. It would generate a cbmc.txt file for you which will have the results enlisted.

2. If you have not modified anything, then the tests should not fail. Every single one of them is supposed to pass. Do note however that some tests use quite a bit of CPU and quite a bit of memory wherein they might be stopped by the OS. In these cases, you'll see `Process Killed` message at the end of `cbmc.txt`.

3. Yes. Your understanding is correct. But now we are moving towards removing as many patches as we can. See the [CBMC folder](https://github.com/FreeRTOS/FreeRTOS/tree/master/FreeRTOS-Plus/Test/CBMC) of FreeRTOS/FreeRTOS. It is identical to the one in amazon-FreeRTOS except the patches.

I hope this answers you questions. Feel free to ping us back with any more questions you might have. :)

rvantonder, 2020-08-14:
Hi @AniruddhaKanhere. That's great, thank you! I think I have everything I need to experiment with some things.

> But except that, cbmc does not require patches. If you modify certain patches, my guess would be that certain proofs might fail since they will not be able to access the static functions under test. [...]

Got it, that's super clear now.

> But now we are moving towards removing as many patches as we can. See the CBMC folder of FreeRTOS/FreeRTOS. It is identical to the one in amazon-FreeRTOS except the patches.

Useful, thanks! Looks like the current set of patches then is to remove `static`, and there's one for `volatile`. I understand the context of those changes now.

I'll close this issue since I don't have more questions, and I don't think I'll have more. If I do, I can just reopen it :-) Cheers!

closed 2020-08-14 by rvantonder