aws/amazon-freertos #2350: Common_io not ported on esp32

YTvW, 2020-08-11:
**Describe the bug**
I am trying to use  aws freeRTOS with an esp32, and I need to use the i2c functionality.
except I can not import the "iot_i2c.h" file since the "abstraction/common_io" module is disabled due to the following error:
```
=========================Resolving dependencies==========================
module disabled: common_io
reason:          common_io::mcu_port is not defined by vendor.
dependency path: common_io->common_io::mcu_port
```

**System information**
- ESP32WROOM
- visual studio Code 
- MacOS

yanjos-dev, 2020-08-12:
Hi @YTvW, thank you for your feedback. Support for common IO on the ESP32 board is currently under development and can be monitored in this PR #1946 .

lundinc2, 2020-11-12:
@YTvW I am closing this issue as we have merged the PR now. Please feel free to re-open here with further questions.

Thanks,

Carl

closed 2020-11-12 by lundinc2