aws/amazon-freertos #2351: IOT_STATIC_MEMORY_ONLY is not obeyed everywhere

lightblu, 2020-08-11:
**Describe the bug**

`IOT_STATIC_MEMORY_ONLY` is not obeyed everywhere, at least not for some task creations. Not sure if this is complete, but the task creations in `abstractions/platform/freertos/iot_threads_freertos.c` for `iot_thread` tasks and in `abstractions/platform/freertos/iot_network_freertos.c` for `NetRecv` tasks use `xTaskCreate`.

**Expected behavior**

All IoT related task creations obey `IOT_STATIC_MEMORY_ONLY` and should only use `xTaskCreateStatic` functions.

**Additional context**

I have two other minor comments I wondered about and like to mention/ask:

1) Why do `IotThreads_Malloc` and `IotThreads_Free` default to `malloc` and `free` in `abstractions/platform/freertos/iot_threads_freertos.c`? Wouldn't it be more reasonable for them to default do `pvPortMalloc` and `pvPortFree`?

2) Seeing 

```
typedef struct iot_mutex_internal
{
    StaticSemaphore_t xMutex; /**< FreeRTOS mutex. */
    BaseType_t recursive;     /**< Type; used for indicating if this is reentrant or normal. */
} iot_mutex_internal_t;
```

and its creation 

```
xSemaphoreCreateMutexStatic( &internalMutex->xMutex );
```

and usage 

```
lockResult = xSemaphoreTake( ( SemaphoreHandle_t ) &internalMutex->xMutex, timeout );
```

made me really wonder if this is correct (as I was searching for my own issue). Needed to double check, agreeding now it is unlikely that this would be  implemented differently.
But still, that `SemaphoreHandle_t` (the handle for a semaphore) actually is just the address of `StaticSemaphore_t` (the memory for a semaphore) is implementation knowledge and not documented or guaranteed, by documentation the buffer and the handle are unrelated things. So if such shortcut is taken it should be at least commented well or maybe even asserted on at creation :)

yanjos-dev, 2020-08-13:
Hi @lightblu, I read through your feedback and agree with what you have said. 

Starting with the IOT_STATIC_MEMORY_ONLY macro, I agree that the value of this macro declines when we have modules that don't use it. I'll look into it to see if I can correct this without altering any of our APIs. At the minimum, I'll update our documentation to be explicit about what modules this macro impacts. 

I looked at the IotThreads_Malloc and IotThreads_Free macros and I agree, this is a mistake. I created a PR ( #2357 ) to fix it. 

Your analysis of the of the way we are using the semaphore handle is correct. Passing the direct address of the semaphore in as a handle is not a good practice and is something that needs to be improved on.

yanjos-dev, 2020-10-23:
@lightblu I'm closing this ticket for now due to inactivity. Please reopen this ticket if you have any further questions.

closed 2020-10-23 by yanjos-dev