aws/amazon-freertos #2361: OTA Agent calls Activate even if final MQTT PUBLISH fails

stevedefrisco, 2020-08-13:
**Describe the bug**
The OTA Agent always calls prvPAL_ActivateNewImage() after trying to PUBLISH final status message to MQTT, regardless of success/failure of the PUBLISH. The MQTT broker did not get the message. The Device Activates the new version. This puts the Device and the MQTT server out of sync.
As a result, when the Device updates the application and reboots, the Job it gets is the same update Job (which it already executed), and after the update/reboot, it fails the self_test as the previous and new versions are the same.

**System information**
- CY8CPROTO_062_4343W
- Using IDT 3.1.0 from command liine
- Linux (Ubuntu)
- 202007.00-13-g0c84d1095
- OTA IDT 3.1 tests OTATestDisconnectAndResume for HTTP and MQTT

**Expected behavior**
If MQTT PUBLISH message for {"status":"IN_PROGRESS","statusDetails":{"self_test":"ready","updatedBy":"0x90002"}} fails, the OTA_ActivateNewImage() function should not be called, and Device should not update the application.

**Screenshots or console output**

Test console Summary:
```
 ========== Test Summary ==========
 Execution Time: 	15m46s
 Tests Completed: 	1
 Tests Passed: 		0
 Tests Failed: 		1
 Tests Skipped: 		0
 ----------------------------------
 Test Groups:
 Optional Test Groups:
     OTADataplaneHTTP: 		FAILED
 ----------------------------------
 Failed Tests:
     Group Name: OTADataplaneHTTP
         Test Name: OTATestDisconnectAndResume
             Reason: rejected: 0x23000000

```

Log output:
```
5195 165250 [OTA Agent] [prvIngestDataBlock] Received final expected block of file.
5196 165250 [OTA Agent] [prvStopRequestTimer] Stopping request timer.
5197 166577 [OTA Agent] [prvIngestDataBlock] File receive complete and signature is valid.
5198 166577 [OTA Agent] [prvStopRequestTimer] Stopping request timer.
5199 166577 [OTA Agent] [prvPublishStatusMessage] Msg: {"status":"IN_PROGRESS","statusDetails":{"self_test":"ready","updatedBy":"0x90002"}}
5200 166577 [iot_threa] [ERROR][NET][lu] Error -27648 while sending data.
5201 166578 [OTA Agent] [INFO ][MQTT][lu] (MQTT connection 0x80442e8) MQTT PUBLISH operation queued.
5202 166578 [OTA Agent] [INFO ][MQTT][lu] (MQTT connection 0x80442e8, PUBLISH operation 0x8045168) Waiting for operation completion.
5203 166578 [OTA Agent] [INFO ][MQTT][lu] (MQTT connection 0x80442e8, PUBLISH operation 0x8045168) Wait complete with result NETWORK ERROR.
5204 166578 [OTA Agent] [prvPublishStatusMessage] Failed: $aws/things/idt-4119149353932166240/jobs/AFR_OTA-idt-2616811108046347440/update
5205 166578 [OTA Agent] [INFO ][DEMO][lu] Received eOTA_JobEvent_Activate callback from OTA Agent.

5206 166578 [OTA Agent] [INFO ][MQTT][lu] (MQTT connection 0x80442e8) Disconnecting connection.
5207 166579 [iot_threa] [ERROR][NET][lu] Error -27648 while sending data.
5216 166582 [iot_threa] [INFO ][DEMO][lu] State: WaitingForFileBlock  Received: 825   Queued: 0   Processed: 0   Dropped: 0

5217 166583 [iot_threa] [prvStopRequestTimer] Stopping request timer.
5218 167581 [OTA Agent] prvPAL_ResetDevice()   RESETTING NOW !!!!
```

**Comments on log output**
Notes:
Line 5200 - error when sending the Status Message libraries/abstractions/platform/freertos/iot_network_freertos.c : IotNetworkAfr_Send()
Line 5203 - NETWORK ERROR
Line 5204 - prvPublishStatusMessage() prints Failed message
Line 5205 - Still calls Activate

**To reproduce**
The error occurred running IDT 3.1.0 for tests OTATestDisconnectAndResume for HTTP and MQTT on a CY8CYPROTO_062_4343W board. I realize this board is not part of the release. This occurs on our test machine, but not when I run the IDT at my desk. While investigating, I came across this issue.

**Code to reproduce the bug**
The call chain is:
prvProcessDataHandler()->xOTA_ControlInterface.prvUpdateJobStatus()->prvUpdateJobStatus_Mqtt()->prvPublishStatusMessage()->prvPublishMessage()

- prvUpdateJobStatus_Mqtt() always returns kOTA_Err_None;
- prvPublishStatusMessage() Prints an error message from prvPublishMessage() call, but is a void function and cannot pass the result up the call chain;

**Possible solution**
If prvProcessDataHandler() received an appropriate error message from the xOTA_ControlInterface.prvUpdateJobStatus() call, it could take appropriate action and not call xOTA_Agent.xPALCallbacks.xCompleteCallback(eOTA_JobEvent_Activate) which calls OTA_ActivateNewImage(), causing the Device to update the code and reboot.

**Additional context**
Add any other context about the problem here.

Cheers,
Steve

pvyawaha, 2020-08-14:
Hello Steve, 

Thank you for reporting this. We are working on propagating the error code from MQTT APIs up in the OTA library to block on resetting the device and will update this issue with the PR details.

stevedefrisco, 2020-08-17:
Hi Prasad,
Thank you for looking into this issue. Along the same lines, for the same IDT test, all of the status updates sent to the MQTT broker fail. The status updates are sent every 128 blocks, and when the status fails (because the MQTT broker disconnected), the OTA Agent needs to recognize this and reconnect to the broker. Example output:

1252 435835 [OTA Agent] [prvUpdateJobStatus_Mqtt] Calling prvPublishStatusMessage() {"status":"IN_PROGRESS","statusDetails":{"receive":"128/825"}}1253 435835 [OTA Agent] [prvPublishStatusMessage] Msg: {"status":"IN_PROGRESS","statusDetails":{"receive":"128/825"}}
1254 435835 [iot_threa] [ERROR][NET][lu] Error -27648 while sending data.
1255 435835 [OTA Agent] [INFO ][MQTT][lu] (MQTT connection 0x8044238) MQTT PUBLISH operation queued.
1256 435835 [OTA Agent] [prvPublishStatusMessage] 'IN_PROGRESS' to $aws/things/idt-419412977478892574/jobs/AFR_OTA-idt-7140749478840665491/update

Cheers,
Steve

pvyawaha, 2020-10-02:
Hello, 

We are working on both the recommendation and I will update this thread when PR is created.

pvyawaha, 2020-10-12:
Hello, 

We have added and merged a check to avoid activating the image when updating the final status fails. The PR that is merged is https://github.com/aws/amazon-freertos/pull/2534

closed 2020-10-15 by pvyawaha