aws/amazon-freertos #2365: OTA update receives random packet size of 4200 bytes instead of 4096 bytes

Sunilpuli188, 2020-08-14:
Hi Team,

When I am performing OTA update, I receive random packet size of 4200 bytes some times against the expected size of 4096 bytes.

I do not see this issue happening when doing OTA update over BLE, but I see this issue happening in OTA update over Wi-Fi.

I am using FreeRTOS Version 202002.00 and I am using ESP32.

Please find the screenshot for the log attached here.
[image]

Kindly let me know what can be the cause of this problem?

Thank you.
SK

pvyawaha, 2020-08-14:
Hello,

Thank you for sharing logs and other details. The error you are getting is due to memory allocation failure in MQTT library on an incoming packet. The size of this incoming packet will be slightly larger than the OTA block size ( 4096 in your case ) due to additional header. So 4200 size of the packet is not the issue. 

This issue is not reproducible on the OTA demo and also from logs I see you have application code and other libraries used reducing available heap space. To fix this we have to understand the memory required for additional libraries and application. We can insert calls to get free heap available and maximum block size available in code to do that. Please refer to the blog post [ESP32-Memory-Blog](https://medium.com/the-esp-journal/esp32-memory-analysis-case-study-eacc75fe5431).

It seems while using BLE enough heap space is available and you are not observing this issue. Do you have a requirement to use BLE while WiFi is on or can it be offloaded?

Also, we can reduce OTA library heap requirement by updating the OTA configuration parameter otaconfigLOG2_FILE_BLOCK_SIZE and set it to (10) 1 KB block size.  [Esp32 config file](https://github.com/aws/amazon-freertos/blob/master/vendors/espressif/boards/esp32/aws_demos/config_files/aws_ota_agent_config.h)

Also from Espressif, It is also possible to utilize residual instruction memory (IRAM) for dynamic allocation or moving data/bss of libraries/objects. You may find relevant config option at, https://github.com/espressif/esp-afr-sdk/blob/release/v3.3/components/esp32/Kconfig#L1080. Please note that based on access pattern (unaligned accesses), it may incur performance penalty.

Sunilpuli188, 2020-08-17:
Hi Prasad,

Thanks for the detailed explanation.
I could manage the Heap correctly now. I do not see any issue in OTA now.

You can close this issue.

Once again thanks a lot for the detailed explanation.

Thank you,
SK.

closed 2020-08-17 by Sunilpuli188