aws/amazon-freertos #2366: FreeRTOS can't publish data to IoT Core through Greengrass

monhack, 2020-08-14:
**Describe the bug**
I'm trying to have my ESP32 connect to a iot core. The problem is while esp32 connect iot core, connection failed because it can't find endpoint or connect a remote core. 

**System information**

- ESP32 WROOM32D
- Visual Studio Code
- Operating System : Linux
- Version of Amazon FreeRTOS : 202007.00 release
- Project : MQTTGreengrass|Sub pub Demo 

**Expected behavior**
FreeRTOS accepted publish data to iot core

**Screenshots or console output**
$ python .vendors/espressif/esp-idf/tools/idf.py monitor -p /dev/ttyUSB0 -B directory
.........
 --- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:7180
load:0x40078000,len:12652
ho 0 tail 12 room 4
load:0x40080400,len:6708
entry 0x40080778
I (72) boot: Chip Revision: 1
I (72) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (40) boot: ESP-IDF HEAD-HASH-NOTFOUND 2nd stage bootloader
I (40) boot: compile time 12:43:26
I (40) boot: Enabling RNG early entropy source...
I (46) boot: SPI Speed      : 40MHz
I (50) boot: SPI Mode       : DIO
I (54) boot: SPI Flash Size : 4MB
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00010000 00006000
I (76) boot:  1 otadata          OTA data         01 00 00016000 00002000
I (84) boot:  2 phy_init         RF data          01 01 00018000 00001000
I (91) boot:  3 ota_0            OTA app          00 10 00020000 00177000
I (99) boot:  4 ota_1            OTA app          00 11 001a0000 00177000
I (106) boot:  5 storage          WiFi data        01 02 00317000 00010000
I (114) boot: End of partition table
I (118) boot: ota rollback check done
I (122) boot_comm: chip revision: 1, min. application chip revision: 0
I (129) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x287e8 (165864) map
I (197) esp_image: segment 1: paddr=0x00048810 vaddr=0x3ffbdb60 size=0x037fc ( 14332) load
I (203) esp_image: segment 2: paddr=0x0004c014 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/monhack/Documents/Magang/FreeRTOS/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/xtensa_vectors.S:1713

I (204) esp_image: segment 3: paddr=0x0004c41c vaddr=0x40080400 size=0x03bf4 ( 15348) load
I (219) esp_image: segment 4: paddr=0x00050018 vaddr=0x400d0018 size=0xb3d34 (736564) map
0x400d0018: _flash_cache_start at ??:?

I (480) esp_image: segment 5: paddr=0x00103d54 vaddr=0x40083ff4 size=0x174e0 ( 95456) load
0x40083ff4: phy_unforce_wifi_chan at /home/aiqin/git_tree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7_ana.c:756

I (535) boot: Loaded app from partition at offset 0x20000
I (535) boot: Disabling RNG early entropy source...
I (536) cpu_start: Pro cpu up.
I (539) cpu_start: Application information:
I (544) cpu_start: Project name:     esp-idf
I (549) cpu_start: App version:      1
I (554) cpu_start: Compile time:     Aug 14 2020 12:42:39
I (560) cpu_start: ELF file SHA256:  2ae223720d7829f4...
[... 87 lines cut ...]
I (21297) wifi:new:<2,0>, old:<2,1>, ap:<255,255>, sta:<2,1>, prof:1
I (21307) WIFI: SYSTEM_EVENT_STA_DISCONNECTED: 8
E (21307) wifi:wifi deinit internal: wifi not stop
30 2097 [iot_thread] [INFO ][MQTT][20970] (MQTT connection 0x3ffb5f84) Network connection destroyed.
31 2097 [iot_thread] [INFO ][DEMO][20970] Disconnected from the broker.
32 2097 [iot_thread] Heap low watermark: 97800. Stack high watermark: 2696.
I (21337) wifi:flush txq
I (21337) wifi:stop sw txq
I (21347) wifi:lmac stop hw txq
33 2097 [iot_thread] [INFO ][MQTT][20970] MQTT library cleanup done.
34 2097 [iot_thread] [INFO ][DEMO][20970] Cleaned up MQTT library.
I (21357) wifi:Deinit lldesc rx mblock:10
35 2097 [iot_thread] [INFO ][DEMO][20970] memory_metrics::freertos_heap::before::bytes::193252
36 2097 [iot_thread] [INFO ][DEMO][20970] memory_metrics::freertos_heap::after::bytes::97800
37 2097 [iot_thread] [INFO ][DEMO][20970] memory_metrics::demo_task_stack::before::bytes::8776
38 2097 [iot_thread] [INFO ][DEMO][20970] memory_metrics::demo_task_stack::after::bytes::2696
39 2097 [iot_thread] [INFO ][DEMO][20970] Demo completed successfully.
40 2108 [iot_thread] [INFO ][INIT][21080] SDK cleanup done.
41 2108 [iot_thread] [INFO ][DEMO][21080] -------DEMO FINISHED-------

**To reproduce**
i follow step by step in https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html
help me guys

Thank you!

yanjos-dev, 2020-08-14:
Hello @monhack, it looks like the demo is running correctly based on the log you posted. Were you able to see the expected messages in the AWS IoT MQTT client? You can find instructions on how to do this in the [AWS IoT Greengrass discovery demo application documentation](https://docs.aws.amazon.com/freertos/latest/userguide/gg-demo.html). 

For your reference, here is a log of the code running successfully with the 202007.00 release on the ESP32 board. 

```
rst:0x1 (POWERON_RESET),boot:0x3e (SPI_FAST_FLASH_BOOT)

flash read err, 1000

ets_main.c 371 

ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x3e (SPI_FAST_FLASH_BOOT)

configsip: 0, SPIWP:0xee

clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

mode:DIO, clock div:2

load:0x3fff0018,len:4

load:0x3fff001c,len:7176

load:0x40078000,len:12652

load:0x40080400,len:6708

entry 0x40080778

[0;32mI (74) boot: Chip Revision: 1[0m

[0;32mI (74) boot_comm: chip revision: 1, min. bootloader chip revision: 0[0m

[0;32mI (40) boot: ESP-IDF HEAD-HASH-NOTFOUND 2nd stage bootloader[0m

[0;32mI (40) boot: compile time 02:33:14[0m

[0;32mI (70) boot: Enabling RNG early entropy source...[0m

[0;32mI (70) boot: SPI Speed      : 40MHz[0m

[0;32mI (70) boot: SPI Mode       : DIO[0m

[0;32mI (72) boot: SPI Flash Size : 4MB[0m

[0;32mI (76) boot: Partition Table:[0m

[0;32mI (80) boot: ## Label            Usage          Type ST Offset   Length[0m

[0;32mI (87) boot:  0 nvs              WiFi data        01 02 00010000 00006000[0m

[0;32mI (95) boot:  1 otadata          OTA data         01 00 00016000 00002000[0m

[0;32mI (102) boot:  2 phy_init         RF data          01 01 00018000 00001000[0m

[0;32mI (110) boot:  3 ota_0            OTA app          00 10 00020000 00177000[0m

[0;32mI (117) boot:  4 ota_1            OTA app          00 11 001a0000 00177000[0m

[0;32mI (125) boot:  5 storage          WiFi data        01 02 00317000 00010000[0m

[0;32mI (133) boot: End of partition table[0m

[0;32mI (137) boot: No factory image, trying OTA 0[0m

[0;32mI (142) boot_comm: chip revision: 1, min. application chip revision: 0[0m

[0;32mI (149) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x27b80 (162688) map[0m
[... 214 lines cut ...]

33 2617 [iot_thread] Heap low watermark: 74192. Stack high watermark: 2616.

34 2617 [iot_thread] [INFO ][MQTT][26170] MQTT library cleanup done.

35 2617 [iot_thread] [INFO ][DEMO][26170] Cleaned up MQTT library.

36 2617 [iot_thread] [INFO ][DEMO][26170] memory_metrics::freertos_heap::before::bytes::193944

37 2617 [iot_thread] [INFO ][DEMO][26170] memory_metrics::freertos_heap::after::bytes::74192

38 2617 [iot_thread] [INFO ][DEMO][26170] memory_metrics::demo_task_stack::before::bytes::8920

39 2617 [iot_thread] [INFO ][DEMO][26170] memory_metrics::demo_task_stack::after::bytes::2616

40 2617 [iot_thread] [INFO ][DEMO][26170] Demo completed successfully.

GAP procedure initiated: stop advertising.

GAP procedure initiated: stop advertising.

41 2630 [iot_thread] [INFO ][INIT][26300] SDK cleanup done.

42 2630 [iot_thread] [INFO ][DEMO][26300] -------DEMO FINISHED-------
```

monhack, 2020-08-17:
i follow instruction in https://docs.aws.amazon.com/freertos/latest/userguide/gg-demo.html number 3
```
Open freertos/vendors/vendor/boards/board/aws_demos/config_files/aws_demo_config.h, comment out #define CONFIG_MQTT_DEMO_ENABLED, and define CONFIG_GREENGRASS_DISCOVERY_DEMO_ENABLED. 
```
thanks @yanjos-dev finally solve

closed 2020-08-17 by monhack