aws/amazon-freertos #2369: ESP32 crashes on mqtt_agent_connect

YTvW, 2020-08-17:
**Describe the bug**
when trying to connect to AWS IoT Core the esp32 has an unhandled exception when calling iotMqtt_connect

**System information**
- ESP32
- VSCode
- Operating System [MacOS]
- Project [Custom Application|MQTT]
- If your project is a Custom Application, please add the relevant code snippet in the section `Code to reproduce the bug`. 

**Expected behavior**
Expect to start the communications task using xTaskCreate() and have the task connect to AWS IoT Core using the credentials in clientcredintial_keys.h. (just for testing purposes)

**Screenshots or console output**
```
55 12123 [commsTask] [INFO ][MQTT][12123] Establishing new MQTT connection.

56 12127 [commsTask] [DEBUG][MQTT][12127] MQTT PINGREQ packet:

57 12133 [commsTask] c0 00 

58 12136 [commsTask] [DEBUG][MQTT][12135] (MQTT connection 0x3ffba6b4) Creating new operation record.

59 12144 [commsTask] [DEBUG][MQTT][12144] (MQTT connection 0x3ffba6b4) Reference count changed from 2 to 3.

60 12154 [commsTask] [DEBUG][MQTT][12154] MQTT CONNECT packet:

61 12160 [commsTask] 10 1d 00 04 4d 51 54 54 04 02 04 b0 00 11 79 77 

62 12166 [commsTask] 6b 2d 63 68 65 63 6b 6f 75 74 2d 30 30 30 31 

Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4008a3c6  PS      : 0x00060f30  A0      : 0x800886d9  A1      : 0x3ffd8c40  
A2      : 0x3ffc77c0  A3      : 0x3ffd527c  A4      : 0x00007530  A5      : 0x3ffb9a3c  
A6      : 0x3ffba6b4  A7      : 0x3ffd8ddc  A8      : 0x00000000  A9      : 0x3ffc77c8  
A10     : 0x00000000  A11     : 0x00000016  A12     : 0x00000000  A13     : 0x400d9d74  
A14     : 0x3ffba940  A15     : 0x3ffba6d0  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

ELF file SHA256: cff4ecab890f69f5

Backtrace: 0x4008a3c6:0x3ffd8c40 0x400886d6:0x3ffd8c60 0x4008a10b:0x3ffd8c80 0x401183f3:0x3ffd8cc0 0x4011840d:0x3ffd8ce0 0x40117ea1:0x3ffd8d00 0x40117ef1:0x3ffd8d20 0x400d9814:0x3ffd8d50 0x400d8386:0x3ffd8d80 0x400d7a92:0x3ffd8db0 0x400d6aab:0x3ffd8e50 0x400d6aec:0x3ffd8ea0

```

**Code to reproduce the bug**

```cpp
BaseType_t prvMQTTConnect()
{
    MQTTAgentReturnCode_t xReturned;
    BaseType_t xReturn = pdFAIL;
    MQTTAgentConnectParams_t xConnectParameters =
        {
            clientcredentialMQTT_BROKER_ENDPOINT,            /* The URL of the MQTT broker to connect to. */
            mqttconfigMQTT_AGENT_CONNECT_FLAGS,              /* Connection flags. */
            pdFALSE,                                         /* Deprecated. */
            clientcredentialMQTT_BROKER_PORT,                /* Port number on which the MQTT broker is listening. Can be overridden by ALPN connection flag. */
            (const uint8_t *)clientcredentialIOT_THING_NAME, /* Client Identifier of the MQTT client. It should be unique per broker. */
            0,                                               /* The length of the client Id, filled in later as not const. */
            pdFALSE,                                         /* Deprecated. */
            NULL,                                            /* User data supplied to the callback. Can be NULL. */
            NULL,                                            /* Callback used to report various events. Can be NULL. */
            NULL,                                            /* Certificate used for secure connection. Can be NULL. */
            0                                                /* Size of certificate used for secure connection. */
        };

    /* Check this function has not already been executed. */
    configASSERT(xMQTTHandle == NULL);

    /* The MQTT client object must be created before it can be used.  The
	* maximum number of MQTT client objects that can exist simultaneously
	* is set by mqttconfigMAX_BROKERS. */
    xReturned = MQTT_AGENT_Create(&xMQTTHandle);
    
    // configPRINTF(("result %s\n", xReturned));
    if (xReturned == eMQTTAgentSuccess)
    {
        /* Fill in the MQTTAgentConnectParams_t member that is not const,
		* and therefore could not be set in the initializer (where
		* xConnectParameters is declared in this function). */
        xConnectParameters.usClientIdLength = (uint16_t)strlen((const char *)clientcredentialIOT_THING_NAME);

        /* Connect to the broker. */
        configPRINTF(("MQTT attempting to connect to %s.\r\n", clientcredentialMQTT_BROKER_ENDPOINT));
        xReturned = MQTT_AGENT_Connect(xMQTTHandle, &xConnectParameters, mqttconfigMQTT_TLS_NEGOTIATION_TIMEOUT);
        configPRINTF(("mqttagent Connect call completed\n"));
        if (xReturned != eMQTTAgentSuccess)
        {
            /* Could not connect, so delete the MQTT client. */
            (void)MQTT_AGENT_Delete(xMQTTHandle);
            configPRINTF(("ERROR:  MQTT  failed to connect.\r\n"));
        }
        else
        {
            configPRINTF(("MQTT echo connected.\r\n"));
            xReturn = pdPASS;
        }
    }

    return xReturn;
}
```

**Additional context**
Add any other context about the problem here.

Thank you!

YTvW, 2020-08-17:
I have found the problem, besides calling MQTT_AGENT_Init() you also need to call IotSdk_Init() otherwise it will fail when trying to connect and cause an unhandled exception. Couldn't find this in the documentation so took me a while to figure out

yanjos-dev, 2020-08-18:
@YTvW Thank you for sharing the solution you found and for the feedback. This requirement can be found in the [iot_init.h header file](https://github.com/aws/amazon-freertos/blob/b11b43ae7685de35c50e70acf8eec39f3ab4f9ec/libraries/c_sdk/standard/common/include/iot_init.h#L44) where the IotSdk_Init function is declared. I see how this could be difficult to find, so I created a PR ( #2378 ) to make this requirement more clear in the demo.

YTvW, 2020-08-18:
Would it be an idea to initialize the iot ark when initializing the MQTT agent since that needs to be done anyway ?

aggarg, 2020-08-19:
This function creates resources needed by all the libraries and not just MQTT and that is why it is called explicitly in the application. 

As you mentioned that your issues is resolved, I am closing this. Feel free to re-open or create a new issue if the problem persists.

Thanks.

closed 2020-08-19 by aggarg