aws/amazon-freertos #2371: Implement speech to text on b-l475e-iot01a

git-cerbero, 2020-08-17:
Hi everyone.
i am struggling with a project on my board b-l475e-iot01a. The basic idea is to record, using the user button, the voice through the microphone and, once the registration is finished, send everything, through appropriate online api, to receive the translated text. I was able, through some online tutorials, to activate the microphone and to save the audio in a buffer for a few seconds.
Now I am faced with a fairly important problem: do you know any free online api provider where it is possible to send the audio and receive the translated text? Also, I have another question ... How can I convert the current format of the array (array of int32_t) into an audio file or other, so that I can adapt it to the possible requests of the online api providers?
I hope I was clear and thank you in advance for the time you will devote to me.

yanjos-dev, 2020-08-19:
Hello @git-cerbero, thank you for sharing your interesting use case. I'm personally unaware of a free online API for this kind of functionality. You may be interested in [Amazon Transcribe](https://aws.amazon.com/transcribe/). While it's not free, there is a trial period where you can try it out.

git-cerbero, 2020-08-19:
Hello yanjos-dev.
Thank you for the suggestion. Could you show me an example in c where amazon transcribe is used?

aggarg, 2020-08-19:
The following should help you get started with the Amazon Transcribe API: https://docs.aws.amazon.com/transcribe/latest/dg/getting-started-asc-api.html

The following blog post shows real-time streaming transcription with the AWS C++ SDK: https://aws.amazon.com/blogs/developer/real-time-streaming-transcription-with-the-aws-c-sdk/

Thanks.

aggarg, 2020-08-24:
I am closing this issue. Feel free to re-open or create a new issue if you need anything else. 

Thanks.

closed 2020-08-24 by aggarg