aws/amazon-freertos #2374: Disconnect callback is not triggered when MQTT connection is being made using BLE using network Manager

kumarfirmware, 2020-08-17:
Hi Team,

I am using AWS freeRTOS version 202007.

Disconnect callback is not triggered when MQTT connection is being made using BLE using network Manager.

Disconnect callback is triggered when MQTT connection is being made using Wi-Fi using network Manager.

static IotNetworkError_t prvNetworkDisconnectCallback( void * pvContext ); This function never gets triggered if MQTT connection is being made using BLE using network Manager.

Can you kindly let me know what can be the issue here?

How can I trigger this disconnect call back for MQTT connection using BLE using network Manager.

Thank you.

ravibhagavandas, 2020-08-19:
Hi,

`prvNetworkDisconnectCallback` is triggered when the response for an MQTT PING request is not received by the device from the broker. For BLE, the MQTT PING request is handled  by the IOT Mobile SDK and auto reconnect is enabled on the mobile app so this handler is unused. 
However, using network manager you can subscribe to a BLE disconnect at physical level using the following API in `iot_network_manager_private.h`:
```
BaseType_t AwsIotNetworkManager_SubscribeForStateChange( uint32_t ulNetworkTypes,
                                                         AwsIotNetworkStateChangeCb_t xCallback,
                                                         void * pvContext,
                                                         IotNetworkManagerSubscription_t * pxHandle );
```
where network type can be `AWSIOT_NETWORK_TYPE_BLE`

ravibhagavandas, 2020-08-26:
Hi,

 > But where should I do the stop OTA and reconnect MQTT for BLE disconnection?

`prvNetworkDisconnectCallback` will be invoked for a BLE disconnect just like WiFi network disconnect as the keep alive times out since a ping request won't be send through the BLE connection.

Are you thinking of handling network disconnect ( such as a WiFi access point disconnect or a BLE disconnect from central device) and to recover from this?  Network manager event callback can be used by subscribing to it as follows:
```c
status = AwsIotNetworkManager_SubscribeForStateChange( AWSIOT_NETWORK_TYPE_BLE | AWSIOT_NETWORK_TYPE_WIFI,
                                                         prvApplicationCallback,
                                                         NULL,
                                                         &xsubscriptionHandle);
```

In `prvApplicationCallback` you can add your business logic of handling a BLE disconnect or WiFi disconnect. There are other network manager helper functions which are useful  for this and documented [here](https://github.com/aws/amazon-freertos/blob/master/demos/network_manager/iot_network_manager_private.h) .

Let us know if you have further questions.

kumarfirmware, 2020-09-21:
This issue is resolved.

closed 2020-09-21 by kumarfirmware