aws/amazon-freertos #2376: Reduce heap usage for tls connection

TBorlan, 2020-08-18:
Hello. I'm using the mbed_tls module for establishing secure connections with the aws endpoint. I see that during

```
BaseType_t TLS_Connect( void * pvContext )
{
    BaseType_t xResult = 0;
    TLSContext_t * pxCtx = ( TLSContext_t * ) pvContext; /*lint !e9087 !e9079 Allow casting void* to other types. */

    /* Ensure that the FreeRTOS heap is used. */
    CRYPTO_ConfigureHeap();

    /* Initialize mbedTLS structures. */
    mbedtls_ssl_init( &pxCtx->xMbedSslCtx );
    mbedtls_ssl_config_init( &pxCtx->xMbedSslConfig );
    mbedtls_x509_crt_init( &pxCtx->xMbedX509CA );

    /* Decode the root certificate: either the default or the override. */
    if( NULL != pxCtx->pcServerCertificate )
    {
        xResult = mbedtls_x509_crt_parse( &pxCtx->xMbedX509CA,
                                          ( const unsigned char * ) pxCtx->pcServerCertificate,
                                          pxCtx->ulServerCertificateLength );

        if( 0 != xResult )
        {
            TLS_PRINT( ( "ERROR: Failed to parse custom server certificates %d \r\n", xResult ) );
        }
    }
    else
    {
        xResult = mbedtls_x509_crt_parse( &pxCtx->xMbedX509CA,
                                          ( const unsigned char * ) tlsVERISIGN_ROOT_CERTIFICATE_PEM,
                                          tlsVERISIGN_ROOT_CERTIFICATE_LENGTH );

        if( 0 == xResult )
        {
            xResult = mbedtls_x509_crt_parse( &pxCtx->xMbedX509CA,
                                              ( const unsigned char * ) tlsATS1_ROOT_CERTIFICATE_PEM,
                                              tlsATS1_ROOT_CERTIFICATE_LENGTH );

            if( 0 == xResult )
            {
                xResult = mbedtls_x509_crt_parse( &pxCtx->xMbedX509CA,
                                                  ( const unsigned char * ) tlsSTARFIELD_ROOT_CERTIFICATE_PEM,
                                                  tlsSTARFIELD_ROOT_CERTIFICATE_LENGTH );
            }
        }

        if( 0 != xResult )
        {
            /* Default root certificates should be in aws_default_root_certificate.h */
            TLS_PRINT( ( "ERROR: Failed to parse default server certificates %d \r\n", xResult ) );
        }
    }
```
If we supply the credential initializer with the server certificate like so 

```
#define AWS_IOT_NETWORK_CREDENTIALS_AFR_INITIALIZER            \
    {                                                          \
        .pAlpnProtos = socketsAWS_IOT_ALPN_MQTT,               \
        .maxFragmentLength = 0,                                \
        .disableSni = false,                                   \
        .pRootCa = tlsSTARFIELD_ROOT_CERTIFICATE_PEM,          \
        .rootCaSize = tlsSTARFIELD_ROOT_CERTIFICATE_LENGTH,    \
        .pClientCert = keyCLIENT_CERTIFICATE_PEM,              \
        .clientCertSize = sizeof( keyCLIENT_CERTIFICATE_PEM ), \
        .pPrivateKey = keyCLIENT_PRIVATE_KEY_PEM,              \
        .privateKeySize = sizeof( keyCLIENT_PRIVATE_KEY_PEM )  \
    }
```

The certificate chain is no longer parsed and stored in memory, which reduce heap usage by a lot. The downside is that i have to modify a core file in the amazon modules. 
Are there any other options which i can use to reduce heap usage for TLS connections?

yanjos-dev, 2020-08-18:
Hello @TBorlan. Have you considered creating an Mbed TLS configuration file that is optimized for your application? We provide a [default Mbed TLS configuration file](https://github.com/aws/amazon-freertos/blob/master/libraries/3rdparty/mbedtls_config/aws_mbedtls_config.h), but it's not optimized for specific boards and applications. This can be a good starting point for reducing the memory footprint of an application. See issue [1468 ](https://github.com/aws/amazon-freertos/issues/1468#issuecomment-571280296) and issue #1250 for examples of how to reduce the memory footprint of your application.

aggarg, 2020-08-24:
I am closing this issue. Feel free to re-open or create a new issue if you need anything else.

closed 2020-08-24 by aggarg