aws/amazon-freertos #2389: IOT_STATIC_MEMORY_ONLY and IOT_LOG_LEVEL_GLOBAL==IOT_LOG_DEBUG not working

lightblu, 2020-08-21:
**Describe the bug**

IOT_STATIC_MEMORY_ONLY and IOT_LOG_LEVEL_GLOBAL==IOT_LOG_DEBUG seem to cause some endless looping / crash (or something similar, endless looping is what I see in IotLog_Generic til its IotLogging_Malloc).

One problem seems to be that that IotMutex_Create (which is used in IotStaticMemory_Init) already wants to log a debug message, but logging itself with IOT_STATIC_MEMORY_ONLY requires that static memory to be initialized already.

It goes somehow like this:
IotSdk_Init -> IotStaticMemory_Init -> IotMutex_Create -> IotLogDebug( "Creating new mutex %p.", pNewMutex ); -> IotLog_Generic -> IotLogging_Malloc -> Iot_MallocMessageBuffer -> IotStaticMemory_FindFree$

Removing the IotLogDebug in IotMutex_Create gets me a little bit further, but there seem to be other problems because I don't get it running until forcing logging to not use static memory via e.g.:

```
#define IotLogging_Malloc                       pvPortMalloc
#define IotLogging_Free                         vPortFree
```

**Expected behavior**
IOT_STATIC_MEMORY_ONLY and IOT_LOG_LEVEL_GLOBAL==IOT_LOG_DEBUG should work.

**To reproduce**
Steps to reproduce the behavior:

1. Define IOT_STATIC_MEMORY_ONLY and IOT_LOG_LEVEL_GLOBAL==IOT_LOG_DEBUG
2.Call IotSdk_Init

muneebahmed10, 2020-09-07:
Hi @lightblu,

Sorry for the late response, I'm looking into this issue and will update you when I have more information on this. When you remove that debug log statement in `IotMutex_Create`, what additional problem did you encounter?

Thanks

muneebahmed10, 2020-09-09:
Hi @lightblu,

I'm unable to reproduce this issue with the MQTT demo on the 202007.00 release, on the Windows simulator. I set `IOT_STATIC_MEMORY_ONLY` to `1`, and all log levels to `IOT_LOG_DEBUG` in `iot_config.h`. What platform/commit/demo are you using in which you encounter this issue? Also, I see that `IotLogging_Malloc` and `IotLogging_Free` are defined in [`iot_config_common.h`](https://github.com/aws/amazon-freertos/blob/master/demos/include/iot_config_common.h#L95) and are [not affected](https://github.com/aws/amazon-freertos/blob/master/demos/include/iot_config_common.h#L89) by the definition of `IOT_STATIC_MEMORY_ONLY`. Is there anything you are doing differently?

Thanks

lightblu, 2020-09-11:
On bc86e638723a7542fa688d4624fd9c9b453ab664 (Thu Jun 6 15:34:51 2019 -0700), will consider a bump soon then. Environment is our own platform (ARM M4) (the IotLogging_Malloc/Free defines may have recently changed due to another request.

We can close this til I get to bumping version and reattempting? Sorry.

muneebahmed10, 2020-09-12:
Hi @lightblu,

I will close this issue then. Please open a new issue if you have further questions.

Thank you

closed 2020-09-12 by muneebahmed10