aws/amazon-freertos #2400: Using FreeRTOS timer in UART ISR results in crash

kumarfirmware, 2020-08-24:
Hi Team,

I am using AWS freeRTOS version 202007 and ESP32.

I am using UART interrupt handler to raise interrupt, when I receive data on UART.

In the UART interrupt handler, I am calling xTimerStart(timer_handle, 0 );

In the initalisation I did xTimerCreate( pcTimerName,
                                            pdMS_TO_TICKS( 30 * 1000 ),
                                            pdFALSE,
                                            NULL,
                                            Timer_Callback );

As soon as the data is received and xTimerStart is called, it crashes. The following is the error I am receiving.

core=0 port_interruptNesting=1

/opt/amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/port.c:315 (vPortAssertIfInISR)- assert failed!
abort() was called at PC 0x4008e3df on core 0

ELF file SHA256: c91d6659ccfed3e4

Backtrace: 0x4008d034:0x3ffbe210 0x4008d265:0x3ffbe230 0x4008e3df:0x3ffbe250 0x400900aa:0x3ffbe270 0x40090597:0x3ffbe290 0x400d342d:0x3ffbe2b0 0x40080f19:0x3ffbe2d0 0x4008177e:0x3ffbe2f0 0x4018aa03:0x00000000

But if I use ESP32 timer in UART ISR it works fine and it does not crash.

So this concludes, we cannot use FreeRTOS timers in UART ISR? 

Can you please confirm this as soon as possible? Am I doing any thing wrong here? Is this the expected behaviour?

Thank you.

n9wxu, 2020-09-01:
Many FreeRTOS API's cannot be called from interrupt and the xTimerStart API is no exception.  However, in most cases, an ISR friendly version of the API function is available.  In this case, the xTimerStartFromISR can be called from the interrupt.  The calling parameters are slightly different so you cannot simply add 'FromISR' to the function call.
For more information consult the API documentation here:
[](https://www.freertos.org/FreeRTOS-timers-xTimerStartFromISR.html)

muneebahmed10, 2020-09-09:
Hi @kumarfirmware,

Did the [above](https://github.com/aws/amazon-freertos/issues/2400#issuecomment-685115178) response resolve your issue? We will close this issue if we don't hear back from you.

Thanks

kumarfirmware, 2020-09-21:
Please close the issue.

kumarfirmware, 2020-09-21:
Issue is resolved.

closed 2020-09-21 by kumarfirmware