aws/amazon-freertos #2401: AWS Greegrass cannot connect to esp32 (Amazon freeRTOS)

monhack, 2020-08-25:
**Describe the bug**
I've done all the instructions from https://docs.amazonaws.cn/en_us/greengrass/latest/developerguide/gg-gs.html and https://docs.aws.amazon.com/freertos/latest/userguide/freertos-getting-started.html, but I do not get a tutorial on how freeRTOS can connect, send, and receive a message to raspberry pi (Greengrass) with the local network. I am finding no way out from here

**System information**
- ESP32 WROOM32D
- Visual Studio Code
- Operating System : Linux
- Version of Amazon FreeRTOS : 202007.00 release
- Project : FreeRTOS|MQTTGreengrass|Sub pub Demo

**Expected behavior**
Amazon freeRTOS devices can connect, send, and receive messages to and from an IoT Greengrass Core device for local processing at the edge.

**Additional context**
I hope you can provide a good tutorial so I can follow it well

Thank you!

tgsong, 2020-09-01:
Hi, sorry for a late response. Have you check the FreeRTOS Greengrass demo doc? https://docs.aws.amazon.com/freertos/latest/userguide/gg-demo.html

monhack, 2020-09-07:
thanks @tgsong

monhack, 2020-09-07:
thanks @tgsong 
I have done all the step by step. the problem is while flashing to esp32, /demos/greengrass_discovery/aws_greengrass_discovery_demo.c don't work. But this file is executed (iot_demo_mqtt.c). How to cmake with the specified files? i'm so confused

muneebahmed10, 2020-09-07:
Hi @monhack,

According to step 3 of the manual configuration section ("Configuring the FreeRTOS demos") in the above [guide](https://docs.aws.amazon.com/freertos/latest/userguide/gg-demo.html), to run the Greengrass demo, you must remove the definition of [`CONFIG_MQTT_DEMO_ENABLED`](https://github.com/aws/amazon-freertos/blob/master/vendors/espressif/boards/esp32/aws_demos/config_files/aws_demo_config.h#L46) and instead define `CONFIG_GREENGRASS_DISCOVERY_DEMO_ENABLED` in [aws_demo_config.h](https://github.com/aws/amazon-freertos/blob/master/vendors/espressif/boards/esp32/aws_demos/config_files/aws_demo_config.h). Does the mqtt demo still run after doing this step?

monhack, 2020-09-08:
Thanks @muneebahmed10,

i have done follow step by step, but esp32(FreeRTOS) doesn't send messages to greengrass core. this is my output

```
2 6 [iot_thread] [INFO ][DEMO][60] ---------STARTING DEMO---------

I (387) BTDM_INIT: BT controller compile version [0fff744]
3 6 [iot_thread] [INFO ][INIT][60] SDK successfully initialized.
I (387) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (497) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0
GAP procedure initiated: advertise; disc_mode=2 adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=187 adv_itvl_max=375
I (797) wifi:wifi driver task: 3ffdb50c, prio:23, stack:3584, core=0
I (797) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (797) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (817) wifi:wifi firmware version: 1afdae0
I (817) wifi:config NVS flash: enabled
I (817) wifi:config nano formating: disabled
I (817) wifi:Init dynamic tx buffer num: 32
I (817) wifi:Init data frame dynamic rx buffer num: 32
I (827) wifi:Init management frame dynamic rx buffer num: 32
I (837) wifi:Init management short buffer num: 32
I (837) wifi:Init static rx buffer size: 1600
I (837) wifi:Init static rx buffer num: 10
I (847) wifi:Init dynamic rx buffer num: 32
I (857) wifi:mode : sta (30:ae:a4:97:6b:b0)
I (857) WIFI: SYSTEM_EVENT_STA_START
I (3517) wifi:new:<11,0>, old:<1,0>, ap:<255,255>, sta:<11,0>, prof:1
I (4497) wifi:state: init -> auth (b0)
I (4507) wifi:state: auth -> assoc (0)
I (4507) wifi:state: assoc -> run (10)
I (4527) wifi:connected with #After21, aid = 1, channel 11, BW20, bssid = a6:d9:90:80:83:0b
I (4527) wifi:security type: 3, phy: bgn, rssi: -20
I (4527) wifi:pm start, type: 1

I (4537) WIFI: SYSTEM_EVENT_STA_CONNECTED
I (4537) wifi:AP's beacon interval = 102400 us, DTIM period = 2
I (6467) event: sta ip: 192.168.43.36, mask: 255.255.255.0, gw: 192.168.43.1
I (6467) WIFI: SYSTEM_EVENT_STA_GOT_IP
GAP procedure initiated: stop advertising.
4 614 [iot_thread] [INFO ][DEMO][6140] Successfully initialized the demo. Network type for the demo: 1
5 615 [iot_thread] [INFO ][MQTT][6150] MQTT library successfully initialized.
6 615 [iot_thread] [INFO ][DEMO][6150] Attempting automated selection of Greengrass device

GAP procedure initiated: stop advertising.
E (6567) PKCS11: failed nvs get file size 4354 0
7 1200 [iot_thread] JSON file retrieval completed
8 1200 [iot_thread] About to close socket.
9 1201 [iot_thread] Socket closed.
10 1201 [iot_thread] Stack high watermark for discovery helper task: 5096.
E (12647) PKCS11: failed nvs get file size 4354 0
11 1484 [iot_thread] About to close socket.
12 1485 [iot_thread] Socket closed.
13 1485 [iot_thread] Stack high watermark for discovery helper task: 2760.
14 1485 [iot_thread] [INFO ][DEMO][14850] Greengrass device discovered.
15 1485 [iot_thread] [INFO ][DEMO][14850] Attempting to establish MQTT connection to Greengrass.
E (15207) PKCS11: failed nvs get file size 4354 0
16 1744 [iot_thread] [INFO ][MQTT][17440] Establishing new MQTT connection.
17 1745 [iot_thread] [INFO ][MQTT][17450] Anonymous metrics (SDK language, SDK version) will be provided to AWS IoT. Recompile with AWS_IOT_MQTT_ENABLE_METRICS set to 0 to disable.
18 1745 [iot_thread] [INFO ][MQTT][17450] (MQTT connection 0x3ffb60d8, CONNECT operation 0x3ffb5eb4) Waiting for operation completion.
19 1747 [iot_thread] [INFO ][MQTT][17470] (MQTT connection 0x3ffb60d8, CONNECT operation 0x3ffb5eb4) Wait complete with result SUCCESS.
20 1747 [iot_thread] [INFO ][MQTT][17470] New MQTT connection 0x3ffb4a50 established.
21 1747 [iot_thread] [INFO ][MQTT][17470] (MQTT connection 0x3ffb60d8) MQTT PUBLISH operation queued.
22 1897 [iot_thread] [INFO ][MQTT][18970] (MQTT connection 0x3ffb60d8) MQTT PUBLISH operation queued.
23 2047 [iot_thread] [INFO ][MQTT][20470] (MQTT connection 0x3ffb60d8) MQTT PUBLISH operation queued.
24 2197 [iot_thread] [INFO ][DEMO][21970] Disconnecting from broker.
25 2197 [iot_thread] [INFO ][MQTT][21970] (MQTT connection 0x3ffb60d8) Disconnecting connection.
26 2197 [iot_thread] [INFO ][MQTT][21970] (MQTT connection 0x3ffb60d8, DISCONNECT operation 0x3ffb5eb4) Waiting for operation completion.
27 2198 [iot_thread] [INFO ][MQTT][21980] (MQTT connection 0x3ffb60d8, DISCONNECT operation 0x3ffb5eb4) Wait complete with result SUCCESS.
28 2198 [iot_thread] [INFO ][MQTT][21980] (MQTT connection 0x3ffb60d8) Connection disconnected.
[... 1 lines cut ...]
30 2202 [iot_thread] [INFO ][MQTT][22020] (MQTT connection 0x3ffb60d8) Network connection destroyed.
I (22347) wifi:state: run -> init (0)
I (22347) wifi:pm stop, total sleep time: 14838113 us / 17822585 us

I (22357) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (22367) WIFI: SYSTEM_EVENT_STA_DISCONNECTED: 8
E (22367) wifi:wifi deinit internal: wifi not stop
31 2202 [iot_thread] [INFO ][DEMO][22020] Disconnected from the broker.
32 2202 [iot_thread] Heap low watermark: 99120. Stack high watermark: 2760.
33 2202 [iot_thread] [INFO ][MQTT][22020] MQTT library cleanup done.
34 2202 [iot_thread] [INFO ][DEMO][22020] Cleaned up MQTT library.
35 2202 [iot_thread] [INFO ][DEMO][22020] memory_metrics::freertos_heap::before::bytes::193256
I (22407) wifi:flush txq
I (22407) wifi:stop sw txq
I (22417) wifi:lmac stop hw txq
36 2202 [iot_thread] [INFO ][DEMO][22020] memory_metrics::freertos_heap::after::bytes::99120
I (22427) wifi:Deinit lldesc rx mblock:10
37 2202 [iot_thread] [INFO ][DEMO][22020] memory_metrics::demo_task_stack::before::bytes::8920
38 2202 [iot_thread] [INFO ][DEMO][22020] memory_metrics::demo_task_stack::after::bytes::2760
39 2202 [iot_thread] [INFO ][DEMO][22020] Demo completed successfully.
40 2214 [iot_thread] [INFO ][INIT][22140] SDK cleanup done.
41 2214 [iot_thread] [INFO ][DEMO][22140] -------DEMO FINISHED-------

I (142367) event: station ip lost
```

muneebahmed10, 2020-09-08:
Hi @monhack,

In your output, I see the strings `New MQTT connection 0x3ffb4a50 established` and `Demo completed successfully.`, indicating the Greengrass device was discovered and an MQTT connection was formed with it. Are you subscribing to the topic `freertos/demos/ggd` from the IoT Console to view the messages? Also, have you created a subscription (IoT Core -> Greengrass -> Groups -> *Your Greengrass Group* -> Subscriptions) between your ESP32 Things as the Source and "IoT Cloud" as the Target, on the topic `freertos/demos/ggd`?

monhack, 2020-09-08:
Thanks, @muneebahmed10,

hmm, I forget to subscription between esp32 and IoT Core, finally solved.

now how to publish a message from esp32-freertos (```aws_greengrass_discovery_demo.c```) to raspi (Greengrass Core) without the internet (local network)? I not found a tutorial in the documentation. Please give me a tutorial about it.

Thanks, bro

muneebahmed10, 2020-09-09:
Hi @monhack,

Since the Greengrass discovery demo relies on the [Greengrass Discovery API endpoint](https://docs.aws.amazon.com/greengrass/latest/developerguide/gg-discover-api.html), we don't support a way to perform local discovery for Greengrass in our demos. The Greengrass discovery demo uses the discovery endpoint to obtain the connection information for the Greengrass Core, and then forms [an MQTT connection](https://github.com/aws/amazon-freertos/blob/master/demos/greengrass_connectivity/aws_greengrass_discovery_demo.c#L189) to it just as it would for IoT Core. So, if you have the local connection information for your Greengrass Core (including certificate, port, and local address), you can try skipping the discovery step and connect directly to the core over your local network.

monhack, 2020-09-10:
Thanks @muneebahmed10,

So, do you have a solution for freertos so that it can send data to greengrass core without internet? or how freertos get local connection information about Greengrass Core? give me some tutorial about this problem.

monhack, 2020-09-12:
Hi @muneebahmed10,

i have done report this problem to AWS Support. Finally solved!

Thanks a lot! @muneebahmed10 @tgsong

closed 2020-09-12 by monhack