aws/amazon-freertos #2402: ESP32 GPIO HAL

YTvW, 2020-08-25:
**Is your feature request related to a problem? Please describe.**  
Using any GPIO function is introducing a hard dependency on the esp specific libraries

**Describe the solution you would like.**  
A GPIO implementation to use with the iot_gpio hardware abstraction 

**Additional context**  
being able to switch between hardware without having to rewrite parts of the software that uses GPIO (most software does) would be very nice 

Thank you!

mingyue86010, 2020-09-10:
Hello,

Thank you for submitting the request, this feature is implemented by this PR https://github.com/aws/amazon-freertos/pull/1946 which will be merged soon.

Thank you!
Regards,

Ming

YTvW, 2020-09-10:
For the most part, except gpio. That PR contains common io for i2c, spi and uart, but not for gpio

mingyue86010, 2020-09-14:
Hi Yrjan,

Sorry I was not check that PR carefully. You are correct. I will make a feature request to the team and follow up with this.