aws/amazon-freertos #2404: [iot_thread] [ERROR][MQTT][14340] Failed to establish new MQTT connection, error TIMEOUT.

SGOSAI602, 2020-08-25:
**Describe the bug**
I am trying to run the MQTT demo on ESP32. I followed this tutorial : https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html
Its configuration is done and build files are successfully generated. But when i run this command to flash and run FreeRTOS : " ./vendors/espressif/esp-idf/tools/idf.py flash monitor -p /dev/ttyUSB0" on serial monitor it shows "Starting Demo" and gets error at

****12 4238 [iot_thread] [INFO ][MQTT][42380] Network connection closed on error.
13 4239 [iot_thread] [INFO ][MQTT][42390] (MQTT connection 0x3ffdf948) Network connection destroyed.
I (42437) wifi:state: run -> init (0)
I (42447) wifi:pm stop, total sleep time: 9958099 us / 13014256 us**

**I (42447) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1****
. I cannot figure out what kind of error it could be.
There's is another
**E (33017) PKCS11: failed nvs get file size 4354 0**

**System information**
-  Hardware - ESP32-WROOM-32 D
- Operating System Linux-Ubuntu
- Version of FreeRTOS (run ``git describe --tags`` to find it)
- Project AWS FreeRTOS MQTT demo

**Console Output**
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:7204
load:0x40078000,len:12652
load:0x40080400,len:6708
entry 0x40080778
I (72) boot: Chip Revision: 1
I (72) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (38) boot: ESP-IDF v3.3.2-235-ga3ce2f400 2nd stage bootloader
I (38) boot: compile time 21:28:40
I (39) boot: Enabling RNG early entropy source...
I (44) boot: SPI Speed      : 40MHz
I (48) boot: SPI Mode       : DIO
I (52) boot: SPI Flash Size : 4MB
I (56) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00010000 00006000
I (75) boot:  1 otadata          OTA data         01 00 00016000 00002000
I (82) boot:  2 phy_init         RF data          01 01 00018000 00001000
I (90) boot:  3 ota_0            OTA app          00 10 00020000 00177000
I (97) boot:  4 ota_1            OTA app          00 11 001a0000 00177000
I (105) boot:  5 storage          WiFi data        01 02 00317000 00010000
I (112) boot: End of partition table
I (117) boot: No factory image, trying OTA 0
I (121) boot_comm: chip revision: 1, min. application chip revision: 0
I (129) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x2bb88 (179080) map
I (200) esp_image: segment 1: paddr=0x0004bbb0 vaddr=0x3ffbdb60 size=0x03848 ( 14408) load
I (206) esp_image: segment 2: paddr=0x0004f400 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/aumrajshital/aws_esp32/amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/xtensa_vectors.S:1713

I (208) esp_image: segment 3: paddr=0x0004f808 vaddr=0x40080400 size=0x00808 (  2056) load
I (217) esp_image: segment 4: paddr=0x00050018 vaddr=0x400d0018 size=0xb76fc (751356) map
0x400d0018: _flash_cache_start at ??:?

I (489) esp_image: segment 5: paddr=0x0010771c vaddr=0x40080c08 size=0x1a8cc (108748) load
I (550) boot: Loaded app from partition at offset 0x20000
I (585) boot: Set actual ota_seq=1 in otadata[0]
I (585) boot: Disabling RNG early entropy source...
I (585) cpu_start: Pro cpu up.
I (588) cpu_start: Application information:
I (593) cpu_start: Project name:     esp-idf
I (598) cpu_start: App version:      202007.00-124-g60f21a68d
I (604) cpu_start: Compile time:     Aug 22 2020 21:27:36
I (610) cpu_start: ELF file SHA256:  4f4efd404e8a7cea...
[... 65 lines cut ...]
12 4238 [iot_thread] [INFO ][MQTT][42380] Network connection closed on error.
13 4239 [iot_thread] [INFO ][MQTT][42390] (MQTT connection 0x3ffdf948) Network connection destroyed.
I (42437) wifi:state: run -> init (0)
I (42447) wifi:pm stop, total sleep time: 9958099 us / 13014256 us

I (42447) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (42457) WIFI: SYSTEM_EVENT_STA_DISCONNECTED: 8
E (42467) wifi:wifi deinit internal: wifi not stop
14 4239 [iot_thread] [ERROR][DEMO][42390] MQTT CONNECT returned error TIMEOUT.
15 4239 [iot_thread] [INFO ][MQTT][42390] MQTT library cleanup done.
16 4239 [iot_thread] [INFO ][DEMO][42390] memory_metrics::freertos_heap::before::bytes::200488
I (42487) wifi:flush txq
I (42487) wifi:stop sw txq
I (42497) wifi:lmac stop hw txq
17 4239 [iot_thread] [INFO ][DEMO][42390] memory_metrics::freertos_heap::after::bytes::76880
I (42507) wifi:Deinit lldesc rx mblock:10
18 4240 [iot_thread] [INFO ][DEMO][42400] memory_metrics::demo_task_stack::before::bytes::5848
19 4240 [iot_thread] [INFO ][DEMO][42400] memory_metrics::demo_task_stack::after::bytes::1896
20 4240 [iot_thread] [ERROR][DEMO][42400] Error running demo.
GAP procedure initiated: stop advertising.
GAP procedure initiated: stop advertising.
21 4251 [iot_thread] [INFO ][INIT][42510] SDK cleanup done.
22 4251 [iot_thread] [INFO ][DEMO][42510] -------DEMO FINISHED-------

Thank you!

tgsong, 2020-09-02:
Hi, sorry for a late response. It is likely that your AWS IoT thing and certificate are not configured correctly. Could you provide information on how you set up the IoT thing? Did you create a certificate from console? Is it activated and attached to the thing? Are they being correctly configured in the source code?

sarenameas, 2020-09-17:
Hello @SGOSAI602 ,

We have not header back from you yet. I am going to close issue for now. Please reopen if you are still having issues.

closed 2020-09-17 by sarenameas

LeonardMichalas, 2020-09-28:
@tgsong @sarenameas I have got the same issue as well. 

I have set up the configuration as follows:

clientcredentialMQTT_BROKER_ENDPOINT = My custom MQTT Endpoint (AWS IoT > Settings)

clientcredentialIOT_THING_NAME = Thing ARN (AWS IoT > Things > [Device Name] > Thing ARN)

clientcredentialMQTT_BROKER_PORT = 8883

clientcredentialGREENGRASS_DISCOVERY_PORT  = 8443

keyCLIENT_CERTIFICATE_PEM = device certificate, which I downloaded when I created the thing in AWS IoT

keyJITR_DEVICE_CERTIFICATE_AUTHORITY_PEM = Amazon Root CA 1 (https://docs.aws.amazon.com/iot/latest/developerguide/server-authentication.html?icmpid=docs_iot_console#server-authentication-certs)

keyCLIENT_PRIVATE_KEY_PEM  = private key, which I downloaded when I created the thing in AWS IoT 

Any support would be highly appreciated.