aws/amazon-freertos #2406: AWS Greengrass using elliptic curves

vvelciu, 2020-08-26:
Hello,

Is there any guide on how to use the Greengrass discovery demo application with EC-based certificates?

Thanks!

muneebahmed10, 2020-09-02:
HI @vvelciu,

AWS IoT supports several [certificate signing algorithms](https://docs.aws.amazon.com/iot/latest/developerguide/x509-client-certs.html#x509-cert-algorithms) for X.509 certs, including ECDSA. Have you tried [creating your own client certificates](https://docs.aws.amazon.com/iot/latest/developerguide/device-certs-your-own.html), and then using them for the [Greengrass Discovery demo](https://docs.aws.amazon.com/freertos/latest/userguide/gg-demo.html)?

muneebahmed10, 2020-09-12:
Hi @vvelciu,

Did the above response resolve your issue? We will close this issue if we don't hear back from you.

closed 2020-09-14 by muneebahmed10

vvelciu, 2021-02-22:
Hello @muneebahmed10 ,

I managed to run the Greengrass Discovery demo with ECDSA (I generated the keys for the device and the Greengrass core as described [here,](https://aws.amazon.com/it/blogs/iot/elliptic-curve-cryptography-and-forward-secrecy-support-in-aws-iot-3/) but I needed to make some changes in the _iot_tls.c_ file. The default root certificates did not include the ATS3 one, so I manually added it:
[image]
Is there any proper way of doing this? Greengrass is hardcoded to using default certificates:
[image]

Thanks,
Veronica