aws/amazon-freertos #2422: Getting started with MQTT and amazon-FreeRTOS on ESP32

KressAlex, 2020-09-02:
Hi everybody, 

I am kind of new to all the FreeRTOS chapter and are struggling a lot with all the MQTT communication.

I got all the Amazon-FreeRTOS demos working for MQTT, and also can modify it to subscribe to a different topic or change buffer counts.

But now my problem is, that it is impossible for me to modify the demo to make a simple MQTT communication and publish or subscribe to a topic. I did not even find out how I can publish/subscribe to two different topics, linked with the correct payload.

I was reading a lot through all the community out there trying to find help for my issue and noticed that I´m not alone with my problems. Many new guys out there struggling the same than me, reading through 844 lines of code without understanding anything.

I would really appreciate if somebody could do a really simple MQTT explanation/example. 
**Let´s say 3 simple examples in the garden managed by ESP32:**
     1. Switching the watering valve On/Off and read status of valve, publish and subscribe to topic (On/Off)
     2. Read an analog water tank level, publish to topic
     3. Pass preset value received from topic to a local variable, subscribe to topic

And the most important would be, to start from an empty page with MQTT and explain the complete flow in the example what has to be done in order to get one message successfully published or subscribed to a topic.

**Starting from scratch and walk through all the steps that have to be done and how.**

1. Get local values that have to be published.
2. create payload
3. link it to the correct topic
4. connect to broker
5. publish MQTT package
6. subscribe to MQTT packages
7. save received values into local variables
8. cleanup MQTT connection

I would be really happy, if somebody could make a youtube video out of it and explain the steps he is doing. 
Or if it can´t be a video maybe a straight forward explanation in between the steps, in normal words for everybody understandable.

So at the end everybody would be able to make a use out of the MQTT and Amazon-FreeRTOS  at home to switch on/off a lamp for example.

Thank you guys a lot, I really appreciate your help in advance!!!

Thanks, Alex

muneebahmed10, 2020-09-03:
Hello @KressAlex,

I'm sorry to hear you had difficulty getting started with the MQTT library. We appreciate your feedback and request for more detailed documentation, and we will consider updating it.

Is there any part of the [FreeRTOS documentation](https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html) specifically that you found confusing or needs clarification? Is the [MQTT API Reference](https://docs.aws.amazon.com/freertos/latest/lib-ref/c-sdk/mqtt/mqtt_design.html) any more helpful in answering your questions?

KressAlex, 2020-09-06:
Hello @muneebahmed10,

thanks for your quick answer!

I don´t know how to publish a customized message, lets say for example i want to press a button and than it should publish a message which includes some data for the weather.

So I press the button, than I want to publish to topic "Home/CurrentWeather" and the status as payload "Rain"
And during a temp change, on another topic I want to publish the Temperature "Home/CurrentTemp", payload "19 °C"

So I expected something like a function to pass the data: 
 -    PublishToTopic("Home/CurrentWeather", "19 °C");
or   d PublishToTopic("Home/CurrentTemp", "Rain");

something like this for example where I can pass my parameters to the Publish function:

typedef struct IotMqttPublishInfo
{
    IotMqttQos_t qos;         /**< @brief QoS of message. Must be 0 or 1. */
    bool retain;              /**< @brief MQTT message retain flag. */

    const char * pTopicName;  /**< @brief Topic name of PUBLISH. */
    uint16_t topicNameLength; /**< @brief Length of #IotMqttPublishInfo_t.pTopicName. */

    const void * pPayload;    /**< @brief Payload of PUBLISH. */
    size_t payloadLength;     /**< @brief Length of #IotMqttPublishInfo_t.pPayload. For LWT messages, this is limited to 65535. */

    uint32_t retryMs;         /**< @brief If no response is received within this time, the message is retransmitted. */
    uint32_t retryLimit;      /**< @brief How many times to attempt retransmission. */
} IotMqttPublishInfo_t;

but I don´t understand how to forward all the information into the iot_demo_mqtt.c, because in the demo it´s always the preset Hello World, and than its just numbers counting up and down in the function.

I know that I also have to Initialize the MQTT connection first and than also cleanup.

Thanks a lot @muneebahmed10 !!!

monhack, 2020-09-07:
hello @KressAlex,
maybe you can modify freertos/demos/greengrass_connectivity/aws_greengrass_discovery_demo.c. the program explain how to pub/sub JSON message and you can import library sensors.

I give you link about example amazon-freertos-esp32 using dht11 from krazers (https://github.com/krazers/freertos-dht11)

keep spirit bro!

KressAlex, 2020-09-07:
Thanks a lot @monhack!

I will take a look into this and will keep you posted, if I got a step closer or not. 

Thanks! 
Regards Alex

muneebahmed10, 2020-09-09:
Hi @KressAlex,

In the MQTT demo function [`_publishAllMessages`](https://github.com/aws/amazon-freertos/blob/master/demos/mqtt/iot_demo_mqtt.c#L611), you can see where the [topic name](https://github.com/aws/amazon-freertos/blob/master/demos/mqtt/iot_demo_mqtt.c#L650) and [payload](https://github.com/aws/amazon-freertos/blob/master/demos/mqtt/iot_demo_mqtt.c#L629) are assigned in the publish info struct before calling the [MQTT Publish function](https://github.com/aws/amazon-freertos/blob/master/demos/mqtt/iot_demo_mqtt.c#L675). You can write your own function using the MQTT API to suit your needs, e.g.

```c
void PublishToTopic( IotMqttConnection_t mqttConnection, const char * topic, const char * payload )
{
    IotMqttError_t publishStatus = IOT_MQTT_STATUS_PENDING;
    IotMqttPublishInfo_t publishInfo = IOT_MQTT_PUBLISH_INFO_INITIALIZER;
    IotMqttCallbackInfo_t publishComplete = IOT_MQTT_CALLBACK_INFO_INITIALIZER;

    publishInfo.qos = IOT_MQTT_QOS_1;
    publishInfo.pTopicName = topic;
    publishInfo.topicNameLength = strlen( topic );
    publishInfo.pPayload = payload;
    publishInfo.payloadLength = strlen( payload );
    publishInfo.retryMs = PUBLISH_RETRY_MS;
    publishInfo.retryLimit = PUBLISH_RETRY_LIMIT;

    //Assume someCallbackFunction is your callback function defined elsewhere.
    //It will be called when the publish completes.
    publishComplete.function = someCallbackFunction;
    //The context should identify the publish that was completed, e.g. the topic (if unique).
    publishComplete.pCallbackContext = ( void * ) topic;

    publishStatus = IotMqtt_Publish( mqttConnection, &publishInfo, 0, &publishComplete, NULL );
}
```

KressAlex, 2020-09-15:
Hi @muneebahmed10, 

I'm still struggling with the MQTT part. 

Do you think that it would be possible, that somebody does an example code that could be included in a custom code, just to send the basic MQTT and call the correct steps that has to be done and also close the connection in a correct manner. 

And recall that code based on an event, for an example a button is pressed or a value has changed!? 

The MQTT example has so much content, that might don't even had to be changed for simple applications and is tightened into so many other things. That would be also like a future request to tell the guys that just get started with the MQTT communication. 

Or maybe you can send me a PN. 

Thanks so much, regards Alex

sarenameas, 2020-09-16:
Hello @KressAlex,

Our documentation relies on on the customer understanding how the MQTT protocol works on a high-level and also an understanding of the C language. 

Unfortunately we do not currently have a blog or sample for an application using a mock or real sensor and I can definitely put into our future planning considerations if this can be created. We have found that interns just starting on the team can quickly create applications to send sensor data to AWS IoT to view using our current demos. It sounds like you are looking for a single topic example that publishes only for you to write over a few lines for your needs? The current demo has multiple topics and I do think you can easily piece together how to do this simply for one topic. 

The current iot_mqtt_demo.c in this repo subscribes to the same topics it is publishing to that that in a single thread we can see that data being submitted is also received. 

[RunMqttDemo()](https://github.com/aws/amazon-freertos/blob/master/demos/mqtt/iot_demo_mqtt.c#L741) - Shows the high level steps of:
1. [connect](https://github.com/aws/amazon-freertos/blob/master/demos/mqtt/iot_demo_mqtt.c#L778)
1. [subscribe](https://github.com/aws/amazon-freertos/blob/master/demos/mqtt/iot_demo_mqtt.c#L792)
1. [publish](https://github.com/aws/amazon-freertos/blob/master/demos/mqtt/iot_demo_mqtt.c#L806)
1. [unsubscribe](https://github.com/aws/amazon-freertos/blob/master/demos/mqtt/iot_demo_mqtt.c#L823)
1. [disconnect](https://github.com/aws/amazon-freertos/blob/master/demos/mqtt/iot_demo_mqtt.c#L832)

Depending on your needs, you may want to publish only and can skip the subscribe examples. 

In RunMqttDemo() as you may already know you can modify your subscriptions here:

```C
    /* Topics used as both topic filters and topic names in this demo. */
    const char * pTopics[ TOPIC_FILTER_COUNT ] =
    {
        IOT_DEMO_MQTT_TOPIC_PREFIX "/topic/1",
        IOT_DEMO_MQTT_TOPIC_PREFIX "/topic/2",
        IOT_DEMO_MQTT_TOPIC_PREFIX "/topic/3",
        IOT_DEMO_MQTT_TOPIC_PREFIX "/topic/4",
    };
```

After connecting to the broker, the next step is to subscribe to these topics.  In _modifySubscriptions() you should be able to find an example of the needed parameters for IotMqtt_TimedSubscribe() being filled out.
```
    for( i = 0; i < TOPIC_FILTER_COUNT; i++ )
    {
        pSubscriptions[ i ].qos = IOT_MQTT_QOS_1;
        pSubscriptions[ i ].pTopicFilter = pTopicFilters[ i ];
        pSubscriptions[ i ].topicFilterLength = TOPIC_FILTER_LENGTH;
        pSubscriptions[ i ].callback.pCallbackContext = pCallbackParameter;
        pSubscriptions[ i ].callback.function = _mqttSubscriptionCallback;
    }

```
This logic to use IotMqtt_TimedSubscribe() under the `if( operation == IOT_MQTT_SUBSCRIBE )` part.

The next thing you need to do is publish. _publishAllMessages() shows how to fill out the parameters needed for IotMqtt_Publish(). From @muneebahmed10's last post, he shows you how to simplify the function a bit more for whatever your needs may be. 

The next steps in the demo is to unsubscribe to the topics it subscribed just before disconnecting.  The logic in _modifySubscriptions() for unsubscribing is shown under the `else if( operation == IOT_MQTT_UNSUBSCRIBE )` part. 

The disconnection with the server is shown in a correct manner with the example code:
```C
    /* Disconnect the MQTT connection if it was established. */
    if( connectionEstablished == true )
    {
        IotMqtt_Disconnect( mqttConnection, 0 );
    }

    /* Clean up libraries if they were initialized. */
    if( librariesInitialized == true )
    {
        _cleanupDemo();
    }
```

KressAlex, 2020-09-17:
hi @sarenameas,

Thank you very much on your feedback, I will work on the flow you provided and I hope to get success!

Thanks very much to all of you!

best regards Alex

sarenameas, 2020-09-17:
@KressAlex 

And thank you for choosing us! I am going to close this issue now. Feel free to open a new one if you run into any other issues.

Thanks,
Sarena

closed 2020-09-17 by sarenameas