aws/amazon-freertos #2426: Where can a client monitor an OTA Job progress?

rrivadeneira-rsc, 2020-09-04:
**Describe the bug**
I was trying 

**System information**
- Which hardware board or part numbers? - ESP32
- IDE used - Eclipse
- Operating System - MacOS
- Version of FreeRTOS (run ``git describe --tags`` to find it) - modified version from `202002.00`
- Project - Custom Appkication
- If your project is a Custom Application, please add the relevant code snippet in the section `Code to reproduce the bug`. 

**Expected behavior**
The OTA agent in Freertos provides us with a list of different topics used to accomplish OTA Jobs such as:
```
static const char pcOTA_JobsGetNextAccepted_TopicTemplate[] = "$aws/things/%s/jobs/$next/get/accepted";
static const char pcOTA_JobsNotifyNext_TopicTemplate[] = "$aws/things/%s/jobs/notify-next";
static const char pcOTA_JobsGetNext_TopicTemplate[] = "$aws/things/%s/jobs/$next/get";
static const char pcOTA_JobStatus_TopicTemplate[] = "$aws/things/%s/jobs/%s/update";
static const char pcOTA_StreamData_TopicTemplate[] = "$aws/things/%s/streams/%s/data/cbor";
static const char pcOTA_GetStream_TopicTemplate[] = "$aws/things/%s/streams/%s/get/cbor";
```
If a client or someone would like to see the status updates the firmware gives to the Job Status topic, they should be able to do that by subscribing to `$aws/things/<thingName>/jobs/<jobID>/update` or even possibly subscribing to `$aws/things/#`. However, subscribing on the console or another MQTT client with full access to the broker does not get any visibility or messages from that particular topic. I am expecting messages like this to come through the status update topic:
```
{"status":"IN_PROGRESS","statusDetails":{"receive":"320/360"}}
```
and
```
{"status":"FAILED","statusDetails":{"reason":"rejected: 0x22000000"}}
```

I can only see two initial messages when the firmware receives the job at `$aws/things/%s/jobs/$next/get/accepted` and `$aws/things/%s/jobs/notify-next` posting the the general information about the job in question.

Can a client not see the messages coming through those topics? Should they be looking through a different topic? Are there special permission AWS puts to these reserved topics under the hood? How can I client get visibility of an OTA Job Update in real time?

Basically, the use case would be that we would like the client to be aware of an OTA Job's failure (with reason) or success for managing update purposes.

**Screenshots or console output**
At the time of updating the job with IN PROGRESS every X blocks:
```
1269 52215 [OTA Agent Task] [prvUpdateJobStatus_Mqtt] Msg: {"status":"IN_PROGRESS","statusDetails":{"receive":"320/360"}}
1270 52216 [OTA Agent Task] [INFO ][MQTT][522160] (MQTT connection 0x3fff5120) MQTT PUBLISH operation queued.
1271 52216 [OTA Agent Task] [prvUpdateJobStatus_Mqtt] 'IN_PROGRESS' to $aws/things/BCDDC2CB4C88/jobs/AFR_OTA-v-0-1-8-OTA-Update/update
```
At the time of updating the job to FAILURE:
```
138 1459 [OTA Agent Task] [prvUpdateJobStatus_Mqtt] Msg: {"status":"FAILED","statusDetails":{"reason":"rejected: 0x22000000"}}
139 1460 [OTA Agent Task] [INFO ][MQTT][14600] (MQTT connection 0x3fff25c4) MQTT PUBLISH operation queued.
140 1461 [OTA Agent Task] [INFO ][MQTT][14610] (MQTT connection 0x3fff25c4, PUBLISH operation 0x3fff4dfc) Waiting for operation completion.
141 1467 [OTA Agent Task] [INFO ][MQTT][14660] (MQTT connection 0x3fff25c4, PUBLISH operation 0x3fff4dfc) Wait complete with result SUCCESS.
142 1467 [OTA Agent Task] [prvUpdateJobStatus_Mqtt] 'FAILED' to $aws/things/BCDDC2CB4C88/jobs/AFR_OTA-v-0-1-8-OTA-Update/update
143 1468 [OTA Agent Task] [prvResetDevice] Attempting forced reset of device...
```

**To reproduce**
Steps to reproduce the behavior:
1. Enable OTA Agent
2. Subscribe through the console or any MQTT client to all Job reserved topics
3. There will be no output coming from the subscription to $aws/things/<thingName>/jobs/<jobID>/update`

**Code to reproduce the bug**
N/A - We are using our custom application, but I can assume that the OTA demo could serve for this purpose.

**Additional context**
Add any other context about the problem here.

Thank you!

tgsong, 2020-09-04:
Hi, please check the AWS Jobs API doc here, https://docs.aws.amazon.com/iot/latest/developerguide/jobs-api.html#mqtt-updatejobexecution. Basically the topic `$aws/things/<thingName>/jobs/<jobID>/update` is for sending the update to the service. To listening for a response, you can subscribe to 2 additional topics,

- `$aws/things/thingName/jobs/jobId/update/accepted`
- `$aws/things/thingName/jobs/jobId/update/rejected`

Upon success, you will be notified in the first topic. Note to get the execution state, you will need to set `includeJobExecutionState` to true when posting update to `$aws/things/<thingName>/jobs/<jobID>/update`, this requires a code change in our current OTA code. An example response looks like this,

```
{
  "clientToken": "123",
  "timestamp": 1599258272,
  "executionState": {
    "status": "IN_PROGRESS",
    "statusDetails": {
      "progress": "10/300"
    },
    "versionNumber": 2
  }
}
```

rrivadeneira-rsc, 2020-09-08:
Hi @tgsong - thank you for your prompt answer! 

I can't seem to find the `includeJobExecutionState` flag in the OTA files I've looked (aws_iot_ota_agent_config.h, aws_iot_ota_agent.h, aws_iot_ota_agent_internal.h, aws_iot_ota_mqtt.c). Would you please give me some guidance on where to find this configuration to enable job execution state to test?

Thank you so much in advance!

tgsong, 2020-09-08:
Yes, that key is not used in our OTA library code, you need to add it. The status update message is constructed in this function, https://github.com/aws/amazon-freertos/blob/4d8b6543b9bd8c4a6be938804a9ec1c497911e2b/libraries/freertos_plus/aws/ota/src/mqtt/aws_iot_ota_mqtt.c#L426
Basically you just need to modify the `pcOTA_JobStatus_ReceiveDetailsTemplate` string (defined at [here](https://github.com/aws/amazon-freertos/blob/4d8b6543b9bd8c4a6be938804a9ec1c497911e2b/libraries/freertos_plus/aws/ota/src/mqtt/aws_iot_ota_mqtt.c#L74)) to include the `includeJobExecutionState` key.

yanjos-dev, 2020-10-23:
Hi @rrivadeneira-rsc, I'm closing this issue for now. Please feel free to reopen this if you have any further questions.

closed 2020-10-23 by yanjos-dev