aws/amazon-freertos #2431: Subscribtion TIMEOUT for topics starting with $aws

AbdeenM, 2020-09-06:
**Describe the bug**
Am sure this is not a bug but rather a silly mistake from my end.
It seems that whenever i try to subscribe to `$aws/certificates/create/json` i get an error with timeout, i am trying to setup the provisioning by claim and everything is going well upto this point.
So i have done a few tests to narrow down my findings, below should should help:
1. I setup the provisioning policy and got the certs as per the guide [here](https://aws.amazon.com/blogs/iot/how-to-automate-onboarding-of-iot-devices-to-aws-iot-core-at-scale-with-fleet-provisioning/).
2. I coded a quick python script to check if the template and AWS server side is setup correctly, all is well and i get the official certs and token.
3. To make sure that my ESP code is correct i added a new topic and topicfilter to my provisioning policy and ran the python script for the topic, as expected it subscribed and was able to push to the test topic, i then changed the topic of subscription on rtos to the test and my device ran perfectly, connected, subscribed and pushed.
4. So know i tried changing the topic to the original endpoint `$aws/certificates/create/json` but again TIMEOUT is what i get. if i change it to anything else say `test/9` it works fine.

**System information**
- Device: ESP32
- IDE: VSCode
- OS: Windows
- Version of FreeRTOS: 201912.00-396-g314275376
- Project: Custom Application

**Expected behavior**
When trying to subscribe to a topic filter of `$aws/certificates/create/json` the device subscribes, same would apply to making a mqtt push.

**Screenshots or console output**
```Bash
GAP procedure initiated: stop advertising.
5 513 [iot_thread] [INFO ][MQTT][5130] MQTT library successfully initialized.
6 513 [iot_thread] [INFO ][Shadow][5130] Shadow library successfully initialized.
7 513 [iot_thread] [INFO ][DEMO][5130] Thing Name is PROV5130 (length 8).
E (5726) PKCS11: failed nvs get file size 4354 0
8 1042 [iot_thread] [INFO ][MQTT][10420] Establishing new MQTT connection.
9 1042 [iot_thread] [INFO ][MQTT][10420] (MQTT connection 0x3ffb34f4, CONNECT operation 0x3ffb3618) Waiting for operation completion.
I (10736) AWS PROV => : Filter Length = 29
10 1072 [iot_thread] [INFO ][MQTT][10720] (MQTT connection 0x3ffb34f4, CONNECT operation 0x3ffb3618) Wait complete with result SUCCESS.
11 1072 [iot_thread] [INFO ][MQTT][10720] New MQTT connection 0x3ffb5038 established.
12 1072 [iot_thread] [INFO ][MQTT][10720] (MQTT connection 0x3ffb34f4) SUBSCRIBE operation scheduled.
13 1072 [iot_thread] [INFO ][MQTT][10720] (MQTT connection 0x3ffb34f4, SUBSCRIBE operation 0x3ffb3618) Waiting for operation completion.
14 1572 [iot_thread] [INFO ][MQTT][15720] (MQTT connection 0x3ffb34f4, SUBSCRIBE operation 0x3ffb3618) Wait complete with result TIMEOUT.
15 1572 [iot_thread] [INFO ][DEMO][15720] Subscribtion topic timed out.
16 1572 [iot_thread] [INFO ][MQTT][15720] (MQTT connection 0x3ffb34f4) MQTT PUBLISH operation queued.
17 1573 [iot_thread] [ERROR][NET][15730] Error -27648 while sending data.
18 1573 [iot_thread] [ERROR][DEMO][15730] MQTT PUBLISH 1 could not be sent. Error NETWORK ERROR.
19 7073 [iot_thread] [ERROR][NET][70730] Error -27648 while sending data.
20 7073 [iot_thread] [ERROR][MQTT][70730] (MQTT connection 0x3ffb34f4) Failed to send PINGREQ.
21 7073 [iot_thread] [INFO ][MQTT][70730] (MQTT connection 0x3ffb34f4) Network connection closed.
```

**To reproduce**
Steps to reproduce the behavior:

You can run the original MQTT Demo but modify the demo main.c, adding the code below to handle subscription or making a mqtt push 

**Code to reproduce the bug**
The code is longer than 30 lines so please review [here](https://gist.github.com/AbdeenM/2beef1dc524717069239a6e7259ee526).

Thank you!

muneebahmed10, 2020-09-07:
Hello @AbdeenM,

I notice the publish operation after the subscribe fails with a network error. If you remove the subscribe, does the publish succeed? If so, then it's possible the server could be closing the network connection due to unexpected data in the subscribe packet. You could try adding a log statements before the subscribe operation, to make sure the topic filter is correct, as well as enabling debug logs for MQTT, so you can view the logs for the subscribe operation.

Second, are you sure you need to subscribe to `$aws/certificates/create/json`? In the AWS blog you referred to, as well as the Python reference client linked in it, there didn't seem to be step for subscribing to that topic, only publishing to it. The Python client looked like it only subscribed to the topic with a `/rejected` suffix. As `$aws` topics are reserved by the broker, it's possible the broker will send the messages without the client actually subscribing to the topic. If the broker isn't expecting a subscription on one of these topics, it could result in the errors you're seeing.

Also, have you looked at the beta [Fleet Provisioning library](https://github.com/aws/aws-iot-device-sdk-embedded-C/tree/FleetProvisioning_beta) in the v4_beta branch of the [Embedded C SDK](https://github.com/aws/aws-iot-device-sdk-embedded-C/tree/v4_beta/libraries/aws/provisioning)? This seems to be what you're trying to do, and might be helpful for reference.

AbdeenM, 2020-09-07:
Hello @muneebahmed10 !

Thank you for replying.
Yes correct, the push operation always yields a network error, if i remove the subscribe it still fails with the network error.
Okay i just tried what you suggested, i logged the data before subscription and push operations to check if the topic string is correct and it was correct for the subscription but wrong for the push. i fixed it but still got the network error for push operation.
I should mention i dont think this is the issue as i actually had them statically defined before my gist (as in `publishInfo.pTopicName = "$aws/certificates/create/%s";` and `pSubscriptions.pTopicFilter = "$aws/certificates/create/%s";` and i was still getting these TIMEOUT and NETWORK ERROR respectively.
Could you guide me on how to enable the debugging MQTT logs you mentioned, thank you.

No i don't need to subscribe to `$aws/certificates/create/json` this was just for testing. i only need to make the push operation to get the token and certs. Originally i didn't have the subscribe code you see in the gist and i was getting the network error a few steps down the line of debugging and i started testing for subscription as well.
My python script is different from the one on the blog (see below). am using a different library `paho` and it works as expected with the suffix of `$aws`

No i haven't seen the library you shared thank you, i will review the code base and maybe find something interesting, I will update with a comment after.

### Python Code
```Python
import sys
import ssl
import paho.mqtt.client as mqtt
import json
import time

def on_connect(mqtt, obj, flags, rc):
    if rc == 0:
        print('Subscriber Connection Status Code: ' + str(rc) + ' | Connection Status: Successful')
    elif rc == 1:
        print('Subscriber Connection Status COde: ' + str(rc) + ' | Connection Status: Refused')

def on_subscribe(mqtt, obj, mid, granted_qos):
    print('Subscribed: ' + str(mid) + ' ' + str(granted_qos) + 'data' + str(obj))

def on_message(mqtt, obj, msg):
    print('Recieved Message From Topic: ' + msg.topic + ' | QOS: ' + str(msg.qos) + ' | Data Received: ' + str(msg.payload))

mqttc = mqtt.Client(client_id='ProvTest')

mqttc.on_connect = on_connect
mqttc.on_message = on_message
mqttc.on_subscribe = on_subscribe

mqttc.tls_set('./AmazonRootCA1.crt',
certfile='./648f37da4a-certificate.pem.crt',
keyfile='./648f37da4a-private.pem.key',
tls_version=ssl.PROTOCOL_TLSv1_2,
ciphers=None)

mqttc.connect('a214zveo184gfp-ats.iot.us-east-1.amazonaws.com', port=8883)

time.sleep(5)

mqttc.loop_start()

# Not needed, was testing...
mqttc.subscribe('$aws/certificates/create/json')

mqttc.publish('$aws/certificates/create/json', '{}', 0)

while 1==1:
	time.sleep(0.5)
```

AbdeenM, 2020-09-07:
Hello Again @muneebahmed10 !

I managed to getting working, the push operation is successful! am attaching the new code [here](https://gist.github.com/AbdeenM/c512287d664549316af5cbe392d066c2).
Only thing is am not sure how to handle the response to the push operation, how can i handle the response and get the payload?

Thanks again!

aggarw13, 2020-09-09:
Hi @AbdeenM, I am glad to know that you were able to resolve the PUBLISH operation issue!

The response from the `$aws/certificates/create/json` topic comes on either of the `$aws/certificates/create/json/rejected` OR ``$aws/certificates/create/json/accepted` topics. Both of these topics can be subscribed to by the application along with a common subscription handler OR separate subscription callback for each topic to process the incoming response from the server. 

( As mentioned earlier by @muneebahmed10), you can refer to the [Fleet Provisioning library](https://github.com/aws/aws-iot-device-sdk-embedded-C/tree/v4_beta_deprecated/libraries/aws/provisioning) in the `v4_beta_deprecated` branch of [Embedded C SDK](https://github.com/aws/aws-iot-device-sdk-embedded-C/tree/v4_beta_deprecated) repository on how the response is handled. The library registers a common subscription callback for both the "rejected" and "accepted" response topic filters, and then parses the topic when invoked for an incoming server response on either of those topics. (You can refer to the implementation of the `AwsIotProvisioning_CreateKeysAndCertificate` API function [here](https://github.com/aws/aws-iot-device-sdk-embedded-C/blob/v4_beta_deprecated/libraries/aws/provisioning/src/aws_iot_provisioning_api.c#L531) for example on handling the server response using callbacks. **Note**: As this is a library API, it does takes the "handler" from the caller, and calls it within the subscription callbacks it registers with the MQTT library for the "accepted" and "rejected" topics. )

If the server response is on the "accepted" topic, then it would contain the new credentials (device cert and device private key in PEM format), certificate ID and ownership token (reference to the API doc: https://docs.aws.amazon.com/iot/latest/developerguide/fleet-provision-api.html#create-keys-cert). In your case, he subscription callback/handler can write the credentials to Non-volatile storage in your device.  If the server response is on the "rejected", the subscription callback can notify the application about the request failure.

AbdeenM, 2020-09-10:
Thank you so much @aggarw13 it is all well now.
I thought the payload was suppose to be handled within the publish operation itself, rather than from subscribing to the accept and reject.
Closing now and thanks again.

closed 2020-09-10 by AbdeenM