aws/amazon-freertos #2433: Increase payload of wifi send_data

git-cerbero, 2020-09-08:
Hi everyone,
I have a board b-l475e-iot01a.
I am developing a project in which I implement a web server. mainly the web server, when it receives the request from the browser, inserts the html code of the page in the uint8_t http[4096] variable.

So, I call the WIFI_SendData function below.

`WIFIReturnCode_t WIFI_SendData(uint8_t socket, uint8_t *pdata, uint16_t Reqlen, uint16_t *SentDatalen, uint32_t Timeout)
{
	WIFIReturnCode_t ret = eWiFiFailure;

    if(ES_WIFI_SendData(&xWiFiModule.xWifiObject, socket, pdata, Reqlen, SentDatalen, Timeout) == ES_WIFI_STATUS_OK)
    {
      ret = eWiFiSuccess;
    }

  return ret;
}`

The error that occurs is that when the html page is larger than 1200 characters, it is not sent correctly.
In the function ES_WIFI_Status_t ES_WIFI_SendData (ES_WIFIObject_t * Obj, uint8_t Socket, uint8_t * pdata, uint16_t Reqlen, uint16_t * SentLen, uint32_t Timeout); present in es_wifi.c I noticed the presence of an if that changes the length of the html variable to a predefined value (ES_WIFI_PAYLOAD_SIZE). So I tried to change this constant, but the problem still occurs around the size of 1500.

I don't know if I have to go and change some other parameter or if there is some other error.
Apparently, there seems to be a timeout problem .. is it possible?
Thank you in advance.

git-cerbero, 2020-09-08:
@aggarg you helped me with the wifi problem. would you be able to help me?

aggarg, 2020-09-08:
Looking at the code, `ES_WIFI_PAYLOAD_SIZE` determines the maximum length of the data that can be sent by `ES_WIFI_SendData`. What is the size of the data you want to send? If it is variable and cannot be determined at compile time, can you try sending in chunks (with maximum size of each chunk being `ES_WIFI_PAYLOAD_SIZE`)?

Thanks.

git-cerbero, 2020-09-08:
Hi, thank you for answering me.
I've thought about it too ... but how could I do?
The size of the data is variable and below is an example of the content of the http variable to be sent during the request.

`void index_page(uint8_t http[]) //0 success, 1 failure
{
  strcpy((char *)http, (char *)"HTTP/1.0 200 OK\r\nContent-Type: text/html\r\nPragma: no-cache\r\n\r\n");
  strcat((char *)http, (char *)"<html>");
  strcat((char *)http, (char *)"<head>");`
  .......

Thanks.

aggarg, 2020-09-08:
The following pseudo code shows a way to send data in chunks:

```
/* http is the buffer which contains the data to send. */
/* httpDataLength is the length of the data to be sent. */

WIFIReturnCode_t ret;
uint32_t dataLengthToSend, dataLengthSent;

while( httpDataLength > 0 )
{
    if( httpDataLength > ES_WIFI_PAYLOAD_SIZE)
    {
        dataLengthToSend = ES_WIFI_PAYLOAD_SIZE;
    }
    else
    {
        dataLengthToSend = httpDataLength;
    }

    ret = WIFI_SendData( socket, http, dataLengthToSend, &( dataLengthSent ), timeout );
    
    if( ret != eWiFiSuccess)
    {
        /* Handle failure (probably print a log). */
        break;
    }
    else
    {
        /* Update what is left to send based on the length of the data actually sent. */
        httpDataLength = httpDataLength - dataLengthSent;
    }
}
```

Thanks.

git-cerbero, 2020-09-08:
Hello. I really thank you very much. I managed to break up the payload and it works fine.
I just ask you one last help: I ​​am afraid that after exceeding a certain size of the http variable, the board is unable to satisfy in terms of memory, so I have noticed that in the last cells of the array there are values ​​at random, or repeated without an apparent reason. What parameters should I adjust to have more headroom regarding the size of the arrays?
I have tried to change the constant 

#define mainECHO_DEMO_STACK_SIZE (8192) //default 1024

but apparently I have not had any improvement ... and the array I need is made up like this:

uint8_t http[3000];

I thank you once again

aggarg, 2020-09-08:
Glad that it worked for you. 

Would you please elaborate what you mean by the following:
```
I have noticed that in the last cells of the array there are values ​​at random, or repeated without an apparent reason. 
```
Do you mean that you get more data at the receiving end than you send? 

Is this a global variable `uint8_t http[3000];`? If so, are there any concurrent accesses? 

Thanks.

git-cerbero, 2020-09-09:
Hi, I'm sorry if I made myself wrong.
The web server consists of a thread initialized as follows:

xTaskCreate( wifi_server,          /* The function that implements the task. */
			"SERVER",                   /* Just a text name for the task to aid debugging. */
			mainECHO_DEMO_STACK_SIZE, /* The stack size. */
			NULL,                     /* The task parameter, not used in this case. */
			mainECHO_DEMO_PRIORITY,   /* The priority assigned to the task. */
			NULL );                   /* The task handle is not used. */

In the wifi_server function, I proceed with starting the web server and when a client connects, I send the web page through the function below.

int SendWebPage_Index() //0 success, 1 failure
{
  uint8_t http[3000];

  index_page((uint8_t *)http);

  return SendWebPage((uint8_t *)http, strlen((char *)http));
}

The index_page () function does nothing but build the html page (as I posted earlier). The SendWebPage function does exactly what you sent me to solve the chunks problem.
Apparently, it is as if the http variable is too large for the thread stack. As you can see, http is not global and is created and destroyed every time the request is made by the client. How do you advise me to do to solve the problem?

Thanks for your patience

aggarg, 2020-09-09:
There are two possible issues:
* First, you are creating a very large buffer `http` on the stack. You may be causing stack overflow - what is the value of `mainECHO_DEMO_STACK_SIZE`?
* Second, does the `index_page` function ensures that a trailing NULL character is placed in the `http` buffer so that `strlen` works correctly?

I have the following two suggestions:
* You should make the `http` buffer static so that it is not created on the stack. Note that this will make the function non-reentrant and you wont be able to call it from multiple tasks (without adding some synchronization primitives to prevent concurrent access to the `http` buffer). If you are not calling this function from more than one task, it should be okay.
* Update your index function to return the number of bytes written to the `http` buffer. This way you will not need to call `strlen`.

```c
int SendWebPage_Index() //0 success, 1 failure
{
  static uint8_t http[3000];
  uint32_t httpLength;

  httpLength = index_page((uint8_t *)http);
  
  return SendWebPage((uint8_t *)http, httpLength);
}
```

Thanks.

aggarg, 2020-09-11:
Hi @git-cerbero, Did the above work for you?

Thanks.

git-cerbero, 2020-09-13:
Hi @aggarg , 
sorry if I answer you only now.
I managed to solve the problem by increasing the value of mainECHO_DEMO_STACK_SIZE and simultaneously reducing the size of the http array.
Basically, I split the index_page function into two functions (index_page1 and index_page2), which write to the http array one after the other. In this way, I was able to use a small array and at the same time send very complex and large pages.

I thank you for your patience with me.

closed 2020-09-13 by git-cerbero