aws/amazon-freertos #2434: AFR MQTT Demo doesn't display the Payload on the Serial console

dipen-1337lab, 2020-09-08:
**Describe the bug**
When running the MQTT Demo downloaded from the AWS IoT portal, the **payload message** does not get displayed on the serial console.

**System information**
- Hardware board: ESP32-DevKitC
- IDE used: N/A
- Operating System [Windows 10 x64 -- Version: 1909 -- OS Build: 18363.959]
- Version of FreeRTOS: AFR Version 202002.00 downloaded from [AWS IoT portal](https://console.aws.amazon.com/) (about 4 months ago).
- Project: MQTT Demo

**Expected behavior**
The possible console log should be as below.
```
Incoming PUBLISH received:
Subscription topic filter: iotdemo/topic/1
Publish topic name: iotdemo/topic/1
Publish retain flag: 0
Publish QoS: 1
Publish payload: <<payload message>>
```

**Screenshots or console output**
Intermittent console outputs are as below. It is observed that the **Publish QoS: 1** statement is also missing at times. The complete demo log can be downloaded here.
[mqtt_demo_console_log.txt](https://github.com/aws/amazon-freertos/files/5188518/mqtt_demo_console_log.txt)

```
25 1669 [iot_thread] [INFO ][DEMO][16690] Incoming PUBLISH received:
Subscription topic filter: iotdemo/topic/1
Publish topic name: iotdemo/topic/1
Publish retain flag: 0
Publish QoS: 1
...
...

28 1672 [iot_thread] [INFO ][DEMO][16720] Incoming PUBLISH received:
Subscription topic filter: iotdemo/topic/2
Publish topic name: iotdemo/topic/2
Publish retain flag: 0
Publish QoS: 1
...
...

103 1915 [iot_thread] [INFO ][DEMO][19150] Incoming PUBLISH received:
Subscription topic filter: iotdemo/topic/1
Publish topic name: iotdemo/topic/1
Publish retain flag: 0
...
...

106 1918 [iot_thread] [INFO ][DEMO][19180] Incoming PUBLISH received:
Subscription topic filter: iotdemo/topic/2
Publish topic name: iotdemo/topic/2
Publish retain flag: 0
...
...

```

**To reproduce**
Steps to reproduce the behaviour:

1. Login and download FreeRTOS from the FreeRTOS console.
2. Create a configuration for a ESP32-DevKitC. On the Configure FreeRTOS Software configuration properties page, under Libraries, choose MQTT Demo. Under Demo projects, choose to enable the MQTT Hello World demo.
3. Download and unzip to extract the folder contents.
4. Open Command Prompt with Administrator privilege.
5. Change the current working directory path to ..\FreeRTOS\
6. Connect the target hardware (ESP32 based DevKitC in my case).
7. Run the following commands. Observe the MQTT demo output on the console.

- cmake -DVENDOR=espressif -DBOARD=esp32_devkitc -DCOMPILER=xtensa-esp32 -DDEBUG=1 -GNinja -S . -B build
- cd build
- ninja
- .\vendors\espressif\esp-idf\tools\idf.py erase_flash flash monitor -B build

**Code to reproduce the bug**
N/A

**Additional context**
1) Below is the code for the information that gets displayed on the serial console. But **Payload** doesn't get displayed on the console. Occasionally, **QoS** is also missing.

```
static void _mqttSubscriptionCallback( void * param1,
                                       IotMqttCallbackParam_t * const pPublish )
{
    int acknowledgementLength = 0;
    size_t messageNumberIndex = 0, messageNumberLength = 1;
    IotSemaphore_t * pPublishesReceived = ( IotSemaphore_t * ) param1;
    const char * pPayload = pPublish->u.message.info.pPayload;
    char pAcknowledgementMessage[ ACKNOWLEDGEMENT_MESSAGE_BUFFER_LENGTH ] = { 0 };
    IotMqttPublishInfo_t acknowledgementInfo = IOT_MQTT_PUBLISH_INFO_INITIALIZER;

    /* Print information about the incoming PUBLISH message. */
    IotLogInfo( "Incoming PUBLISH received:\r\n"
                "Subscription topic filter: %.*s\r\n"
                "Publish topic name: %.*s\r\n"
                "Publish retain flag: %d\r\n"
                "Publish QoS: %d\r\n"
                "Publish payload: %.*s",
                pPublish->u.message.topicFilterLength,
                pPublish->u.message.pTopicFilter,
                pPublish->u.message.info.topicNameLength,
                pPublish->u.message.info.pTopicName,
                pPublish->u.message.info.retain,
                pPublish->u.message.info.qos,
                pPublish->u.message.info.payloadLength,
                pPayload );

```

2) After code modification for a custom subscribed topic, the callback function does get called but I am unable to view the payload.

3) On the AWS IoT portal, what is the format of the message, so that it is correctly decoded by the firmware? Is the below default message, on the AWS IoT portal for Publishing a message, a valid format? If yes, should the payload on the serial console be displayed as **Hello from AWS IoT console** once successfully received by the firmware?

```
{
  "message": "Hello from AWS IoT console"
}

```

Thank you!

AbdeenM, 2020-09-08:
Same issue

muneebahmed10, 2020-09-09:
Hi @dipen-1337lab,

Can you try splitting the log message into multiple calls, and see if the issue is still present? E.g.
```
    IotLogInfo( "Incoming PUBLISH received:\r\n"
                "Subscription topic filter: %.*s\r\n"
                "Publish topic name: %.*s\r\n"
                "Publish retain flag: %d\r\n"
                "Publish QoS: %d\r\n"
                "Publish payload: %.*s",
                pPublish->u.message.topicFilterLength,
                pPublish->u.message.pTopicFilter,
                pPublish->u.message.info.topicNameLength,
                pPublish->u.message.info.pTopicName,
                pPublish->u.message.info.retain,
                pPublish->u.message.info.qos,
                pPublish->u.message.info.payloadLength,
                pPayload );
```
to
```
    IotLogInfo( "Incoming PUBLISH received:\r\n"
                "Subscription topic filter: %.*s\r\n"
                "Publish topic name: %.*s\r\n"
                "Publish retain flag: %d\r\n",
                pPublish->u.message.topicFilterLength,
                pPublish->u.message.pTopicFilter,
                pPublish->u.message.info.topicNameLength,
                pPublish->u.message.info.pTopicName,
                pPublish->u.message.info.retain );

    IotLogInfo( "Publish QoS: %d\r\n"
                "Publish payload: %.*s",
                pPublish->u.message.info.qos,
                pPublish->u.message.info.payloadLength,
                pPayload );
```
Additionally, with the unchanged logging call, can you try increasing the value of the [`configLOGGING_MAX_MESSAGE_LENGTH`](https://github.com/aws/amazon-freertos/blob/master/vendors/espressif/boards/esp32/aws_demos/config_files/FreeRTOSConfig.h#L152) macro to a larger value to see if the entire log message is printed?

Thanks

dipen-1337lab, 2020-09-10:
Namaste @muneebahmed10,

Thank you for your reply on this issue. My observations are provided below.

> Can you try splitting the log message into multiple calls, and see if the issue is still present?

I am able to view the payload message after splitting the log message into multiple calls. Payload message is displayed correctly as **Publish payload: Hello world n!** where n varies from 0 to 19. Partial console log is shared below.

```
24 1005 [iot_thread] [INFO ][DEMO][10050] Incoming PUBLISH received:
Subscription topic filter: iotdemo/topic/1
Publish topic name: iotdemo/topic/1
Publish retain flag: 0

25 1005 [iot_thread] [INFO ][DEMO][10050] Publish QoS: 1
Publish payload: Hello world 0!
26 1007 [iot_thread] [INFO ][MQTT][10070] (MQTT connection 0x3ffdebf0) MQTT PUBLISH operation queued.
27 1007 [iot_thread] [INFO ][DEMO][10070] Acknowledgment message for PUBLISH 0 will be sent.
28 1009 [iot_thread] [INFO ][DEMO][10090] MQTT PUBLISH 1 successfully sent.
29 1011 [iot_thread] [INFO ][DEMO][10110] Incoming PUBLISH received:
Subscription topic filter: iotdemo/topic/2
Publish topic name: iotdemo/topic/2
Publish retain flag: 0

30 1011 [iot_thread] [INFO ][DEMO][10110] Publish QoS: 1
Publish payload: Hello world 1!

```

> With the unchanged logging call, can you try increasing the value of the configLOGGING_MAX_MESSAGE_LENGTH macro to a larger value to see if the entire log message is printed?

After modifying the below maximum length from 192 to 255, I am able to view the payload message as desired. Partial console log is shared below.

```
/* Sets the length of the buffers into which logging messages are written - so
 * also defines the maximum length of each log message. */
#define configLOGGING_MAX_MESSAGE_LENGTH            255

140 1797 [iot_thread] [INFO ][DEMO][17970] MQTT PUBLISH 18 successfully sent.
141 1797 [iot_thread] [INFO ][DEMO][17970] MQTT PUBLISH 19 successfully sent.
142 1797 [iot_thread] [INFO ][DEMO][17970] Incoming PUBLISH received:
Subscription topic filter: iotdemo/topic/3
Publish topic name: iotdemo/topic/3
Publish retain flag: 0
Publish QoS: 1
Publish payload: Hello world 18!
143 1797 [iot_thread] [INFO ][MQTT][17970] (MQTT connection 0x3ffdece8) MQTT PUBLISH operation queued.
144 1797 [iot_thread] [INFO ][DEMO][17970] Acknowledgment message for PUBLISH 18 will be sent.
145 1801 [iot_thread] [INFO ][DEMO][18010] Incoming PUBLISH received:
Subscription topic filter: iotdemo/topic/4
Publish topic name: iotdemo/topic/4
Publish retain flag: 0
Publish QoS: 1
Publish payload: Hello world 19!

```

Many thanks for the RCA (Root Cause Analysis) and the associated resolution.

I am curious to understand, why this issue has not been addressed. As part of the MQTT demo, it is expected to display the complete incoming Publish message, which includes the payload message. Any specific reason for setting the configLOGGING_MAX_MESSAGE_LENGTH macro to 192?

Thanks | Regards,
Dipen

muneebahmed10, 2020-09-12:
Hi @dipen-1337lab,

There's no specific reason for 192, it's arbitrary. There's a tradeoff between a smaller memory footprint and longer log messages, and 192 seems like a sane default.
<details>
<summary>I did a grep and these seem to be the defaults for various boards:</summary>
<br>
<pre>
vendors/nordic/boards/nrf52840-dk/aws_tests/config_files/FreeRTOSConfig.h:#define configLOGGING_MAX_MESSAGE_LENGTH            192
vendors/nordic/boards/nrf52840-dk/aws_demos/config_files/FreeRTOSConfig.h:#define configLOGGING_MAX_MESSAGE_LENGTH            256
vendors/nxp/boards/lpc54018iotmodule/aws_tests/config_files/FreeRTOSConfig.h:#define configLOGGING_MAX_MESSAGE_LENGTH            128
vendors/nxp/boards/lpc54018iotmodule/aws_demos/config_files/FreeRTOSConfig.h:#define configLOGGING_MAX_MESSAGE_LENGTH            128
vendors/cypress/boards/CY8CKIT_064S0S2_4343W/aws_tests/config_files/FreeRTOSConfig.h:#define configLOGGING_MAX_MESSAGE_LENGTH 256
vendors/cypress/boards/CY8CKIT_064S0S2_4343W/aws_demos/config_files/FreeRTOSConfig.h:#define configLOGGING_MAX_MESSAGE_LENGTH            100
vendors/cypress/boards/CYW943907AEVAL1F/aws_tests/config_files/FreeRTOSConfig.h:    #define configLOGGING_MAX_MESSAGE_LENGTH            100
vendors/cypress/boards/CYW943907AEVAL1F/aws_demos/config_files/FreeRTOSConfig.h:    #define configLOGGING_MAX_MESSAGE_LENGTH            100
vendors/cypress/boards/CYW954907AEVAL1F/aws_tests/config_files/FreeRTOSConfig.h:    #define configLOGGING_MAX_MESSAGE_LENGTH            100
vendors/cypress/boards/CYW954907AEVAL1F/aws_demos/config_files/FreeRTOSConfig.h:    #define configLOGGING_MAX_MESSAGE_LENGTH            100
vendors/xilinx/boards/microzed/aws_tests/config_files/FreeRTOSConfig.h:#define configLOGGING_MAX_MESSAGE_LENGTH            100
vendors/xilinx/boards/microzed/aws_demos/config_files/FreeRTOSConfig.h:#define configLOGGING_MAX_MESSAGE_LENGTH            100
vendors/marvell/boards/mw300_rd/aws_tests/config_files/FreeRTOSConfig.h:#define configLOGGING_MAX_MESSAGE_LENGTH            100
vendors/marvell/boards/mw300_rd/aws_demos/config_files/FreeRTOSConfig.h:#define configLOGGING_MAX_MESSAGE_LENGTH            100
vendors/mediatek/boards/mt7697hx-dev-kit/aws_tests/config_files/FreeRTOSConfig.h:#define configLOGGING_MAX_MESSAGE_LENGTH            100
vendors/mediatek/boards/mt7697hx-dev-kit/aws_demos/config_files/FreeRTOSConfig.h:#define configLOGGING_MAX_MESSAGE_LENGTH	(120)
vendors/st/boards/stm32l475_discovery/aws_tests/config_files/FreeRTOSConfig.h:#define configLOGGING_MAX_MESSAGE_LENGTH            128
vendors/st/boards/stm32l475_discovery/aws_demos/config_files/FreeRTOSConfig.h:#define configLOGGING_MAX_MESSAGE_LENGTH            160
vendors/microchip/boards/curiosity_pic32mzef/aws_tests/config_files/FreeRTOSConfig.h:    #define configLOGGING_MAX_MESSAGE_LENGTH            256
vendors/microchip/boards/curiosity_pic32mzef/aws_demos/config_files/FreeRTOSConfig.h:    #define configLOGGING_MAX_MESSAGE_LENGTH            128
vendors/nuvoton/boards/numaker_iot_m487_wifi/aws_tests/config_files/FreeRTOSConfig.h:#define configLOGGING_MAX_MESSAGE_LENGTH            80
vendors/nuvoton/boards/numaker_iot_m487_wifi/aws_demos/config_files/FreeRTOSConfig.h:#define configLOGGING_MAX_MESSAGE_LENGTH            128
vendors/infineon/boards/xmc4800_iotkit/aws_tests/config_files/FreeRTOSConfig.h:#define configLOGGING_MAX_MESSAGE_LENGTH            256
vendors/infineon/boards/xmc4800_iotkit/aws_demos/config_files/FreeRTOSConfig.h:#define configLOGGING_MAX_MESSAGE_LENGTH            256
vendors/infineon/boards/xmc4800_plus_optiga_trust_x/aws_tests/config_files/FreeRTOSConfig.h:#define configLOGGING_MAX_MESSAGE_LENGTH            256
vendors/infineon/boards/xmc4800_plus_optiga_trust_x/aws_demos/config_files/FreeRTOSConfig.h:#define configLOGGING_MAX_MESSAGE_LENGTH            256
vendors/renesas/boards/rx65n-rsk/aws_tests/config_files/FreeRTOSConfig.h:#define configLOGGING_MAX_MESSAGE_LENGTH            192
vendors/renesas/boards/rx65n-rsk/aws_demos/config_files/FreeRTOSConfig.h:#define configLOGGING_MAX_MESSAGE_LENGTH            192
vendors/espressif/boards/esp32/aws_tests/config_files/FreeRTOSConfig.h:#define configLOGGING_MAX_MESSAGE_LENGTH            192
vendors/espressif/boards/esp32/aws_demos/config_files/FreeRTOSConfig.h:#define configLOGGING_MAX_MESSAGE_LENGTH            192
vendors/vendor/boards/board/aws_tests/config_files/FreeRTOSConfig.h:#define configLOGGING_MAX_MESSAGE_LENGTH            100
vendors/vendor/boards/board/aws_demos/config_files/FreeRTOSConfig.h:#define configLOGGING_MAX_MESSAGE_LENGTH            100
vendors/ti/boards/cc3220_launchpad/aws_tests/config_files/FreeRTOSConfig.h:#define configLOGGING_MAX_MESSAGE_LENGTH            256
vendors/ti/boards/cc3220_launchpad/aws_demos/config_files/FreeRTOSConfig.h:#define configLOGGING_MAX_MESSAGE_LENGTH            192
</pre>
</details>

The `configLOGGING_MAX_MESSAGE_LENGTH` macro is part of `FreeRTOSConfig.h`, and meant to be user-configurable. If anyone wants to view longer log messages in their terminal, its value can just be increased. I've created an internal ticket to add a tip to increase this macro value in the troubleshooting steps of the AWS documentation.

Let me know if you have any further questions.

dipen-1337lab, 2020-09-14:
Namaste @muneebahmed10, 

> I did a grep and these seem to be the defaults for various boards:

Appreciate this useful input.

> I've created an internal ticket to add a tip to increase this macro value in the troubleshooting steps of the AWS documentation.

That's really nice and shall surely be helpful.

Once again, many thanks for your assistance on this issue! You have resolved my queries. No further questions 👍 

Thanks | Regards,
Dipen

closed 2020-09-14 by dipen-1337lab