aws/amazon-freertos #2443: Using OTA without aws network manager

YTvW, 2020-09-11:
**Describe the bug**
I am unable to use the OTA Agent without the aws demo network manager.

**System information**
- ESP32
- VSCode
- MacOS
- Project:Custom Application using MQTT, Greengrass, OTA 

**Expected behavior**
I expect to be able to include the OTA Agent library and use the init function to start the OTA Agent

** Actual problem **
the OTA Agent Init function requires the IotNetworkInterface_t parameter to be set in the OTA_ConnectionContext_t content. but this information is not available when using the mqtt agent to connect to IoT Core or Greengrass

YTvW, 2020-09-11:
A solution would be to allow the use of the MQTT_AGENT_Getv2Connection function as defined in mqtt_agent.c,  but that is not available to the rest of the application

pvyawaha, 2020-09-14:
Hello,

Thank you for sharing details regarding your integration issue. 

The latest release of OTA demo/library does not directly call network manager APIs and only uses the network interface. The network interface that is required for creating the HTTP connection when HTTP is selected for file download while creating the OTA Job.

If you are only using MQTT for file download then you do not need to pass this interface from your application when calling OTA_AgentInit.

YTvW, 2020-09-15:
Thanks,

This is not immediately clear from the documentation which made me think I had to specify it.

pvyawaha, 2020-09-15:
Thank you for feedback , this is only in the API docs and we will make it more clear in the demo documentation as well.

closed 2020-09-28 by pvyawaha