aws/amazon-freertos #2449: AFR & ESP32: How to INCLUDE -isystem ../libraries/3rdparty/jsmn for using JSMN library with MQTT Demo?

dipen-1337lab, 2020-09-16:
Namaste,

Below is a sample **MQTT Publish payload**, which I would like to **publish in JSON format** on the AWS IoT web console.

```
    {
      "timestamp": 1600176543,
      "deviceId": "esp32DevKitCBoard01",
      "param01": 012345.67,
      "param02": 678901.23
    }

```

After browsing the OTA Update demo code, I found the **../libraries/3rdparty/jsmn** library, which the OTA Update demo leverages for receiving an OTA update. So, I am trying to use this library for converting my Publish payload in JSON format, before calling the **IotMqtt_Publish()**.

`#include "jsmn.h"` in the **iot_demo_mqtt.c** source file generates `fatal error: jsmn.h: No such file or directory`

In the `build.ninja` log for OTA Update demo, I have observed that the OTA Update source files build with an additional `INCLUDE =  -isystem ../libraries/3rdparty/jsmn`.

Sample build command for the **aws_iot_ota_agent.c** source file is provided below. Scroll to the extreme right to view the same.

```
build CMakeFiles/afr_ota.dir/libraries/freertos_plus/aws/ota/src/aws_iot_ota_agent.c.obj: C_COMPILER__afr_ota_ ../libraries/freertos_plus/aws/ota/src/aws_iot_ota_agent.c || cmake_object_order_depends_target_afr_ota
  DEFINES = -DAFR_ESP_LWIP -DESP_PLATFORM -DGCC_NOT_5_2_0=0 -DHAVE_CONFIG_H -DIDF_VER=\"v3.3-163-g601a03e\" -DMBEDTLS_CONFIG_FILE=\"D:/gitHub/dipen-1337lab/repositories/evcharging/devV2/evcharging/device/firmware/aws/proj04/FreeRTOS/vendors/espressif/boards/esp32/components/mbedtls/port/include/mbedtls/esp_config.h\"
  DEP_FILE = CMakeFiles\afr_ota.dir\libraries\freertos_plus\aws\ota\src\aws_iot_ota_agent.c.obj.d
  FLAGS = -mlongcalls   -mlongcalls -Og -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -std=gnu99 -Wno-old-style-declaration -Wno-maybe-uninitialized
  INCLUDES = -I../libraries/freertos_plus/aws/ota/src -I../libraries/3rdparty/mbedtls/include -I../libraries/freertos_plus/aws/ota/include -I../libraries/freertos_plus/standard/crypto/include -I../freertos_kernel/include -I../vendors/espressif/esp-idf/components/app_update/include -I../vendors/espressif/esp-idf/components/bootloader_support/include -I../vendors/espressif/esp-idf/components/esp32/include -I../vendors/espressif/esp-idf/components/esp_event/include -I../vendors/espressif/esp-idf/components/efuse/include -I../vendors/espressif/esp-idf/components/driver/include -I../vendors/espressif/esp-idf/components/heap/include -I../vendors/espressif/esp-idf/components/log/include -I../vendors/espressif/esp-idf/components/nvs_flash/include -I../vendors/espressif/esp-idf/components/pthread/include -I../vendors/espressif/esp-idf/components/newlib/include -I../vendors/espressif/esp-idf/components/newlib/platform_include -I../vendors/espressif/esp-idf/components/soc/esp32/include -I../vendors/espressif/esp-idf/components/soc/include -I../vendors/espressif/esp-idf/components/spi_flash/include -I../vendors/espressif/esp-idf/components/vfs/include -I../vendors/espressif/boards/esp32/components/freertos/include -I../vendors/espressif/boards/esp32/components/mbedtls/port/include -I../vendors/espressif/esp-idf/components/tcpip_adapter/include -I../freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/include -I../firmware/include -I../vendors/espressif/boards/esp32/proj04/config_files -Iconfig -I../libraries/freertos_plus/standard/pkcs11/include -I../libraries/abstractions/pkcs11/include -I../libraries/c_sdk/standard/mqtt/include -I../libraries/c_sdk/standard/common/include -I../libraries/abstractions/platform/include -I../libraries/abstractions/platform/freertos/include -I../libraries/abstractions/secure_sockets/include -I../libraries/c_sdk/standard/common/include/private -I../libraries/c_sdk/standard/serializer/include -I../libraries/3rdparty/tinycbor -I../libraries/3rdparty/mbedtls/include/mbedtls -I../libraries/abstractions/pkcs11/mbedtls -I../libraries/c_sdk/standard/ble/include -I../libraries/abstractions/ble_hal/include -I../libraries/c_sdk/standard/https/include -isystem ../libraries/3rdparty/pkcs11 -isystem ../libraries/3rdparty/jsmn
  OBJECT_DIR = CMakeFiles\afr_ota.dir
  OBJECT_FILE_DIR = CMakeFiles\afr_ota.dir\libraries\freertos_plus\aws\ota\src
  TARGET_COMPILE_PDB = CMakeFiles\afr_ota.dir\afr_ota.pdb
  TARGET_PDB = afr_ota.pdb
```

The `build.ninja` log for the **iot_demo_mqtt.c** source file does include the `-isystem ../libraries/3rdparty/pkcs11`. The log is again shared below; scroll right towards the end. **How do I also include the '-isystem ../libraries/3rdparty/jsmn', so as to leverage the JSMN library?**

```
build CMakeFiles/proj02.dir/firmware/mqtt/iot_demo_mqtt.c.obj: C_COMPILER__proj02_ ../firmware/mqtt/iot_demo_mqtt.c || cmake_object_order_depends_target_proj02
  DEFINES = -DESP_PLATFORM -DGCC_NOT_5_2_0=0 -DHAVE_CONFIG_H -DIDF_VER=\"v3.3-163-g601a03e\" -DMBEDTLS_CONFIG_FILE=\"D:/aws/proj02/FreeRTOS/vendors/espressif/boards/esp32/components/mbedtls/port/include/mbedtls/esp_config.h\"
  DEP_FILE = CMakeFiles\proj02.dir\firmware\mqtt\iot_demo_mqtt.c.obj.d
  FLAGS = -mlongcalls   -mlongcalls -Og -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -std=gnu99 -Wno-old-style-declaration -Wno-maybe-uninitialized
  INCLUDES = -I../freertos_kernel/include -I../vendors/espressif/esp-idf/components/app_update/include -I../vendors/espressif/esp-idf/components/bootloader_support/include -I../vendors/espressif/esp-idf/components/esp32/include -I../vendors/espressif/esp-idf/components/esp_event/include -I../vendors/espressif/esp-idf/components/efuse/include -I../vendors/espressif/esp-idf/components/driver/include -I../vendors/espressif/esp-idf/components/heap/include -I../vendors/espressif/esp-idf/components/log/include -I../vendors/espressif/esp-idf/components/nvs_flash/include -I../vendors/espressif/esp-idf/components/pthread/include -I../vendors/espressif/esp-idf/components/newlib/include -I../vendors/espressif/esp-idf/components/newlib/platform_include -I../vendors/espressif/esp-idf/components/soc/esp32/include -I../vendors/espressif/esp-idf/components/soc/include -I../vendors/espressif/esp-idf/components/spi_flash/include -I../vendors/espressif/esp-idf/components/vfs/include -I../vendors/espressif/boards/esp32/components/freertos/include -I../vendors/espressif/boards/esp32/components/mbedtls/port/include -I../vendors/espressif/boards/esp32/components/freertos_tcpip/ethernet/include -I../vendors/espressif/boards/esp32/components/freertos_tcpip/smartconfig_ack/include -I../vendors/espressif/boards/esp32/components/freertos_tcpip/tcpip_adapter/include -I../libraries/freertos_plus/standard/freertos_plus_tcp/source/portable/Compiler/GCC -I../freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/include -I../firmware/include -I../vendors/espressif/boards/esp32/proj02/config_files -Iconfig -I../vendors/espressif/esp-idf/components/bt/bluedroid/api/include/api -I../vendors/espressif/esp-idf/components/bt/include -I../vendors/espressif/esp-idf/components/nimble/nimble/porting/nimble/include -I../vendors/espressif/esp-idf/components/nimble/port/include -I../vendors/espressif/esp-idf/components/nimble/nimble/nimble/include -I../vendors/espressif/esp-idf/components/nimble/nimble/nimble/host/include -I../vendors/espressif/esp-idf/components/nimble/nimble/nimble/host/services/ans/include -I../vendors/espressif/esp-idf/components/nimble/nimble/nimble/host/services/bas/include -I../vendors/espressif/esp-idf/components/nimble/nimble/nimble/host/services/gap/include -I../vendors/espressif/esp-idf/components/nimble/nimble/nimble/host/services/gatt/include -I../vendors/espressif/esp-idf/components/nimble/nimble/nimble/host/services/ias/include -I../vendors/espressif/esp-idf/components/nimble/nimble/nimble/host/services/lls/include -I../vendors/espressif/esp-idf/components/nimble/nimble/nimble/host/services/tps/include -I../vendors/espressif/esp-idf/components/nimble/nimble/nimble/host/util/include -I../vendors/espressif/esp-idf/components/nimble/nimble/nimble/host/store/ram/include -I../vendors/espressif/esp-idf/components/nimble/nimble/nimble/host/store/config/include -I../vendors/ [... cut ...] ../vendors/espressif/esp-idf/components/ulp/include -I../firmware/network_manager -I../firmware/dev_mode_key_provisioning/include -I../libraries/freertos_plus/standard/pkcs11/include -I../libraries/abstractions/pkcs11/include -I../libraries/c_sdk/standard/mqtt/include -isystem ../libraries/3rdparty/pkcs11
  OBJECT_DIR = CMakeFiles\proj02.dir
  OBJECT_FILE_DIR = CMakeFiles\proj02.dir\firmware\mqtt
  TARGET_COMPILE_PDB = CMakeFiles\proj02.dir\
  TARGET_PDB = proj02.pdb
```

Thanks | Regards,
Dipen

sarenameas, 2020-09-17:
Hello @dipen-1337,

To include jsmn just for the demo you may update [demos\mqtt\CMakeLists.txt](https://github.com/aws/amazon-freertos/blob/master/demos/mqtt/CMakeLists.txt) to include the jsmn module:

```cmake
afr_module_dependencies(
    ${AFR_CURRENT_MODULE}
    INTERFACE
        AFR::mqtt
        3rdparty::jsmn
)
```

Then regenerate the build files with CMake according to: https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html#build-and-run-example-espressif

dipen-1337lab, 2020-09-17:
Namaste @sarenameas, 

Thank you for resolving my issue. I confirm that I'm able to include the JSMN library in the iot_demo_mqtt.c source file and, generate the binary image.

Thanks | Regards,
Dipen

closed 2020-09-17 by dipen-1337lab