aws/amazon-freertos #2459: Added serial communication (UART) to aws_greengrass_discovery_demo.c

monhack, 2020-09-21:
Hallo @everyone,

I've managed to run esp32 (amazon freertos) or send data with the local network. Now I want to make a serial communication between Arduino and Esp32 (amazon freertos). Arduino is used to reading sensors and esp32 is used to send data from sensor to Greengrass with the local network. Now i am confused about how to add serial communication (UART) to esp32 so that it can connect with Arduino. I have followed the tutorial on https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/uart.html#uart-api-setting-communication-parameters but the program does not work. is there any solution so that serial communication (UART) between esp32 (amazon freertos) and arduino work well?

sarenameas, 2020-09-22:
Hello @monhack,

If you are facing issues with the espressif UART library I would ask you to please contact Espressif on their forums or on their ESP-IDF repo for help. The ESP-IDF has lots of examples of using their UART library which you may already know about: https://github.com/espressif/esp-idf/tree/master/examples/peripherals/uart . I would imagine these demos working with amazon-freertos if you placed the code you needed in your main application. 

AWS does not currently have instructions or a demonstration of how to use the esp32 UART with amazon-freertos. We have recently added a set of libraries to abstract different I/O peripherals into a common API. The UART peripheral header is found here in **common_io**:  https://github.com/aws/amazon-freertos/blob/master/libraries/abstractions/common_io/include/iot_uart.h. Espressif has recently pushed a port for the UART abstraction and it currently at the bleeding edge on master: https://github.com/aws/amazon-freertos/blob/master/vendors/espressif/boards/esp32/ports/common_io/iot_uart.c.

On master the esp32 project should have the common_io UART port available. You can try and use the functions in iot_uart.h in your esp32-amazon-freertos program. 

Let me know if this helps.

Thanks,
Sarena

alfred2g, 2020-09-25:
I will close the issue , @monhack if you have more questions please feel free to re-open the issue

closed 2020-09-25 by alfred2g