aws/amazon-freertos #2460: Device is not able to connect to AWS IOT using MQTT, when MQTT end points are changed in the device and flashed again

kumarfirmware, 2020-09-21:
Hi Team,

I am using AWS FreeRTOS version 202007 and ESP32.

I am trying to connect to  MQTT end point "XXXXXXXXX-ats.iot.us-east-1.amazonaws.com" from ESP32. It works fine.

But when I change the MQTT end point "ÝYYYYYYYY-ats.iot.us-east-1.amazonaws.com-' from ESP32  and build again and flash again, it does not work.

Is the device remembering certificates ? This is the reason it is not connecting to the AWS.

I am using MQTT over BLE.

I am using this function in the code      vDevModeKeyProvisioning(); Is this the function creating problem?

Why does device MQTT connection fail when I change MQTT end points? Is it something related to vDevModeKeyProvisioning() this function?

Thank you.

sarenameas, 2020-09-22:
Hello @kumarfirmware ,

When you switch to the `YYYYYYYYY-ats.iot.us-east-1.amazonaws.com` endpoint does your new endpoint need a different set of certificate to connect? So do you also change the certificates in [aws_clientcredential.h](https://github.com/aws/amazon-freertos/blob/202007.00/demos/include/aws_clientcredential_keys.h)?

kumarfirmware, 2020-09-22:
Hi Sarenameas,

I am connecting the device through Mobile app, which acts as proxy, MQTT over BLE.

So ideally I need not change any certificates in aws_clientcredential_keys.h.

I just need to change #define clientcredentialMQTT_BROKER_ENDPOINT.

Please let me know if my understanding is correct or not.

It is clearly mentioned in the document, we do not need any certificates when connecting MQTT over BLE.

Thanks.

sarenameas, 2020-09-22:
Hello @kumarfirmware ,

It is not enough to just switch the endpoint with #define clientcredentialMQTT_BROKER_ENDPOINT. You don't need to worry about vDevModeKeyProvisioning() that is already part of the code.

The new endpoint is a different account with new credentials. so you will need to setup Cognito on the account again, create the policy again, and reconfigure your Mobile app. Please see the instructions that you followed for the first account:
https://docs.aws.amazon.com/freertos/latest/userguide/ble-demo.html#ble-demo-prereqs

kumarfirmware, 2020-09-23:
Hi Sarenameas,

I have done cognito setup again perfectly for another account also and reconfigured mobile app also.

Meaning two different accounts, two different mobile apps, two different firmwares are all ready and tested and works separately on two different devices.

But when I test it on the same device consecutively it does not work, meaning it does not connect to AWS IOT, showing MQTT connection failed.

I suspect vDevModeKeyProvisioning() is the problem. This is some how creating the problem, may be storing certificates and keys of one account and when next time account details are changed, it is trying to access previous account certificates and keys stored in the flash. This might be the issue. Is this assumption correct? How to resolve this issue?

Also when I am connecting through mobile app using MQTT over BLE, should I use this function vDevModeKeyProvisioning() in the code? or should I comment this? what is the role of this function when using MQTT over BLE and mobile app.

Please explain in details to avoid any confusion in understanding this.

Thanks,

kumarfirmware, 2020-09-28:
Hi Sarenameas,

Good Morning. Hope you had an excellent weekend.

Can you please let me know what could be the issue?

Also can you please explain if my understanding is correct theoretically and clear my confusion?

Also can you please give me steps to debug?

Can you please tell at least my implementation is correct theoretically and we see from there what steps to follow to debug this issue?

Ideally this should be very simple for you to find out the issue, if we know the background of the function vDevModeKeyProvisioning() what it is doing exactly.

Thanks

lundinc2, 2020-09-29:
@kumarfirmware could you please provide your device side logs, as well as your app logs. Are you using the iOS app or the Android app? What is the SDK version of the app you are using?

The BLE connection does not used the client certificates provisioned using `vDevModeKeyProvisioning` so it is okay to comment that function.

Are you flashing the device with Espressif's IDF tool? If so using the `erase flash` command will ensure all the flash on your device is erased.

The MQTT endpoint and the thing name are not written to external flash by the example application, so if you have built new firmware with new values for those particular macros, you need not worry about the flash using an old value. The BLE stack does write data to external flash, for example bonding data. 

I hope this helps,

Carl

kumarfirmware, 2020-10-05:
Hi Team,

As I mentioned in the above email, I am using custom IOS app custom firmware on device, obviously base is the sample code for app and device. This is clear.

In the logs I can see MQTT connection not successful. Apart from this I do not see any thing in the logs. If MQTT connection is successful everything works, but why MQTT connection is failing I do not know.

MQTT connection is failing because of change of MQTT end point in the device and in the app and cognito is taken care as well.

But why is MQTT connection fails I do not know. This is the problem.

How can I debug this? what could be the problem for MQTT connection failure?

Thanks, Regards
Kumar

lundinc2, 2020-10-05:
Hi @kumarfirmware,

Generally the error will occur in the TLS handshake. If you enable this macro, you will see the mbed TLS stack's error log https://github.com/aws/amazon-freertos/blob/178f6f34891143c1f2148537f1da047eb100b567/vendors/espressif/boards/esp32/components/mbedtls/port/include/mbedtls/esp_config.h#L1827. This can be configured manually or through kconfig.

The device side and app side logs should help you narrow down the issue, but unfortunately I can only speculate until I see the logs.

Thanks,

Carl

kumarfirmware, 2020-10-05:
Hi Team,

Can I place this  #define MBEDTLS_DEBUG_C  in any file or I should place this in only esp_config.h file?

Also should I place this in any file or should I place this only in iot_demo_logging.h ?
#define IOT_LOG_LEVEL_NETWORK              IOT_LOG_DEBUG
#define IOT_LOG_LEVEL_MQTT                      IOT_LOG_DEBUG

I tried placing these Macros in iot_demo_logging.h but I do not see logs for MQTT and Network?

My iot_demo_logging.h is as follows

#define IOT_LOG_LEVEL_DEMO 1

/* Configure logs for the demos. The demos will have a log level of:
 * - IOT_LOG_LEVEL_DEMO if defined.
 * - IOT_LOG_LEVEL_GLOBAL if defined and IOT_LOG_LEVEL_DEMO is undefined.
 * - IOT_LOG_NONE if neither IOT_LOG_LEVEL_DEMO nor IOT_LOG_LEVEL_GLOBAL are defined.
 */
#ifdef IOT_LOG_LEVEL_DEMO
    #define LIBRARY_LOG_LEVEL        IOT_LOG_INFO
#else
    #ifdef IOT_LOG_LEVEL_GLOBAL
        #define LIBRARY_LOG_LEVEL    IOT_LOG_DEBUG
    #else
        #define LIBRARY_LOG_LEVEL    IOT_LOG_DEBUG
    #endif
#endif

#define IOT_LOG_LEVEL_NETWORK                   IOT_LOG_DEBUG
#define IOT_LOG_LEVEL_MQTT                      IOT_LOG_DEBUG

/* Set the library name to print with the demos. */
#define LIBRARY_LOG_NAME    ( "DEMO" )

Thanks.

lundinc2, 2020-10-05:
Hi @kumarfirmware,

You are correct, you will want to place that macro in `esp_config.h` as it is the configuration header file consumed by mbed TLS on that platform.

You should update the logging macro's in this file https://github.com/aws/amazon-freertos/blob/202007.00/vendors/espressif/boards/esp32/aws_demos/config_files/iot_config.h. Otherwise this file will overwrite the defaults mentioned in `iot_demo_logging.h`

Thanks,

Carl

kumarfirmware, 2020-10-05:
Hi Carl,

This will help. I will try to get the logs as soon as possible and get back to you.

Thanks, Regards,
Kumar.

lundinc2, 2020-10-05:
Hello @kumarfirmware,

This function is included for developer convenience and should not be used in production. It is included to give you an easy to use option for writing certificates and keys to your device. It is called every time because the credentials will be removed from external flash should you wipe it, for example on ESP32, if you used the `erase flash` command.

Ideally you will use a device with a (HSM)[https://en.wikipedia.org/wiki/Hardware_security_module] that stores those for you. For you, it does not matter, since you are doing MQTT over BLE. If you were to do MQTT over WiFi, you would need to provide the correct credentials to AWS IoT in order to establish a TLS connection that has mutual verification.

Thanks,

Carl

lundinc2, 2020-10-07:
Hello @kumarfirmware,

In case you haven't seen it yet, there is a README associated with said code (here)[https://github.com/aws/amazon-freertos/tree/master/demos/dev_mode_key_provisioning].

Thanks,

Carl

lundinc2, 2020-10-21:
I am closing this issue as it has gone stale. Please feel free to reopen it should you want more help.

closed 2020-10-23 by lundinc2