aws/amazon-freertos #2461: ESP receives MQTT messages from AWS IOT cloud reliably in Asia Pacific (Singapore) ap-southeast-1 with 100 % success but in Virginia region it does not receive messages reliably and success rate is 40%

Sunilpuli188, 2020-09-21:
Hi Team,

I am using AWS FreeRTOS version 202007 and ESP32.

I am facing a critical issue in my setup.

I have following two setup.

Setup 1:-

1. In the setup 1 AWS cloud is configured to Asia Pacific (Singapore) ap-southeast-1.

ESP is connected to AWS IOT cloud through Mobile app using MQTT over BLE. 

ESP subscribes to a topic and receives messages from the cloud reliably means 100% with out missing even once.

ESP subscribed to the topic with QOS 1. Messages to be delivered from the cloud are delivered with QOS 1.

Setup 2:- In this setup AWS configured in US East (N. Virginia) us-east-1.

ESP is connected to AWS IOT cloud through Mobile app using MQTT over BLE.

ESP subscribes to a topic and receives messages from the cloud unreliably means 40% success. Most of the time messages are missed.

I am clueless why messages with QOS 1 are lost despite having stable internet connection and stable BLE connection between mobile and ESP.

ESP subscribed to the topic with QOS 1. Messages to be delivered from the cloud are delivered with QOS 1.

Problem :-

Messages subscribed by ESP with QOS1 are lost only in Virginia region and 100 % success in Singapore region. 

Can you kindly let me know what could be the problem as soon as possible?

Thank you.

sarenameas, 2020-09-22:
Hello @Sunilpuli188 ,

Are you closer to the ap-southeast-1 region? If you are close to Singapore than Virgina, then packets could be dropped because your socket receive timeout is too small. The socket timeout is set when the connection is created. This is controlled by the configurable macro: IOT_NETWORK_SOCKET_POLL_MS which is currently 1000ms by default. You can increase this by adding it to [vendors\espressif\boards\esp32\aws_demos\config_files\iot_config.h](https://github.com/aws/amazon-freertos/blob/master/vendors/espressif/boards/esp32/aws_demos/config_files/iot_config.h):
```
#define IOT_NETWORK_SOCKET_POLL_MS                    2000
```

If increasing the receive timeout does not help, could you also please provide some logs for your failing runs in us-east-1?
When providing logs could you please turn on debug logging in MQTT and the network interface in [vendors\espressif\boards\esp32\aws_demos\config_files\iot_config.h](https://github.com/aws/amazon-freertos/blob/master/vendors/espressif/boards/esp32/aws_demos/config_files/iot_config.h)
Please change IOT_LOG_LEVEL_NETWORK and IOT_LOG_LEVEL_MQTT to IOT_LOG_DEBUG (and the demo also if you are using the demo):
```C
#define IOT_LOG_LEVEL_DEMO                       IOT_LOG_DEBUG
#define IOT_LOG_LEVEL_NETWORK                    IOT_LOG_DEBUG
#define IOT_LOG_LEVEL_MQTT                       IOT_LOG_DEBUG
```

Thanks,
Sarena

htibosch, 2020-09-22:
Could this be caused by the underlying TCP layer?
Looking at a PCAP, do you see many packets that are out-of-order, or retransmissions?

When the quality of Internet is low, it would help to increase time-out value, but it may also help to decrease the allowed TCP window sizes.
When there are many retransmissions while a TCP window is large, the communication can become a mess.
With a small window size of e.g. 1400 bytes, there will be at most 1 outstanding packet in both direction.

kumarfirmware, 2020-09-22:
Hi sarenameas/htibosch,

This is terrific and good news. The following fix helped in fixing the issue. How ever I will try to do regression testing and double confirm and then close the issue.

#define IOT_NETWORK_SOCKET_POLL_MS                    2000

aggarw13, 2020-10-02:
Hi @kumarfirmware, were you able to confirm that your issue was resolved with a higher socket receive timeout?

Sunilpuli188, 2020-10-05:
Hi Team,

Is 2000 the maximum for IOT_NETWORK_SOCKET_POLL_MS ?

Please tell me the maximum value what I can keep?

I see occasionally, device misses shadow message from cloud.

Thanks.

sarenameas, 2020-10-22:
Hello @Sunilpuli188 ,

2000 is the not the maximum. You can change this to whatever value suites your needs. You may increase it as much as you need. You may also try @htibosch 's suggestion and decrease the TCP window size. Let us know if you need help doing so.

Thanks,
Sarena

Sunilpuli188, 2020-10-22:
Hi Sarena,

Thanks for your response. 

#define IOT_NETWORK_SOCKET_POLL_MS 2000

With this value in long term testing I am seeing messages missing in both the directions, i.e. means on both the publish side and subscribe side.

Can you please help in this as soon as possible?

Thanks.

sarenameas, 2020-10-22:
Have you tried increasing to 4000? 10000? Go ahead and experiment with that time and let me know your results.

For the other option of changing the TCP window size. You can find an example here: https://github.com/aws/amazon-freertos/blob/master/demos/tcp/aws_tcp_echo_client_single_task.c#L252.

Sunilpuli188, 2020-10-22:
Hi Sarena,

Does this increase of timeout to 10,000 will help to address missing of messages for both device publish and subscribe ?

Also why QOS 1 is not honored in terms of subscribe and publish? QOS1 should ensure automatic retry until successful delivery but messages clearly miss here despite waiting for long time.

How does changing TCP window size can help, can you kindly explain, because device does not have internet at all. Device is doing MQTT over BLE using mobile app and connecting to cloud.

Thanks.

ravibhagavandas, 2020-10-23:
Hi @Sunilpuli188 
Could you provide more information on the following? 
1. How did you configure the device and mobile app in setup 2? Did you configure the broker endpoint in that AWS region in the device [configuration](https://github.com/aws/amazon-freertos/blob/master/demos/include/aws_clientcredential.h#L34) ? Also for the mobile app, the cognito user pool and identity pool needs to be created for that region and the credentials needs to be configured in the mobile app.
2. Mobile SDK version used and is it  for android or IOS ?
3. Could you share the device logs and mobile SDK logs for the missing packets ?

Sunilpuli188, 2020-10-23:
Hi Ravi,

Device/Mobile App/Cloud/Cognito User Pool and Identity Pool are configured in us-east-1 region. But the important this is Device is physically located in Asia pacific region.

Aws cdk at backend 1.54. Aws-sdk is 2.716. We are using it for IOS. 

Is there a mandate if device is physically located in Asia pacific region, Device/Mobile App/Cloud/Cognito should be configured to Asia pacific region only? Please confirm this clearly.

Also why QOS 1 is not honored in terms of subscribe and publish? Is it not honored in MQTT over BLE?

This issue is running from long time now, it is affecting production, can you kindly help in fixing this as soon as possible. Hopefully it should not take long time now is finding root cause and solving the problem.

I can provide the logs on the device side when packets are missing.  On the mobile app what logs you need? Because it is just a proxy connection to cloud and MQTT over BLE and I don't think there will be any logs in the mobile app. Please let me know if you need any additional information.

Thanks.

Sunilpuli188, 2020-10-28:
Hi Ravi,

Good Morning. Hope you are doing well.

Can you quickly help on this. This has stopped production.

Thanks.

ravibhagavandas, 2020-10-29:
Hi,
Sorry for a delay in response that may have caused you to wait.

> Device/Mobile App/Cloud/Cognito User Pool and Identity Pool are configured in us-east-1 region. But the important this is Device is physically located in Asia pacific region.

You can access the service endpoints in us-east-1 region from asia pacific region, but this will incur additional latency. So its highly recommended to provision devices and cognito resources in the same region as the  device physical location.
You can take a look at this [blog](https://aws.amazon.com/blogs/iot/provision-devices-globally-with-aws-iot/) to see how to provision device globally.

> Also why QOS 1 is not honored in terms of subscribe and publish? Is it not honored in MQTT over BLE?
 
MQTT over BLE should honor QOS1 as long as that subscription is acknowledged by the server as QOS1. Please not that the client specify the maximum qos in the SUBSCRIBE request and server can choose to acknowledge with a qos value less than or equal to that. This is defined in MQTT spec 3.1.1.1 section 3.8.4.

> I can provide the logs on the device side when packets are missing. On the mobile app what logs you need? 

Could you log the messages send between the Mobile APP to device and mobile app to the cloud at the time of the failure. This will help us troubleshoot if the message was received at mobile app but not send to device.

Sunilpuli188, 2020-10-29:
Hi Ravi,

Thanks for the response. We need to solve this long pending issue which is stopping production. 

Noted your suggestion it is good to have device physical location and region same, but if not same, **1. can you quantify the latency please, this will help to analyze in deep and can take decisions accordingly.**

2. Device subscribed to the topic with QOS 1 and server is publishing with QOS1, so this is clear.

QOS1 is not honored, so this is definitely an issue in FreeRTOS firmware or mobile App. This is very clear.

3. I doubt what logs I can get from mobile app, because mobile app acts as proxy connection for MQTT over BLE between device and cloud. Can you please explain what logs you are expecting on mobile app and how should I log and which modules I should log.

4. There is no need of logs on cloud, because cloud is publishing message 100%. I verified this by subscribing to # in aws Test console.

5. I can provide MQTT and Network debug logs on the device side when message is failed to receive.

We need to fix this as soon as possible with all your experience, otherwise we are risking production which is already halted because of this critical issue.

Thanks.

paulbartell, 2022-06-15:
I am closing this issue due to inactivity. We have not been able to reproduce the problem described. Feel free to re-open this issue if you have additional debug logs that we can look over to root cause the problem.

closed 2022-06-15 by paulbartell