aws/amazon-freertos #2471: how to update RSSI value on AWS things Shadow

SGOSAI602, 2020-09-23:
Describe the bug
I am trying to run the Shadow demo on ESP32 and update wifi RSSI value on AWS shadow . 
The shadow demo on ESP32 ran well for its work and updated the shadows on AWS shadows, and now has to update the RSSI value on AWS shadows, but I don't know how to update.
System information

    Hardware - ESP32-WROOM-32 D
    Operating System Linux-Ubuntu
    Version of FreeRTOS (run git describe --tags to find it)
    Project AWS FreeRTOS shadow demo

Thank you!

aggarg, 2020-09-24:
The shadow demo uses the following JSON documents to update desired and reported states: 
* https://github.com/aws/amazon-freertos/blob/master/demos/shadow/aws_iot_demo_shadow.c#L100
* https://github.com/aws/amazon-freertos/blob/master/demos/shadow/aws_iot_demo_shadow.c#L125

As you can see it is sending "powerOn" key as part of desired/reported portion of the JSON document. You can send whatever key describes the state of your device - in your case you probably want to send "rssi" key and the corresponding value should be your RSSI value. 

Is that what you are looking for?

Thanks.

SGOSAI602, 2020-09-28:
thank you for valuable reply.
but my Question is which function print RSSI value in which file for AWS freeRTOS shadow example.
Thanks.
*(3120) wifi:security type: 3, phy: bgn, rssi: -49*

[image]

Thank you!

aggarg, 2020-09-28:
That log is printed from ESP code. You can use `WIFI_Scan` API to get RSSI value: https://github.com/aws/amazon-freertos/blob/master/libraries/abstractions/wifi/include/iot_wifi.h#L454

If you do not want to perform a scan but just want to get the RSSI value, you can use `esp_wifi_sta_get_ap_info` API from Espressif: https://github.com/espressif/esp-afr-sdk/blob/29889b575e9f6f72aada23ab24d82d8d177498f0/components/esp32/include/esp_wifi.h#L452

Thanks.

SGOSAI602, 2020-10-12:
thank you,
get rssi value and store in int variable , now how to rssi variable value update on AWS cloud.
my AWS output is below
and now rssi :0 replace to my rssi: rssi value ().

[image]

> The shadow demo uses the following JSON documents to update desired and reported states:
>
>     * https://github.com/aws/amazon-freertos/blob/master/demos/shadow/aws_iot_demo_shadow.c#L100
>
>     * [...]

aggarg, 2020-10-12:
As I mentioned before, you need to update `SHADOW_DESIRED_JSON` and `SHADOW_REPORTED_JSON` JSON documents sent from the device:
* https://github.com/aws/amazon-freertos/blob/master/demos/shadow/aws_iot_demo_shadow.c#L100
* https://github.com/aws/amazon-freertos/blob/master/demos/shadow/aws_iot_demo_shadow.c#L125

Here are the relevant pieces from our sample to show you how to update the `SHADOW_DESIRED_JSON`:

```c
#define SHADOW_DESIRED_JSON     \
    "{"                         \
    "\"state\":{"               \
    "\"desired\":{"             \
    "\"rssi\":%03d"               \
    "}"                         \
    "},"                        \
    "\"clientToken\":\"%06lu\"" \
    "}"

#define EXPECTED_DESIRED_JSON_SIZE    ( sizeof( SHADOW_DESIRED_JSON ) - 1 )

static char pUpdateDocument[ EXPECTED_DESIRED_JSON_SIZE + 1 ] = { 0 };

status = snprintf( pUpdateDocument,
                   EXPECTED_DESIRED_JSON_SIZE + 1,
                   SHADOW_DESIRED_JSON,
                   ( int ) rssi,
                   ( long unsigned ) ( IotClock_GetTimeMs() % 1000000 ) );
```

Thanks.

SGOSAI602, 2020-10-13:
thank you Gaurav-Aggarwal,
rssi value update on cloud , now how to change shadow update time interval? 

Thanks

aggarg, 2020-10-13:
`AWS_IOT_DEMO_SHADOW_UPDATE_PERIOD_MS` is used to control the time interval which can be defined in your `iot_config.h`: https://github.com/aws/amazon-freertos/blob/master/vendors/espressif/boards/esp32/aws_demos/config_files/iot_config.h#L40

Thanks.

SGOSAI602, 2020-10-13:
thank you Gaurav-Aggarwal, for support and valuable reply.

SGOSAI602, 2020-10-21:
hello Gaurav-Aggarwal,
rssi value update on cloud , now how to if rssi is less than 50 then update on cloud rssi near = xyz otherwise update on cloud rssi far  = xyz.
```
#define SHADOW_DESIRED_JSON     \
    "{"                         \
    "\"state\":{"               \
    "\"desired\":{"             \
    "\"wifi\":{"		\
    "\"rssi_near\":%03d"          \
    "\"rssi_far\":%03d"          \
    "}"			\
    "}"                         \
    "},"                        \
    "\"clientToken\":\"%06lu\"" \
    "}"

```
Thanks

aggarg, 2020-10-22:
Something like this:

```c
#define SHADOW_DESIRED_FAR_JSON     \
    "{"                             \
    "\"state\":{"                   \
    "\"desired\":{"                 \
    "\"rssi far\":%03d"             \
    "}"                             \
    "},"                            \
    "\"clientToken\":\"%06lu\""     \
    "}"

#define SHADOW_DESIRED_NEAR_JSON    \
    "{"                             \
    "\"state\":{"                   \
    "\"desired\":{"                 \
    "\"rssi near\":%03d"            \
    "}"                             \
    "},"                            \
    "\"clientToken\":\"%06lu\""     \
    "}"

#define EXPECTED_DESIRED_FAR_JSON_SIZE        ( sizeof( SHADOW_DESIRED_FAR_JSON ) - 1 )
#define EXPECTED_DESIRED_NEAR_JSON_SIZE       ( sizeof( SHADOW_DESIRED_NEAR_JSON ) - 1 )

/* Near JSON document is one character larger than the far one. */
#define EXPECTED_DESIRED_MAX_JSON_SIZE        ( EXPECTED_DESIRED_NEAR_JSON_SIZE )

static char pUpdateDocument[ EXPECTED_DESIRED_MAX_JSON_SIZE + 1 ] = { 0 };

if( rssi < -50 )
{
    status = snprintf( pUpdateDocument,
                       EXPECTED_DESIRED_NEAR_JSON_SIZE + 1,
                       SHADOW_DESIRED_NEAR_JSON,
                       ( int ) rssi,
                       ( long unsigned ) ( IotClock_GetTimeMs() % 1000000 ) );
}
else
{
    status = snprintf( pUpdateDocument,
                       EXPECTED_DESIRED_FAR_JSON_SIZE + 1,
                       SHADOW_DESIRED_FAR_JSON,
                       ( int ) rssi,
                       ( long unsigned ) ( IotClock_GetTimeMs() % 1000000 ) );
}
```

Thanks.

aggarg, 2020-10-22:
You need to check the return value of snprintf for what you are writing to the buffer.

```c
static int _sendShadowUpdates( IotSemaphore_t * pDeltaSemaphore,
                               IotMqttConnection_t mqttConnection,
                               const char * const pThingName,
                               size_t thingNameLength )
{
    int status = EXIT_SUCCESS;
    int32_t i = 0, desiredState = 0;
    AwsIotShadowError_t updateStatus = AWS_IOT_SHADOW_STATUS_PENDING;
    AwsIotShadowDocumentInfo_t updateDocument = AWS_IOT_SHADOW_DOCUMENT_INFO_INITIALIZER;

    /* A buffer containing the update document. It has static duration to prevent
     * it from being placed on the call stack. */
  /*****  static char pUpdateDocument[ EXPECTED_DESIRED_JSON_SIZE + 1 ] = { 0 };********/

    static char pUpdateDocument[ EXPECTED_DESIRED_MAX_JSON_SIZE + 1 ] = { 0 };
    /* Set the common members of the Shadow update document info. */
    updateDocument.pThingName = pThingName;
    updateDocument.thingNameLength = thingNameLength;
    updateDocument.u.update.pUpdateDocument = pUpdateDocument;

    /* Publish Shadow updates at a set period. */
    for( i = 1; i <= AWS_IOT_DEMO_SHADOW_UPDATE_COUNT; i++ )
    {
        /* Toggle the desired state. */
        desiredState = !( desiredState );

        /* Generate a Shadow desired state document, using a timestamp for the client
         * token. To keep the client token within 6 characters, it is modded by 1000000. */
     /*****   status = snprintf( pUpdateDocument,
                           EXPECTED_DESIRED_JSON_SIZE + 1,
                           SHADOW_DESIRED_JSON,
                           ( int ) rssi_wifi,
                           ( long unsigned ) ( IotClock_GetTimeMs() % 1000000 ) );******/

    if( rssi_wifi < -50 )
    {
        updateDocument.u.update.updateDocumentLength = EXPECTED_DESIRED_NEAR_JSON_SIZE;
        status = snprintf( pUpdateDocument,
                           EXPECTED_DESIRED_NEAR_JSON_SIZE + 1,
                           SHADOW_DESIRED_NEAR_JSON,
                           ( int ) rssi_wifi,
                           ( long unsigned ) ( IotClock_GetTimeMs() % 1000000 ) );
        
        /* Check for errors from snprintf. The expected value is the length of
         * the desired JSON document less the format specifier for the state. */
        if( ( size_t ) status != EXPECTED_DESIRED_NEAR_JSON_SIZE )
        {
            IotLogError( "Failed to generate desired state document for Shadow update"
                         " %d of %d.", i, AWS_IOT_DEMO_SHADOW_UPDATE_COUNT );

            status = EXIT_FAILURE;
            break;
        }
        else
        {
            status = EXIT_SUCCESS;
        }
    }
    else
    {
        updateDocument.u.update.updateDocumentLength = EXPECTED_DESIRED_FAR_JSON_SIZE;
        status = snprintf( pUpdateDocument,
                           EXPECTED_DESIRED_FAR_JSON_SIZE + 1,
                           SHADOW_DESIRED_FAR_JSON,
                           ( int ) rssi_wifi,
                           ( long unsigned ) ( IotClock_GetTimeMs() % 1000000 ) );

        /* Check for errors from snprintf. The expected value is the length of
         * the desired JSON document less the format specifier for the state. */
        if( ( size_t ) status != EXPECTED_DESIRED_FAR_JSON_SIZE )
        {
            IotLogError( "Failed to generate desired state document for Shadow update"
                         " %d of %d.", i, AWS_IOT_DEMO_SHADOW_UPDATE_COUNT );

            status = EXIT_FAILURE;
            break;
        }
        else
        {
            status = EXIT_SUCCESS;
        }
    }

       /* IotLogInfo( "Sending Shadow update %d of %d: %s",
                    i,
                    AWS_IOT_DEMO_SHADOW_UPDATE_COUNT,
                    pUpdateDocument );*/

        /* Send the Shadow update. Because the Shadow is constantly updated in
         * this demo, the "Keep Subscriptions" flag is passed to this function.
         * Note that this flag only needs to be passed on the first call, but
         * passing it for subsequent calls is fine.
         */
        updateStatus = AwsIotShadow_TimedUpdate( mqttConnection,
                                                 &updateDocument,
                                                 AWS_IOT_SHADOW_FLAG_KEEP_SUBSCRIPTIONS,
                                                 TIMEOUT_MS );

        /* Check the status of the Shadow update. */
        if( updateStatus != AWS_IOT_SHADOW_SUCCESS )
        {
            IotLogError( "Failed to send Shadow update %d of %d, error %s.",
                         i,
                         AWS_IOT_DEMO_SHADOW_UPDATE_COUNT,
                         AwsIotShadow_strerror( updateStatus ) );

            status = EXIT_FAILURE;
            break;
        }
        else
        {
            IotLogInfo( "Successfully sent Shadow update %d of %d.",
                        i,
                        AWS_IOT_DEMO_SHADOW_UPDATE_COUNT );

            /* Wait for the delta callback to change its state before continuing. */
            if( IotSemaphore_TimedWait( pDeltaSemaphore, TIMEOUT_MS ) == false )
            {
                IotLogError( "Timed out waiting on delta callback to change state." );

                status = EXIT_FAILURE;
                break;
            }
        }

        IotClock_SleepMs( AWS_IOT_DEMO_SHADOW_UPDATE_PERIOD_MS );
    }

    return status;
}
```

Thanks.

SGOSAI602, 2020-10-22:
thank you gaurav,
RSSI near and  far value update on cloud but in shadow both value can saw .and now at a time only one can show on shadow output, So that's how i do it.
if far then only rssi far value can saw and if it's change then rssi replace with rssi near.

```
{
  "desired": {
    "rssi near": -59,
    "rssi far": -37
  }
}
```
Thanks.

SGOSAI602, 2020-10-29:
hi gaurav ,
i have customised json file, now how add my (json) custom file in AWS freeRTOS?
how to create or add custom source and header file in AWS freeRTOS.

Thank you!

yanjos-dev, 2021-03-11:
This question appears to be answered so I'm closing this issue. Please reopen this if you would like any additional assistance.

closed 2021-03-11 by yanjos-dev

SGOSAI602, 2021-06-04:
> `AWS_IOT_DEMO_SHADOW_UPDATE_PERIOD_MS` is used to control the time interval which can be defined in your `iot_config.h`: https://github.com/aws/amazon-freertos/blob/master/vendors/espressif/boards/esp32/aws_demos/config_files/iot_config.h#L40
> 
> Thanks.

Hi gaurav,

we change in below file but shadow update in only two times , so how we can update shadow more then two times.
/* Shadow demo configuration. The demo publishes periodic Shadow updates and responds
` * to changing Shadows. */
#define AWS_IOT_DEMO_SHADOW_UPDATE_COUNT        ( 30 )   /* Number of updates to publish. */
#define AWS_IOT_DEMO_SHADOW_UPDATE_PERIOD_MS    ( 10000 ) /* Period of Shadow updates. */ `

aggarg, 2021-06-07:
As you can see in the line below, you need to change `AWS_IOT_DEMO_SHADOW_UPDATE_COUNT` to control the number of times the shadow is updated:

```c
    /* Publish Shadow updates at a set period. */
    for( i = 1; i <= AWS_IOT_DEMO_SHADOW_UPDATE_COUNT; i++ )
```

Thanks.

SGOSAI602, 2021-06-07:
Hi, Gaurav,
In which file I can found the below function.
 `    /* Publish Shadow updates at a set period. */
    for( i = 1; i <= AWS_IOT_DEMO_SHADOW_UPDATE_COUNT; i++ )`
**Note: now we are using the latest version of amazon-freertos.**

Thanks.

aggarg, 2021-06-07:
In the latest version you can use `SHADOW_MAX_DEMO_COUNT ` : https://github.com/aws/amazon-freertos/blob/main/demos/device_shadow_for_aws/shadow_demo_main.c#L1048

> It is possible to download and use the old- version of Amazon-Freertos.

Yes. You can download any version from here: https://github.com/aws/amazon-freertos/tags. Alternatively, you can use FreeRTOS Console too: https://docs.aws.amazon.com/freertos/latest/userguide/freertos-ocw.html

Thanks.

SGOSAI602, 2021-06-07:
Hi, Gaurav,
as per you suggested I change my file but still not working, update shadow only two times.
`/*
 * FreeRTOS V202012.00
 * Copyright (C) 2020 Amazon.com, Inc. or its affiliates.  All Rights Reserved.
 *
 * Permission is hereby granted, free of charge, to any person obtaining a copy of
 * this software and associated documentation files (the "Software"), to deal in
 * the Software without restriction, including without limitation the rights to
 * use, copy, modify, merge, publish, distribute, sublicense, and/or sell copies of
 * the Software, and to permit persons to whom the Software is furnished to do so,
 * subject to the following conditions:
 *
 * The above copyright notice and this permission notice shall be included in all
 * copies or substantial portions of the Software.
 *
 * THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
 * IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY, FITNESS
 * FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE AUTHORS OR
 * COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER LIABILITY, WHETHER
 * IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM, OUT OF OR IN
 * CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE SOFTWARE.
 */

/**
 * @file shadow_demo_main.c
 *
 * @brief Demo for showing how to use the Device Shadow library's API. This version
 * of the Device Shadow API provides macros and helper functions for assembling MQTT topics
 * strings, and for determining whether an incoming MQTT message is related to the
 * device shadow. The Device Shadow library does not depend on a MQTT library,
 * therefore the code for MQTT connections are placed in another file (shadow_demo_helpers.c)
 * to make it easy to read the code using Device Shadow library.
 *
 * This example assumes there is a powerOn state in the device shadow. It does the
 * following operations:
 * 1. Establish a MQTT connection by using the helper functions in shadow_demo_helpers.c.
 * 2. Assemble strings for the MQTT topics of device shadow, by using macros defined by the Device Shadow library.
 * 3. Subscribe to those MQTT topics by using helper functions in shadow_demo_helpers.c.
 * 4. Publish a desired state of powerOn by using helper functions in shadow_demo_helpers.c.  That will cause
 * a delta message to be sent to device.
 * 5. Handle incoming MQTT messages in prvEventCallback, determine whether the message is related to the device
 * shadow by using a function defined by the Device Shadow library (Shadow_MatchTopic). If the message is a
 * device shadow delta message, set a flag for the main function to know, then the main function will publish
 * a second message to update the reported state of powerOn.
 * 6. Handle incoming message again in prvEventCallback. If the message is from update/accepted, verify that it
 * has the same clientToken as previously published in the update message. That will mark the end of the demo.
 *
 * @note This demo uses retry logic to connect to AWS IoT broker if connection attempts fail.
 * The FreeRTOS/backoffAlgorithm library is used to calculate the retry interval with an exponential
 * backoff and jitter algorithm. For generating random number required by the algorithm, the PKCS11
 * module is used as it allows access to a True Random Number Generator (TRNG) if the vendor platform
 * supports it.
 * It is RECOMMENDED to seed the random number generator with a device-specific entropy source so that
 * probability of collisions from devices in connection retries is mitigated.
 *
 */

/* Standard includes. */
#include <assert.h>
#include <stdlib.h>
#include <string.h>

/* Demo Specific configs. */
#include "shadow_demo_config.h"

#include "aws_demo.h"

/* Kernel includes. */
#include "FreeRTOS.h"
[... 1114 lines cut ...]
FAILED: CMakeFiles/afr_ota.dir/libraries/freertos_plus/aws/ota/src/aws_iot_ota_agent.c.obj 
/home/shital/AWS/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -DAFR_ESP_LWIP -DESP_PLATFORM -DGCC_NOT_5_2_0=1 -DHAVE_CONFIG_H -DIDF_VER=\"v3.3.2-235-ga3ce2f400\" -DMBEDTLS_CONFIG_FILE=\"/home/shital/old_rep/amazon-freertos/vendors/espressif/boards/esp32/components/mbedtls/port/include/mbedtls/esp_config.h\" -I../libraries/freertos_plus/aws/ota/src -I../libraries/3rdparty/mbedtls/include -I../libraries/freertos_plus/aws/ota/include -I../libraries/freertos_plus/standard/crypto/include -I../freertos_kernel/include -I../vendors/espressif/esp-idf/components/app_update/include -I../vendors/espressif/esp-idf/components/bootloader_support/include -I../vendors/espressif/esp-idf/components/esp32/include -I../vendors/espressif/esp-idf/components/esp_event/include -I../vendors/espressif/esp-idf/components/efuse/include -I../vendors/espressif/esp-idf/components/driver/include -I../vendors/espressif/esp-idf/components/heap/include -I../vendors/espressif/esp-idf/components/log/include -I../vendors/espressif/esp-idf/components/nvs_flash/include -I../vendors/espressif/esp-idf/components/pthread/include -I../vendors/espressif/esp-idf/components/newlib/include -I../vendors/espressif/esp-idf/components/newlib/platform_include -I../vendors/espressif/esp-idf/components/soc/esp32/include -I../vendors/espressif/esp-idf/components/soc/include -I../vendors/espressif/esp-idf/components/spi_flash/include -I../vendors/espressif/esp-idf/components/vfs/include -I../vendors/espressif/boards/esp32/components/freertos/include -I../vendors/espressif/boards/esp32/components/mbedtls [... cut ...] variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -std=gnu99 -Wno-old-style-declaration -Wno-maybe-uninitialized -o CMakeFiles/afr_ota.dir/libraries/freertos_plus/aws/ota/src/mqtt/aws_iot_ota_cbor.c.obj -c ../libraries/freertos_plus/aws/ota/src/mqtt/aws_iot_ota_cbor.c
/bin/sh: 1: /home/shital/AWS/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: not found
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

aggarg, 2021-06-07:
> as per you suggested I change my file but still not working, update shadow only two times.

Okay, my bad. That macro only controls the number of times to retry in case of failure. This demo is written to first update the desired state and then report the updated state - read the function `RunDeviceShadowDemo`. You will need to update this function to continuously update the shadow as per your application's requirement. 

Thanks.

SGOSAI602, 2021-07-22:
Hello,
how to update the SHADOW_DESIRED_JSON: if my sensor data value is **1234** what and where the changes in code.
we changed in below function but geeting shadow rejected error.

`#define SHADOW_DESIRED_JSON     \
    "{"                         \
    "\"state\":{"               \
    "\"desired\":{"             \
    "\"rssi\":%04d"               \
    "}"                         \
    "},"                        \
    "\"clientToken\":\"%06lu\"" \
    "}"

`
I want shadow update like below.

`{
 "desired": {
   "wifi": {
     "rssi ": 1234

   }
 }
}`
thanks