aws/amazon-freertos #2488: AFR & ESP32 - How to defer/ postpone an OTA update on the device?

dipen-1337lab, 2020-09-28:
Namaste Community Member,

For my product under development, I am able to update my application firmware on the ESP32-DevKitC target board. This is achieved by running the **AFR OTA Update Thread** on the board and, leveraging the **Jobs** and **Amazon S3 Bucket** on the AWS IoT console.

The firmware version is **FreeRTOS V202002.00** and, is downloaded from the AWS IoT console (FreeRTOS Device Software). The OTA Update demo code is leveraged and, integrated in my firmware.

**Observations (in line to the OTA Update Demo implementation)**
1. The OTA Update Thread is continuously running on the device.
2. An OTA Update on the device begins immediately once a Job is created on the AWS IoT console.
3. Device restarts immediately once the firmware signature is validated and, applies the new firmware as the active image upon a successful reboot.

**Requirement**
_At the device end_
1. Either disable the OTA Update thread when the device is in the Active state.
2. Or, accept the firmware received via OTA update but, restart the device only when the device is in the Standby state.

_At the AWS IoT console end_

1. Schedule the OTA Update Job at a specific time.

**Code modification attempts**
Two attempts to defer the OTA update were unsuccessful. The **vTaskDelay()** statement was added [here](https://github.com/aws/amazon-freertos/blob/master/demos/ota/aws_iot_ota_update_demo.c#L329) and [here](https://github.com/aws/amazon-freertos/blob/master/libraries/freertos_plus/aws/ota/src/aws_iot_ota_agent.c#L1165). These two references are related to the **master branch.**

Your inputs for my above requirements or, a better recommendation are truly appreciated!

Thanks | Regards,
Dipen

pvyawaha, 2020-09-28:
Hello,

Thank you for sharing details regarding your requirement. 

When the update file is downloaded and its signature is verified and the application callback `App_OTACompleteCallback` is called with event` eOTA_JobEvent_Activate` .In the OTA demo the MQTT connection is disconnected and `OTA_ActivateNewImage` is called to reset the device. 

The use case that you have is at a specific state the device should call `OTA_ActivateNewImage`. We can do this in several ways depending on how your application is designed. One way is to loop with a delay for the device to be in specific state before calling this function in the `App_OTACompleteCallback`. You can also use direct task notification or semaphore to block the task before calling `OTA_ActivateNewImage `and send the notification from other application task. Another way is instead of calling `OTA_ActivateNewImage` from `App_OTACompleteCallback` you can notify and call it from your application task. 

If you do not want to keep OTA Agent running you can start and shutdown the OTA Agent depending on your application requirements.

Please let me know if this helps or if you have more questions.

dipen-1337lab, 2020-09-29:
Namaste @pvyawaha,

Many thanks for your prompt assistance on my requirement. I'll explore your suggestions and, validate the successful deferring of the OTA update in my firmware.

Closing this issue for now. In case of any issues, I shall re-open this issue again!

Thanks | Regards,
Dipen

closed 2020-09-29 by dipen-1337lab

dipen-1337lab, 2021-01-20:
Namaste @pvyawaha,

 

> One way is to loop with a delay for the device to be in specific state before calling `OTA_ActivateNewImage` in the `App_OTACompleteCallback`.

I'm happy to share the below console log after implementing the logic suggested by you, to defer the device restart after receiving the entire firmware as part of OTA update.

```
Result           : Test Passed (Restart is deferred during active session. Later, new version of firmware runs as desired -- OTA completed successfully.)
                   5843 61085 [OTA Agent Task] [INFO ][DEMO][610850] INFO : Session is active. Deferring device restart.
                   ...
                   ...
                   Swipe RFID card to end session
                   5989 64085 [OTA Agent Task] [INFO ][DEMO][640850] INFO : Closing MQTT connection and activating new image.
                   ...
                   ... <Device restarts>
                   ...
                   54 884 [OTA Agent Task] [prvUpdateJobStatus_Mqtt] Msg: {"status":"SUCCEEDED","statusDetails":{"reason":"accepted v0.9.3"}}
```

The code block for the above behaviour is as below

```
static void App_OTACompleteCallback( OTA_JobEvent_t eEvent )
{
    OTA_Err_t xErr = kOTA_Err_Uninitialized;

    DEFINE_OTA_METHOD_NAME( "App_OTACompleteCallback" );

    /* OTA job is completed. so delete the MQTT and network connection. */
    if( eEvent == eOTA_JobEvent_Activate )
    {
        IotLogInfo( "Received eOTA_JobEvent_Activate callback from OTA Agent.\r\n" );

        /*
         * Activate the new image received over OTA once active session ends.
         * Periodically check for the state in this OTA Complete Callback.
         */
        while(1)
        {
            if( ( STATE_STANDBY == deviceState ) )
            {
                break;
            }

            IotLogInfo( "INFO : App_OTACompleteCallback - File receive complete "
                    "and signature is valid. Deferring the device restart since "
                    "the session is active.\r\n" );

            // Delay for periodic check.
            vTaskDelay( 1000 );
        }

        IotLogInfo( "INFO : App_OTACompleteCallback - Device in STATE_STANDBY "
                "state. Closing MQTT connection and activating new image.\r\n" );

        /* OTA job is completed. so delete the network connection. */
        if( _mqttConnection != NULL )
        {
            IotMqtt_Disconnect( _mqttConnection, false );
        }

        /* Activate the new firmware image. */
        OTA_ActivateNewImage();

        /* We should never get here as new image activation must reset the device.*/
        IotLogError( "New image activation failed.\r\n" );

        for( ; ; )
        {
        }
    }
    else if( eEvent == eOTA_JobEvent_Fail )
    {
        IotLogInfo( "Received eOTA_JobEvent_Fail callback from OTA Agent.\r\n" );

        /* Nothing special to do. The OTA agent handles it. */
    }
    else if( eEvent == eOTA_JobEvent_StartTest )
    {
        /* This demo just accepts the image since it was a good OTA update and networking
         * and services are all working (or we wouldn't have made it this far). If this
         * were some custom device that wants to test other things before calling it OK,
         * this would be the place to kick off those tests before calling OTA_SetImageState()
         * with the final result of either accepted or rejected. */

        IotLogInfo( "Received eOTA_JobEvent_StartTest callback from OTA Agent.\r\n" );
        xErr = OTA_SetImageState( eOTA_ImageState_Accepted );

        if( xErr != kOTA_Err_None )
        {
            IotLogError( " Error! Failed to set image state as accepted.\r\n" );
        }
    }
}
```

### Question from AWS IoT Jobs perspective

With the above approach, the device restart now gets delayed till an active session is in progress. As a consequence, the below response to the AWS IoT Jobs also gets delayed.
`54 884 [OTA Agent Task] [prvUpdateJobStatus_Mqtt] Msg: {"status":"SUCCEEDED","statusDetails":{"reason":"accepted v0.9.3"}}`

How long does the AWS IoT Jobs wait for the above response from the device? Is there a timeout associated? If yes, what is the timeout value in seconds?

The question arises from a scenario such that, if the active session goes on for a significant amount of time (few hours), it is possible that the AWS IoT Jobs concludes that the OTA has failed. AWS IoT Jobs will then provide a wrong status about the OTA Update, which the device will be sending on restart, after the currently-active session ends.

**PS**. Let me know if I need to open another issue for this query. Posting this query here, since the context is in line to the original question raised.

Thanks | Regards,
Dipen

dipen-1337lab, 2021-01-20:
Namaste @pvyawaha,

So from this [JobsExecution](https://docs.aws.amazon.com/iot/latest/apireference/API_JobExecution.html) documentation and as per the below Job Creation step, I understand that it is possible to configure the **Timeout** duration.

[image]

Considering that this parameter is _optional_, will the OTA Job indefinitely remain in the _In Progress_ state, awaiting a response from the device, if _No timeout duration configured_ is opted for?

Thanks | Regards,
Dipen