aws/amazon-freertos #2489: how to create or add custom source and header file in AWS freeRTOS.

SGOSAI602, 2020-09-28:
**Describe the bug**
how to create or add custom source and header file in AWS freeRTOS.

**System information**
Hardware - ESP32-WROOM-32 D
Operating System Linux-Ubuntu
Version of FreeRTOS (run git describe --tags to find it)
Project AWS FreeRTOS shadow demo

Thank you!

alfred2g, 2020-09-28:
What exactly are you trying to achieve here?
you wanna add source and headers to a demo projects? test project?
amazon-freertos uses cmake as the build system, you will have to add the source files you want to one of the CMakelists.txt

yanjos-dev, 2020-10-22:
@SGOSAI602 I'm going to close this for now. Please feel free to reopen this issue if you have any further issues or questions.

closed 2020-10-22 by yanjos-dev