aws/amazon-freertos #2492: Not able to add more than 4 characteristics in a service in ble gatt server demo in aws freertos

Padmavathi-Balachandran, 2020-09-28:
I am using the ble_gatt_server demo in the demos/ble in amazon freertos. I need to add five characteristics to the service. 3 Read and 2 Write characteristics. I can succesfully get the first four characteristics in the mobile application and not the last one. Is there any limit to the number of characteristics or do i need to change any configurations? It will be of great use if someone could help me in this issue. Thanks.

alfred2g, 2020-09-28:
Yes there are 2 configurations to control the table size and field size
```c
#define NRF_SDH_BLE_GATTS_ATTR_TAB_SIZE (1300*6)
#define NRF_SDH_BLE_VS_UUID_COUNT 19
```
Which could be different than the values above depending on the the board you are using.
you should override the default values in sdk_config.h from your config file iot_ble_config.h

Hope this helps!

Padmavathi-Balachandran, 2020-09-29:
Thanks for Your response. I am using esp32 board. I checked the iot_ble_config.h , iot_ble_config_defaults.h. 
I couldnt find the configurations you have mentioned. May be these macros names varies for the esp32.Can you please tell me for the esp32 board?

alfred2g, 2020-09-30:
What error are you getting ?

Padmavathi-Balachandran, 2020-09-30:
I didn't get any error. Everything is working fine. I could see the Four characteristics.
[image] 
 
I used this to see the memory---> ESP_LOGI(TAG, "RAM left-initial %d", esp_get_free_heap_size());
When starting the current project, i have 163156 bytes. At the end of the project i have 119728 bytes.

alfred2g, 2020-10-01:
what kind of device are you using?
IOS keeps the cache of the  GATT attribute table on the phone, could you try to remove and re-pair the bond and turn off/on  BLE?

yanjos-dev, 2020-10-22:
@Padmavathi-Balachandran I'm going to close this for now. Please feel free to reopen this if you have any further issues or questions.

closed 2020-10-22 by yanjos-dev