aws/amazon-freertos #2494: Error demo application's build files with CMake

abhi-ratnman, 2020-09-29:
**Describe the bug**
I am using ESP-IDF V4.2 I have installed it using the windows installer. But while building demo application usiong 
cmake -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 -GNinja -S . -B build-directory

it shows the error printed below. 
**System information**
- Which hardware board or part numbers?
- Operating System [Windows]
- Version of FreeRTOS (run ``git describe --tags`` to find it): 202007.00-187-g4ffd042a8
- Project [While Installation]

**Screenshots or console output**
CMake Error at vendors/espressif/esp-idf/tools/cmake/kconfig.cmake:38 (message):
  Windows requires a prebuilt mconf-idf for your platform on the PATH, or an
  MSYS2 version of gcc on the PATH to build mconf-idf.  Consult the setup
  docs for ESP-IDF on Windows.
Call Stack (most recent call first):
  vendors/espressif/esp-idf/tools/cmake/idf_functions.cmake:25 (include)
  vendors/espressif/boards/esp32/CMakeLists.txt:39 (include)
  CMakeLists.txt:70 (include)

Thank you!

PS. I am new to Amazon freertos sorry for any missing information

aggarw13, 2020-09-30:
Hi @abhi-ratnman, from the error message you have shared, it seems like the `mconf-idf.exe` binary is not being in the PATH that is required for the build. 
Can you make sure that the `mconf-idf.exe` is added to your system PATH environment before running the build? 

Also, we do not support ESP-IDF V4.2 toolchain for the build but instead the ESP-IDF V3.3 toolchain as mentioned in our [Getting Started Guide AWS docs](https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html#setup-env-esspressif). Please make sure to use the supported toolchain version for building the demo application.

Please let us know if you have any further questions.

abhi-ratnman, 2020-10-01:
Hello @aggarw13 the error was with the version of IDF i used v4.2 earlier. After installing IDF v3.3 the amazonfreertos worked. 

Thanks for the help.

Also one more thing I am planning to use ESP32 S2 for my upcoming project in near future that was the reason to install idf v4.2. So can we expect the support for ESP32 S2 boards soon?

aggarw13, 2020-10-01:
Hi, @abhi-ratnman, I am happy to know that you were able to fix the issue.

Regarding your inquiry about the ESP32 S2, we cannot currently provide a timeline about qualification of the board. Thank you for your feedback about it. I will add the board qualification request to our internal queue. Please keep an eye for future updates from us about new board qualifications.

closed 2020-10-01 by aggarw13

cobusve, 2020-10-05:
> Hello, @aggarw13 the error was with the version of IDF I used v4.2 earlier. After installing IDF v3.3 I was able to build and flash.
> 
> Thanks for the help.
> 
> Also one more thing I am planning to use ESP32 S2 for my upcoming project in the near future that was the reason to install idf v4.2. So can we expect the support for ESP32 S2 boards soon?

There is alredy a branch for supporting v4.2 and the ESP32 S2 which you can find here https://github.com/aws/amazon-freertos/tree/feature/idf_uplevel_v4.2