aws/amazon-freertos #2498: asw_demo and aws_test for running BLE, why ucChannelMap = 0 ?

wingwu2379, 2020-09-30:
**Describe the bug**
For running BLE, I'm curious about that why aws_demo and aws_test all set **ucChannelMap = 0**
**aws_demo** --> _setAdvData()  in \\amazon-freertos\libraries\c_sdk\standard\ble\src\iot_ble_gap.c   
**aws_test** -->  xAdvertisementConfigA and xAdvertisementConfigB \\amazon-freertos\libraries\abstractions\ble_hal\test\src\iot_test_ble_hal_common.c

From spec, it said "at least one channel bit shall be set"
[image]

aggarw13, 2020-10-02:
Hi @wingwu2379, thank you for pointing out the issue. We are investigating into the use of advertising channel map value, and will provide an update once we have a resolution for it.

ravibhagavandas, 2020-11-19:
@wingwu2379 
The `ucChannelMap` is set to zero as an indication to use vendor stack specific values. I have updated the documentation for it to be more clear in PR #2768 .

ravibhagavandas, 2020-12-15:
The documentation is updated and merged to master.
I will close this issue. Please feel free to reopen if you have more questions.

closed 2020-12-15 by ravibhagavandas