aws/amazon-freertos #2508: common_io not available

YTvW, 2020-10-04:
**Describe the bug**
when using amazon-freertos as a library the common_io library is not available to use in custom components

**System information**
- Which hardware board or part numbers?
- VSCode
- Operating System :MacOS
- Project: Custom Application

**Expected behavior**
When using target_link_libraries() is should not get the following error:

**Screenshots or console output**
Target "base" links to target "AFR::common_io" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

**Additional context**
I think this is due to the cmakelist.txt of the common_io library having afr_module(INTERFACE) instead of afr_module() and
```
afr_module_include_dirs(
    ${AFR_CURRENT_MODULE}
    INTERFACE "${inc_dir}"
)
```
instead of 
```
afr_module_include_dirs(
    ${AFR_CURRENT_MODULE}
    PUBLIC "${inc_dir}"
)
```

Thank you!

aggarw13, 2020-10-04:
Hi @YTvW, yes, the `AFR::common_io` module is currently configured as an `INTERFACE` instead of a library module that can be linked against. Thank you for letting us know about your use-case. We will look into updating it as a library target, and provide an update soon.

YTvW, 2020-10-05:
I am curious about the reasoning for that, since this way the user is not able to write fully portable code that is reusable

aggarw13, 2020-10-05:
The `AFR::common_io` was created as an `INTERFACE` to only represent the set of header files, present [here](https://github.com/aws/amazon-freertos/tree/master/libraries/abstractions/common_io/include), that act as the abstraction interface.  The actual implementation of the interface is represented by the `AFR::common_io::mcu_port` module that contains the silicon-vendor specific port of the `AFR::common_io` interface.

With the CMake configuration in **202007.00** version, the linking dependency can be achieved by linking against the `AFR::common_io::mcu_port` module which is exposed as a library target, and also having an interface dependency with `AFR::common_io` (to obtain the library interface header files). 

To make it easier for users to not require adding dependency on both `AFR::common_io` (as the interface) and `AFR::common_io::mcu_port`, I have created a PR #2509  to update the CMake configuration that will allow users to just link against `AFR::common_io` with `target_link_libraries`

aggarw13, 2020-10-05:
@YTvW, we have merged the CMake update for `AFR::common_io` module to be built as a library in 54407b7.  Please let us know if you have any further questions on this.

closed 2020-10-06 by aggarw13

ChristyBrouwer, 2020-11-30:
I'm having a similar issue on the ESP32, the common_io library can't be linked nor found in custom components. My build is currently up to date with master.