aws/amazon-freertos #2510: When I force cancel OTA Job from AWS IOT, device crashes.

kumarfirmware, 2020-10-05:
Hi Team,

I am using AWS FreeRTOS version 202007 and ESP32.

When I force cancel OTA Job from AWS IOT, device crashes. Can you please help me with this?

This is the error log

339 4173 [OTA Agent Task] [prvParseJSONbyModel] Extracted parameter [ clientToken: 1:AA:BB:CC:DD:EE:FF ]
340 4173 [OTA Agent Task] [prvParseJSONbyModel] Extracted parameter [ timestamp: 1601468899 ]
341 4173 [OTA Agent Task] [prvParseJSONbyModel] parameter not present: execution
342 4174 [OTA Agent Task] [prvParseJSONbyModel] parameter not present: jobId
343 4174 [OTA Agent Task] [prvParseJSONbyModel] parameter not present: jobDocument
344 4174 [OTA Agent Task] [prvParseJSONbyModel] parameter not present: afr_ota
345 4174 [OTA Agent Task] [prvParseJSONbyModel] parameter not present: protocols
346 4174 [OTA Agent Task] [prvParseJSONbyModel] parameter not present: files
347 4174 [OTA Agent Task] [prvParseJSONbyModel] parameter not present: filepath
348 4174 [OTA Agent Task] [prvParseJSONbyModel] parameter not present: filesize
349 4174 [OTA Agent Task] [prvParseJSONbyModel] parameter not present: fileid
350 4174 [OTA Agent Task] [prvParseJSONbyModel] parameter not present: certfile
351 4174 [OTA Agent Task] [prvParseJSONbyModel] parameter not present: sig-sha256-ecdsa
352 4174 [OTA Agent Task] Custom Job Found:
353 4174 [OTA Agent Task] [prvDefaultCustomJobCallback] Received Custom Job inside OTA Demo.
354 4174 [OTA Agent Task] [prvParseJobDoc] Job does not have context or has no ID but has been processed
355 4174 [OTA Agent Task] [prvParseJobDoc] Ignoring job without ID.
356 4174 [OTA Agent Task] [INFO ][MQTT][41740] (MQTT connection 0x3fff49a8) UNSUBSCRIBE operation scheduled.
357 4174 [OTA Agent Task] [prvUnSubscribeFromJobNotificationTopic] OK: $aws/things/AA:BB:CC:DD:EE:FF/jobs/notify-next
358 4174 [OTA Agent Task] [INFO ][MQTT][41740] (MQTT connection 0x3fff49a8) UNSUBSCRIBE operation scheduled.
359 4174 [OTA Agent Task] [prvUnSubscribeFromJobNotificationTopic] OK: $aws/things/AA:BB:CC:DD:EE:FF/jobs/$next/get/accepted
360 4174 [OTA Agent Task] [INFO ][MQTT][41740] (MQTT connection 0x3fff49a8, UNSUBSCRIBE operation 0x3ffed8ac) Waiting for operation completion.
361 4210 [OTA Agent Task] [INFO ][MQTT][42100] (MQTT connection 0x3fff49a8, UNSUBSCRIBE operation 0x3ffed8ac) Wait complete with result SUCCESS.
362 4210 [OTA Agent Task] [INFO ][MQTT][42100] (MQTT connection 0x3fff49a8, UNSUBSCRIBE operation 0x3fff5078) Waiting for operation completion.
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400014fd  PS      : 0x00060430  A0      : 0x800f7028  A1      : 0x3fffab30
A2      : 0x00000000  A3      : 0xfffffffc  A4      : 0x000000ff  A5      : 0x0000ff00
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x00000011
A10     : 0x3fffaf30  A11     : 0x3ffbfd40  A12     : 0x3fffaf41  A13     : 0x00000000
A14     : 0x3f4172b4  A15     : 0x3fff49a8  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff

ELF file SHA256: f69f911690f6de31

Backtrace: 0x400014fd:0x3fffab30 0x400f7025:0x3fffab40 0x400f50be:0x3fffae50 0x40143c0c:0x3fffaf10 0x40143f7c:0x3fffb050 0x4014145d:0x3fffb070 0x40142172:0x3fffb090 0x40142186:0x3fffb110 0x40142a18:0x3fffb140 0x40141757:0x3fffb170 0x401423be:0x3fffb190

Thank you.

pvyawaha, 2020-10-05:
Hello, 

Thanks for sharing call stack  and details, can you please confirm what network interface you are using WiFi or BLE?

kumarfirmware, 2020-10-05:
Hi Team,

I guess, I am able to reproduce this by using Wi-Fi network.

Thanks.

tgsong, 2020-10-05:
Hi, can you try to see if master branch has this issue too? We merged a fix couple weeks ago that addressed a crash, https://github.com/aws/amazon-freertos/pull/2454. Meanwhile, can you also get your device output with ESP32's `idf_monitor.py`? It's under `vendors/espressif/esp-idf/tools`. It will help print the call stack so we could know exactly where it crashed.

kumarfirmware, 2020-10-08:
Hi Team,

I Could see there are changes in OTA library files between the release version and the latest master version.

I will merge those changes in to my project and I will test multiple times and let you know the result.

Thanks.

kumarfirmware, 2020-10-13:
Hi Team,

With the latest master branch files, issue is solved. OTA does not crash any more if it is force cancelled.

I have a question, 

Is there a function in OTA which will give me the current status of OTA like OTA job in progress or OTA job not in progress or OTA Job started or OTA job not started like this? Is there any API which will return me current status of the OTA Job?

Thanks.

pvyawaha, 2020-10-16:
Hello,

You can use the OTA library API OTA_GetAgentState described here - https://freertos.org/Documentation/api-ref/ota/aws__iot__ota__agent_8h.html#ae15e4aa0f46a43e9282e5bc6ee70c2b9

kumarfirmware, 2020-10-20:
Hi Team,

Thanks for the response.

Is this function  OTA_GetAgentState() thread safe?

In thread1 I am calling this function  OTA_AgentInit_internal(), that means thread1 is running OTA.

In thread2 I need to check the state of OTA. can I use this function for error free results or should I use any Mutex. Please advise.

Thanks.

pvyawaha, 2020-10-21:
Hello,

The OTA_GetAgentState function is thread safe.

Can you share the logs and call stack for the crash ? Do you have the above mentioned BLE fix in your build when you observe the crash?

kumarfirmware, 2020-10-21:
Hi Team,

Thanks for the response. 

I can safely use OTA_GetAgentState  function.

What is the BLE fix you are saying about?

Thanks.

pvyawaha, 2020-10-23:
The PR mentioned in one of the above comment https://github.com/aws/amazon-freertos/pull/2454

kumarfirmware, 2020-10-23:
Hi Team,

The latest code is taken and OTA does not crash any more if it is force cancelled.

But crash is observed this time at the beginning of the OTA, I will provide you logs as soon as possible.

Thanks

kumarfirmware, 2020-10-28:
Hi Team,

I am not able to get the ESP32's idf_monitor.py running from AWS FreeRTOS workspace. 

Can you please tell me how to get this up and running with AWS FreeRTOS workspace. 

Because AWS  FreeRTOS workspace builds with cmake command. We do not use idf_monitor.py. But how do we combine both for debugging. 

kindly advice on the same.

Thanks.

pvyawaha, 2020-10-30:
Hello , 

Do you still have issue with OTA after BLE fix merged?

ravibhagavandas, 2020-10-30:
Hi,

Regarding 
>I am not able to get the ESP32's idf_monitor.py running from AWS FreeRTOS workspace.

You can run the command `./vendors/espressif/esp-idf/tools/idf_monitor.py -p <serial port> build/<binary name>.elf` from the project home folder .

Please let us know if this works

pvyawaha, 2020-12-17:
Hello Kumar, 

Do you still have the issue or is it resolved ?

closed 2020-12-18 by pvyawaha