aws/amazon-freertos #2511: sometimes creating MQTT connection does not happen through Network manager using Bluetooth

kumarfirmware, 2020-10-05:
Hi Team,

I am using AWS FreeRTOS version 202007 and ESP32.

I am using Network Manager to manage BLE and Wi-Fi networks.

I am working through MQTT over BLE.

When I connect through mobile app and give BLE connectivity to device it connects through MQTT and push messages to AWS IOT.

But after some time, when I connect through mobile app and give BLE connectivity to device it cannot connect through MQTT, I mean it does not even attempt to create MQTT connection. But device gets BLE connectivity.

May be multiple MQTT connect and disconnects is the reason?

If I do a power reset, this problem will be solved and everything works fine. Again problem starts to happen after several MQTT connect and disconnect

Can you please let me know what can be the issue.

Please find the code snippet attached to the email.

Thanks, Regards,
Kumar.

aggarw13, 2020-10-05:
Hi @kumarfirmware, to better understand your issue, can you clarify about how you are detecting the problem (i.e. whether through logs on the device OR from the phone)? Are there steps that we can follow to re-produce the issue?
Are you able to see logs on the device from the MQTT library of creating a new MQTT connection? If you have device (and our mobile application logs). please share with us device logs that can help us better understand the issue.
That will help understand whether the issue is in the network manager layer or in the application that is unable to call the MQTT API. 

Besides, looking at the shared code snippet, there seems to be a state machine in the device application. 
Can you share whether the application is expected to make multiple MQTT connect/disconnect calls in the `while` loop forever depending on the application state? Also, are there simultaneous MQTT connections being created from the device. (If yes, please make sure to use different client IDs for each parallel connection; otherwise, there will be client ID collisions causing between the open MQTT connections that will cause connections to be closed by the broker.)

kumarfirmware, 2020-10-06:
Hi Team,

Thanks for the reply. 

I am detecting the problem through the logs on the device. When ever I see the issue, I could clearly see from the logs, device has BLE connectivity from mobile. Still it does not try to create MQTT connection.

May be network  manager is unable to detect quick BLE connect, BLE disconnect and again BLE connect. I could think this might be the reason.

Yes issue might be in the network manager layer. Application is unable to call MQTT API to create connection may be because network manager does not detect existence of BLE connection I guess.

Yes application will make multiple MQTT connect/disconnect calls, only if there is loss of BLE connectivity i.e if BLE connect/disconnect/connect happens otherwise at any given point of time it will be always single MQTT connection exists.

There are no simultaneous MQTT connections exist. There is only one MQTT connection at any given point of time. the logic or state machine is to ensure when MQTT connection gets disconnected because of loss of BLE it should do MQTT again when BLE connection is back.

Please let me know what can be the issue?

Thanks.

aggarw13, 2020-10-06:
Hi @kumarfirmware, thank you for clarifying on the issue you are noticing, and expected behavior of your device application logic. 

Regarding your use of the network manager, are you registering a `AwsIotNetworkStateChangeCb_t` callback with the network manager (through the [AwsIotNetworkManager_SubscribeForStateChange](https://github.com/aws/amazon-freertos/blob/1fecc0e8b723deb28ffd0e9e41dcb8a99d491ae8/demos/network_manager/iot_network_manager_private.h#L95-L110) function)? This callback is invoked by the network manager on network connection state change (for both connect and disconnect state changes). This callback should provide a mechanism of detecting BLE connection/disconnection, and taking action of MQTT re-connection.

You can reference the callback we use in the demos [here](https://github.com/aws/amazon-freertos/blob/1fecc0e8b723deb28ffd0e9e41dcb8a99d491ae8/demos/demo_runner/iot_demo_freertos.c#L120) for example of how the network connection can be re-enabled on being invoked with the `eNetworkStateDisabled` status value.

Hope that helps!

kumarfirmware, 2020-10-14:
Hi Team,

I have already registered a call back with AwsIotNetworkManager_SubscribeForStateChange. But I still have the issue.

As I already shared the code in the above, there is a reconnection logic as follows if the  xNetworkConnected flag is false.
 
But still the issue exists, when there is a disconnection some times xNetworkConnected  flag is not set to false and reconnect logic is not triggered.

This is extremely critical. Kindly let me know as soon as possible, how this could be solved?

Reconnection Logic:-

 if(xNetworkConnected == pdFALSE)
            {
                configPRINTF( ( "MQTT connection got disconnected\r\n" ) );
                //configPRINTF( ( "Calling OTA agent Shutdown function \r\n" ) );

                AwsIotShadow_Cleanup();
                AwsIotShadow_Init(0);

                OTA_AgentShutdown(0);

                while( ( eState = OTA_GetAgentState() ) != eOTA_AgentState_Stopped )
                {
                     vTaskDelay( myappONE_SECOND_DELAY_IN_TICKS );
	            }
                //vTaskDelay( pdMS_TO_TICKS( 10000UL ) );
                //configPRINTF( ( "OTA agent Shutdown successfully\r\n" ) );

                IotMqtt_Disconnect( xConnection.xMqttConnection, false );
                vMqttDemoDeleteNetworkConnection( &xConnection );
                vTaskDelay(  pdMS_TO_TICKS( 5000UL ) );
            }

Thanks.

ravibhagavandas, 2020-10-23:
@kumarfirmware 
In the above code snippet, I see only the disconnect and cleanup logic for all libraries. From the code snippet `Test.c.txt` posted above I see the reconnect logic but that gets triggered only when `state = CONN_STATE`. Where is the `state` variable populated to `CONN_STATE` based on the `xNetworkConnected` flag. Could you share the code snippet for that.
Please note that network manager will only notify of the physical BLE disconnection and re-connection. After BLE reconnects, you need to still create a network interface connection for MQTT transport and then an MQTT connection.

Also this logic is not required as the keep alive is handled transparently for a BLE or WIFI connection.
```
    if( xConnection.ulNetworkType == AWSIOT_NETWORK_TYPE_BLE )
     {
                        xConnectInfo.awsIotMqttMode = false;
                        xConnectInfo.keepAliveSeconds = 0;
     }
```

kumarfirmware, 2020-10-28:
Hi Ravi,

Thanks for the response. I am not using any pairing or bonding.

Please find the updated code attached here.

I found out a critical issue, when BLE is successfully connected and for some reason MQTT connection failed and BLE disconnected and connected again, there is a deadlock with respect to semaphore. This is a critical issue and code hangs there.
Please let me know if you see the same behavior.

Thanks.

ravibhagavandas, 2020-11-02:
Hi @kumarfirmware 

I think at line 75 in the attached code after deleting the network connection you should also set `xNetworkConnected = false` so that it will retry in the next loop. This will also ensure that if physical network is reconnected it will give the semaphore correctly so it won't be a deadlock.

kumarfirmware, 2020-11-02:
Hi Ravi,

Thanks for the response. This fixes the issue. 

You can do this change from your side on the code also and publish in the master version, so that others will get benefited and they will not loose time again behind this issue.

I am closing this issue.

Thanks.

closed 2020-11-02 by kumarfirmware