aws/amazon-freertos #2516: IllegalInstruction error on semaphore wait

EtienneMdv, 2020-10-06:
Hi all,

I've recently been facing an error which seems to happen "randomly" on my ESP32 board. After some time publishing and receiving messages on topics the ESP32 catches the following error :
`Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40119494: 00f01d02 f01d120c 56004136
Backtrace: 0x4011949a:0x3ffe9500 0x000000c6:0x3ffe9520 0x40119203:0x3ffe9540`

The decoded backtrace is the following :
`0x4011949a: IotSemaphore_TimedWait at ../amazon-freertos/libraries/abstractions/platform/freertos/iot_threads_freertos.c line 344
0x40119203: _threadRoutineWrapper at ../amazon-freertos/libraries/abstractions/platform/freertos/iot_threads_freertos.c line 85
`

After looking at the ESP IDF docs, it looks like it may come from a task returning instead of being deleted. Before going too deep in my assumptions I thought I'd reach for better help.

Thanks in advance.
Best,
Etienne

htibosch, 2020-10-06:
> it looks like it may come from a task returning instead of being deleted

That could be a valid explanation. When a tasks ends, it is supposed to call `vTaskDelete( NULL )`, in order to get it removed from the lists of tasks.
The call to `vTaskDelete()` will not return.
In the past users have asked why this doesn't happen automatically.  There are various reasons, one reason is to save code.

Can you check if any of your tasks returns to the caller?

Maybe like this:

~~~c
void myTask( void * pvParameter )
{
	for( ;; )
	{
	}
	configASSERT( pdFALSE );
}
~~~

EtienneMdv, 2020-10-06:
Hi @htibosch 
Thanks for the answer.
I can't spot any task returning to the caller. I'm still trying to find the root of this issue, whether it's a firmware or hardware issue.
I've been facing some really strange behaviors lately that I don't really understand, especially related to SPIFFS functionalities.
The ESP IDF doc also mentions that the `IllegalInstruction` error could come from the configuration of the SPI flash pins.

> Failure to load next instruction from SPI flash. This usually happens if:
> 
> Application has reconfigured SPI flash pins as some other function (GPIO, UART, etc.). Consult Hardware Design Guidelines and the Datasheet for the chip or module for details about SPI flash pins.
> 
> Some external device was accidentally connected to SPI flash pins, and has interfered with communication between ESP32 and SPI flash.

I would be interested in knowing if anyone had similar issues.

To be continued...

dachalco, 2020-10-12:
Hi @EtienneMdv 

Have you made any new discoveries? Please let us know if there's some way we can reliably reproduce the issue -- Helpful for debugging.

It is interesting that the exception occurs on a call to `IotSemaphore_TimedWait`. If you can access a debugger, it may be useful to track down _which_ semaphore is being queried during at this crash, then seeing where it used. 

If you can attach a memory watchpoint to that semaphore, you can even break into code when it accesses the sem. This may reveal any invalid sequences (i.e. using sem before it's created, using sem after its been destroyed etc.).

yanjos-dev, 2020-10-22:
@EtienneMdv I'm going to close this for now. Please feel free to reopen this issue if you have any further issues or questions.

closed 2020-10-22 by yanjos-dev