aws/amazon-freertos #2526: AwsIotShadow_TimedUpdate returns TIMEOUT even though shadow updates fine

chinniwhites, 2020-10-07:
**Describe the bug**
I took the example code from here: https://github.com/aws/amazon-freertos/tree/master/demos and modified for 110 values to update to shadow instead of the one power on on my stm32L475E eval board. I did it slowly as to make sure everything came along well. I got to adding the last bit of info to the shadow reported and delta json when the demo started failing. The update sends and updates shadow correctly so I know the update actually works but I started noticing a TIMEOUT failure even though the update was successful. I narrowed it down and it seems like the AwsIotShadow_TimedUpdate is timing out. When looking at declaration of this function, it looks like both AwsIotShadow_Update and AwsIotShadow_Assert are returning TIMEOUT or value of 7. This looks it traces back to this:

/* Wait for a response to the Shadow operation. */
if( IotSemaphore_TimedWait( &( operation->notify.waitSemaphore ),
timeoutMs ) == true )
{
status = operation->status;
}
else
{
status = AWS_IOT_SHADOW_TIMEOUT;
}

/*
 * FreeRTOS V202007.00
 * Copyright (C) 2020 Amazon.com, Inc. or its affiliates.  All Rights Reserved.
 *
 * Permission is hereby granted, free of charge, to any person obtaining a copy of
 * this software and associated documentation files (the "Software"), to deal in
 * the Software without restriction, including without limitation the rights to
 * use, copy, modify, merge, publish, distribute, sublicense, and/or sell copies of
 * the Software, and to permit persons to whom the Software is furnished to do so,
 * subject to the following conditions:
 *
 * The above copyright notice and this permission notice shall be included in all
 * copies or substantial portions of the Software.
 *
 * THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
 * IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY, FITNESS
 * FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE AUTHORS OR
 * COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER LIABILITY, WHETHER
 * IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM, OUT OF OR IN
 * CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE SOFTWARE.
 *
 * http://aws.amazon.com/freertos
 * http://www.FreeRTOS.org
 */

/**
 * @file aws_iot_demo_shadow.c
 * @brief Demonstrates usage of the Thing Shadow library.
 *
 * This program demonstrates the using Shadow documents to toggle a state called
 * "powerOn" in a remote device.
 */

/* The config header is always included first. */
#include "iot_config.h"

/* Standard includes. */
#include <stdbool.h>
#include <stdio.h>
#include <stdlib.h>
#include <string.h>

/* Set up logging for this demo. */
#include "iot_demo_logging.h"

/* Platform layer includes. */
#include "platform/iot_clock.h"
#include "platform/iot_threads.h"

/* MQTT include. */
#include "iot_mqtt.h"

/* Shadow include. */
#include "aws_iot_shadow.h"

/* JSON utilities include. */
[... 1417 lines cut ...]
**Expected behavior**
No TIMEOUT

**Screenshots or console output**
If appropriate, please cut-and-paste the console output to explain the issue. If possible, include the call stack. 

**To reproduce**
Steps to reproduce the behavior:

1. ...
2. ...

**Code to reproduce the bug**
The code should be wrapped in the cpp tag in order to be displayed clearly. For example:

```cpp
printf("Hello World")
```

If the code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

**Additional context**
Add any other context about the problem here.

Thank you!

dachalco, 2020-10-08:
Hello @chinniwhites 

Formatted original post to surround code and terminal output in code blocks.
I'm looking into this now and will follow up soon, providing my analysis or fetching input from one of our devs.

dachalco, 2020-10-08:
Hello @chinniwhites 

Regarding this snippet:
```c
        /* Check for errors from snprintf. The expected value is the length of
         * the desired JSON document less the format specifier for the state. */
        //if( ( size_t ) status != EXPECTED_DESIRED_JSON_SIZE )
    	//if( ( unsigned int ) status != EXPECTED_REPORTED_JSON_SIZE )
      //  {
       //     IotLogError( "Failed to generate desired state document for Shadow update"
       //                  " %d of %d.", i, AWS_IOT_DEMO_SHADOW_UPDATE_COUNT );

            //WB ADDED
            IotLogError( "Status: %d, Expected: %d", status, EXPECTED_REPORTED_JSON_SIZE);

       //     status = EXIT_FAILURE;
       //     break;
       // }
       // else
       // {
            status = EXIT_SUCCESS;
       // }
```
The results of this `snprintf` can be safely assumed in subsequent code, given the original source sets `status = EXIT_FAILURE` and exits early upon failure. In this case, execution proceeds regardless of `snprintf` failure, which may violate assumptions made by subsequent code. Moreover, the formatted buffer is JSON. Truncation might lead to JSON syntax errors which is non-ideal.

That being said, a good starting point is to investigate this failure of `snprintf`.
1. Can you share the result of `strlen(SHADOW_REPORTED_JSON)` and `sizeof(SHADOW_REPORTED_JSON)`?
    - `strlen` to see length at runtime which depends on string literal storage. Curious if this is getting truncated.
    - `sizeof` is likely being evaluated at compile-time, given string literal argument.

chinniwhites, 2020-10-08:
//WB ADDED
            IotLogError( "Status: %d, Expected: %d, Strlen reported: %d, Sizeof reported: %d",
            		status, EXPECTED_REPORTED_JSON_SIZE, strlen(SHADOW_REPORTED_JSON), sizeof(SHADOW_REPORTED_JSON));

29 10222 [iot_thread] [ERROR][DEMO][10221] Status: 1578, Expected: 1908, Strlen reported: 1910, Sizeof reported: 1911

dachalco, 2020-10-13:
@chinniwhites

Thanks again for pointing this out. I believe I found answer to the first issue.
This particular block of code, especially its comments, is misleading and wrong. 
```
/**
 * @brief The expected size of #SHADOW_DESIRED_JSON.
 *
 * Because all the format specifiers in #SHADOW_DESIRED_JSON include a length,
 * its full size is known at compile-time.
 */
#define EXPECTED_DESIRED_JSON_SIZE    ( sizeof( SHADOW_DESIRED_JSON ) - 3 )
```

the `sizeof()` does not operate in this fashion, and instead returns size of format string irrespective of format specifiers (includes them 'as is' in size), as it should.

- I'm preparing a PR to rid prevent this confusion in the future.
----
A correct way to get the size of the `snprintf` is to do a dry run:
```c
// Write 0 bytes into NULL. Returns how many bytes WOULD have been written.
size_t ulActualFormattedSize = snprintf(NULL, 0, YOUR_FORMAT_STRING, arg1, arg2, arg3); // , argN) etc.
```
You can then `malloc` this size, or if you have an upper bound you statically allocate too.

To reflect that actual payload size, can you replace:
```c
    updateDocument.u.update.updateDocumentLength = EXPECTED_REPORTED_JSON_SIZE;
```
with
```
    // Where ulActualActualFormattedSize = snprintf( .. )
    updateDocument.u.update.updateDocumentLength = ulActualFormattedSize;
```

chinniwhites, 2020-10-13:
Did what you said minus the malloc part.
Looks like the value is the same with old or new size approach and still timesout but still published to shadow....

//WB ADDED
    size_t ulActualFormattedSize = snprintf(NULL, 0, SHADOW_REPORTED_JSON,
    			( unsigned int ) UART4_reportBuffer[0],
				( unsigned int ) UART4_rxBuffer[1],
			   ( unsigned int ) UART4_rxBuffer[2],
			   ( unsigned int ) UART4_rxBuffer[3],
			   ( unsigned int ) UART4_rxBuffer[4],
			   ( unsigned int ) UART4_rxBuffer[5],
			   ( unsigned int ) UART4_rxBuffer[6],
			   ( unsigned int ) UART4_rxBuffer[7],
			   ( unsigned int ) UART4_rxBuffer[8],
			   ( unsigned int ) UART4_rxBuffer[9],
			   ( unsigned int ) UART4_rxBuffer[10],
			   ( unsigned int ) UART4_rxBuffer[11],
			   ( unsigned int ) UART4_rxBuffer[12],
			   ( unsigned int ) UART4_rxBuffer[13],
			   ( unsigned int ) UART4_rxBuffer[14],
			   ( unsigned int ) UART4_rxBuffer[15],
			   ( unsigned int ) UART4_rxBuffer[16],
			   ( unsigned int ) UART4_rxBuffer[17],
			   ( unsigned int ) UART4_rxBuffer[18],
			   ( unsigned int ) UART4_rxBuffer[19],
			   ( unsigned int ) UART4_rxBuffer[20],
			   ( unsigned int ) UART4_rxBuffer[21],
			   ( unsigned int ) UART4_rxBuffer[22],
			   ( unsigned int ) UART4_rxBuffer[23],
			   ( unsigned int ) UART4_rxBuffer[24],
			   ( unsigned int ) UART4_rxBuffer[25],
			   ( unsigned int ) UART4_rxBuffer[26],
			   ( unsigned int ) UART4_rxBuffer[27],
			   ( unsigned int ) UART4_rxBuffer[28],
			   ( unsigned int ) UART4_rxBuffer[29],
			   ( unsigned int ) UART4_rxBuffer[30],
			   ( unsigned int ) UART4_rxBuffer[31],
			   ( unsigned int ) UART4_rxBuffer[32],
			   ( unsigned int ) UART4_rxBuffer[33],
			   ( unsigned int ) UART4_rxBuffer[34],
			   ( unsigned int ) UART4_rxBuffer[35],
			   ( unsigned int ) UART4_rxBuffer[36],
			   ( unsigned int ) UART4_rxBuffer[37],
			   ( unsigned int ) UART4_rxBuffer[38],
			   ( unsigned int ) UART4_rxBuffer[39],
			   ( unsigned int ) UART4_rxBuffer[40],
			   ( unsigned int ) UART4_rxBuffer[41],
			   ( unsigned int ) UART4_rxBuffer[42],
			   ( unsigned int ) UART4_rxBuffer[43],
			   ( unsigned int ) UART4_rxBuffer[44],
			   ( unsigned int ) UART4_rxBuffer[45],
			   ( unsigned int ) UART4_rxBuffer[46],
			   ( unsigned int ) UART4_rxBuffer[47],
			   ( unsigned int ) UART4_rxBuffer[48],
			   ( unsigned int ) UART4_rxBuffer[49],
			   ( unsigned int ) UART4_rxBuffer[50],
			   ( unsigned int ) UART4_rxBuffer[51],
			   ( unsigned int ) UART4_rxBuffer[52],
			   ( unsigned int ) UART4_rxBuffer[53],
			   ( unsigned int ) UART4_rxBuffer[54],
			   ( unsigned int ) UART4_rxBuffer[55],
			   ( unsigned int ) UART4_rxBuffer[56],
			   ( unsigned int ) UART4_rxBuffer[57],
			   ( unsigned int ) UART4_rxBuffer[58],
			   ( unsigned int ) UART4_rxBuffer[59],
			   ( unsigned int ) UART4_rxBuffer[60],
			   ( unsigned int ) UART4_rxBuffer[61],
			   ( unsigned int ) UART4_rxBuffer[62],
			   ( unsigned int ) UART4_rxBuffer[63],
			   ( unsigned int ) UART4_rxBuffer[64],
			   ( unsigned int ) UART4_rxBuffer[65],
			   ( unsigned int ) UART4_rxBuffer[66],
			   ( unsigned int ) UART4_rxBuffer[67],
			   ( unsigned int ) UART4_rxBuffer[68],
			   ( unsigned int ) UART4_rxBuffer[69],
			   ( unsigned int ) UART4_rxBuffer[70],
			   ( unsigned int ) UART4_rxBuffer[71],
			   ( unsigned int ) UART4_rxBuffer[72],
			   ( unsigned int ) UART4_rxBuffer[73],
			   ( unsigned int ) UART4_rxBuffer[74],
			   ( unsigned int ) UART4_rxBuffer[75],
			   ( unsigned int ) UART4_rxBuffer[76],
			   ( unsigned int ) UART4_rxBuffer[77],
			   ( unsigned int ) UART4_rxBuffer[78],
			   ( unsigned int ) UART4_rxBuffer[79],
			   ( unsigned int ) UART4_rxBuffer[80],
			   ( unsigned int ) UART4_rxBuffer[81],
			   ( unsigned int ) UART4_rxBuffer[82],
			   ( unsigned int ) UART4_rxBuffer[83],
			   ( unsigned int ) UART4_rxBuffer[84],
			   ( unsigned int ) UART4_rxBuffer[85],
			   ( unsigned int ) UART4_rxBuffer[86],
			   ( unsigned int ) UART4_rxBuffer[87],
			   ( unsigned int ) UART4_rxBuffer[88],
			   ( unsigned int ) UART4_rxBuffer[89],
			   ( unsigned int ) UART4_rxBuffer[90],
			   ( unsigned int ) UART4_rxBuffer[91],
			   ( unsigned int ) UART4_rxBuffer[92],
			   ( unsigned int ) UART4_rxBuffer[93],
			   ( unsigned int ) UART4_rxBuffer[94],
			   ( unsigned int ) UART4_rxBuffer[95],
			   ( unsigned int ) UART4_rxBuffer[96],
			   ( unsigned int ) UART4_rxBuffer[97],
			   ( unsigned int ) UART4_rxBuffer[98],
			   ( unsigned int ) UART4_rxBuffer[99],
			   ( unsigned int ) UART4_rxBuffer[100],
			   ( unsigned int ) UART4_rxBuffer[101],
			   ( unsigned int ) UART4_rxBuffer[102],
			   ( unsigned int ) UART4_rxBuffer[103],
			   ( unsigned int ) UART4_rxBuffer[104],
			   ( unsigned int ) UART4_rxBuffer[105],
			   ( unsigned int ) UART4_rxBuffer[106],
			   ( unsigned int ) UART4_rxBuffer[107],
			   ( unsigned int ) UART4_rxBuffer[108],
			   ( unsigned int ) UART4_rxBuffer[109],
			   ( unsigned int ) UART4_rxBuffer[110],
			   ( long unsigned ) ( IotClock_GetTimeMs() % 1000000 ) );

updateDocument.u.update.updateDocumentLength = ulActualFormattedSize;

//WB ADDED
            IotLogError( "Status: %d, Expected: %d, Strlen reported: %d, Sizeof reported: %d, actualformatted: %d",
            status, EXPECTED_REPORTED_JSON_SIZE, strlen(SHADOW_REPORTED_JSON), sizeof(SHADOW_REPORTED_JSON), ulActualFormattedSize );

29 9969 [iot_thread] [ERROR][DEMO][9969] Status: 1578, Expected: 1908, Strlen reported: 1910, Sizeof reported: 1911, actualformatted: 1578

chinniwhites, 2020-10-19:
Hello, any update here?
Not so much concerned about snprintf checking as to why i get the timeout even though publish to shadow works...

dachalco, 2020-10-19:
Hi @chinniwhites 

At this rate, it's best that I try to locally reproduce the issue. Thought the payload data length may have had an effect.
Will try this out later today and get back to you.

dachalco, 2020-10-21:
Checking back in, I've been able to reproduce the issue and am debugging. It's a strange one. So far it appears to be an issue with the mqtt or tcp transport layer. Will continue to update here with findings.

chinniwhites, 2020-10-29:
Thanks David.Any updates?

dachalco, 2020-11-09:
Hello @chinniwhites 

After running your code on all of our supported boards, I noticed this TIMEOUT does not occur for some boards, reinforcing my suspicions that it's a problem specific to our `stm32L4` port. My remote debugging capabilities have been limited to remote execution, with no step-mode debugging. The boards I do have handy (NXP) don't exhibit this issue.

I ordered the `stm32L475E` which should be arriving in the next 1-2 days and will check back in then, after I get an opportunity to step-mode debug with the board.

dan4thewin, 2021-03-05:
@chinniwhites , I notice this issue is still open and apologize for the lack of progress.  We've had library changes and a release since you first encountered the issue.  Are you still having trouble, and if so, have you tried using the latest release?

chinniwhites, 2021-03-05:
Thanks for follow up!I have not tried the latest release.Can you refresh my memory where to get new library/project from?I believe i will have to get this new library with github and copy over my code again.Its been awhile since i went through process.
Thanks!

Sent from AT&T Yahoo Mail for iPhone

dan4thewin, 2021-03-06:
The library is part the C SDK ( https://github.com/aws/aws-iot-device-sdk-embedded-C ) and is also included in the reference integrations ( https://github.com/aws/amazon-freertos ).  These both include the library as a submodule from https://github.com/aws/Jobs-for-AWS-IoT-embedded-sdk .

yanjos-dev, 2021-03-15:
I'm closing this issue due to inactivity. Please reopen this if you would like any further assistance.

closed 2021-03-15 by yanjos-dev