aws/amazon-freertos #2539: SOCKET_Recv and SOCKET_Send hardware timeouts lead to "assert"

psavyhin, 2020-10-09:
We found one sensitive issue which could lead to many hours of struggling with connectivity problems in the FreeRTOS.

1)	Originally we found it in version 201908.00 but  there are same issues in current version too (steps to reproduce are shown at the end)
2)	MQTT Agent starts_networkReceiveTask in function IotMqtt_Connect by calling setReceiveCallback
3)	_networkReceiveTask calls SOCKET_Recv to receive network data, and finally prvNetworkRecv function (part of vendor secure sockets port) is called.
4)	As prvNetworkRecv function is part of vendor secure sockets port, so it uses vendors synchronization object to synchronize data flow.
5)	Similarly prvNetworkSend is called thru Thread Pool mechanism
6)	At some moment user see that WiFi connection is lost and wants to reconnect the system via calling: 
a.	AwsIotNetworkManager_DisableNetwork which calls _wifiDisable and all synchronization objects are destroyed.
b.	Here is the problem. 
c.	AwsIotNetworkManager_EnableNetwork
d.	System has no connectivity.

Problem description:
prvNetworkRecv and prvNetworkSend know nothing about network disabling. If prvNetworkRecv is still working while network is shut downed (receive procedure due to some reasons require more time than socket timeout value) then it tries to use destroyed synchronization object and networkReceiveTask stalls at assert.

There are some pictures in attach to simplify understanding of the problem.

Also I added the patch to reproduce the problem for next STM32 IoT Node FreeRTOS source version 202007.00: https://console.aws.amazon.com/iot/home?region=us-east-1#/freertos/configuration/Connect_to_AWS_IoT_-_STM32-B-L475E-IOT01A?type=predefined

1)	Apply patch by “patch -p 1 -i ../ba.diff”
2)	Open project and add credentials
3)	Compile and Start Debug
4)	Start the application and wait until demo finishes
5)	Press Pause at the debug toolbar and check the problem with assert that I described in previous mail.

[PatchAndPics.zip](https://github.com/aws/amazon-freertos/files/5353859/PatchAndPics.zip)

dachalco, 2020-10-13:
Hi @psavyhin 

Thanks for bringing this to our attention, and many more thanks for being so thorough with details! 
Going to assign this to the appropriate team so they will investigate.

leegeth, 2020-12-24:
Hi @psavyhin,

Thanks for your patience and apologize for the delay in getting back to you. Thanks for the details and the patch for showing the issue to us. You would be able to solve the issue by calling [IotMqtt_Disconnect](https://docs.aws.amazon.com/freertos/latest/lib-ref/embedded-csdk/v4.0_beta_deprecated/lib-ref/c-sdk/mqtt/mqtt_function_disconnect.html) API with the flag [IOT_MQTT_FLAG_CLEANUP_ONLY](https://docs.aws.amazon.com/freertos/latest/lib-ref/embedded-csdk/v4.0_beta_deprecated/lib-ref/c-sdk/mqtt/iot__mqtt__types_8h.html#a821c60fc92e849e3d6c4258360047f93) to clean up the networkReceiveTask before calling AwsIotNetworkManager_DisableNetwork to turn off Wifi. Can you please let us know if this solution helps with your problem?

We have a new [MQTT library](https://github.com/FreeRTOS/coreMQTT/tree/v1.1.0) integrated in this repository. This library has new [demos](https://github.com/aws/amazon-freertos/tree/202012.00/demos/coreMQTT), which doesn't have any dependency on the networkReceiveTask. The LTS(Long Term Support) version of the library is integrated in the release [202012.00](https://github.com/aws/amazon-freertos/releases/tag/202012.00). Please check it out and see if it will be a better fit for your application.

psavyhin, 2020-12-28:
Dear Leegeth,

Thanks a lot for your detailed comment, we will chek it. Hoping that this
is completely eliminated in new version of FreeRTOS.

—

Regards,
Pavel Savyhin
Head of Embedded Systems Engineering, Klika Tech, Inc.
Mobile/WhatsApp: +375-29-7-509-661
Email: psavyhin@klika-tech.com

www.klika-tech.com

NOTICE: This email may contain confidential and privileged material for the
sole use of the intended recipient(s). Any review, use, distribution or
disclosure by others is strictly prohibited. If you are not the intended
recipient (or authorized to receive for the recipient), please contact the
sender by reply email and delete all copies of this message.

*From:* leegeth <notifications@github.com>
*Sent:* Thursday, December 24, 2020 6:24 AM
*To:* aws/amazon-freertos <amazon-freertos@noreply.github.com>
*Cc:* Pavel Savyhin <psavyhin@klika-tech.com>; Mention <
mention@noreply.github.com>
*Subject:* Re: [aws/amazon-freertos] SOCKET_Recv and SOCKET_Send hardware
timeouts lead to "assert" (#2539)

Hi @psavyhin <https://github.com/psavyhin>,

Thanks for your patience and apologize for the delay in getting back to
you. Thanks for the details and the patch for showing the issue to us. You
would be able to solve the issue by calling IotMqtt_Disconnect
<https://docs.aws.amazon.com/freertos/latest/lib-ref/embedded-csdk/v4.0_beta_deprecated/lib-ref/c-sdk/mqtt/mqtt_function_disconnect.html>
API with the flag IOT_MQTT_FLAG_CLEANUP_ONLY
<https://docs.aws.amazon.com/freertos/latest/lib-ref/embedded-csdk/v4.0_beta_deprecated/lib-ref/c-sdk/mqtt/iot__mqtt__types_8h.html#a821c60fc92e849e3d6c4258360047f93>
to clean up the networkReceiveTask before calling
AwsIotNetworkManager_DisableNetwork to turn off Wifi. Can you please let us
know if this solution helps with your problem?

We have a new MQTT library
<https://github.com/FreeRTOS/coreMQTT/tree/v1.1.0> integrated in this
repository. This library has new demos
<https://github.com/aws/amazon-freertos/tree/202012.00/demos/coreMQTT>,
which doesn't have any dependency on the networkReceiveTask. The LTS(Long
Term Support) version of the library is integrated in the release 202012.00
<https://github.com/aws/amazon-freertos/releases/tag/202012.00>. Please
check it out and see if it will be a better fit for your application.

leegeth, 2021-01-04:
@psavyhin,

Thanks. Yea the issue that you brought up is not present in the new MQTT library. Please let us know if you need any further help.

leegeth, 2021-01-20:
@psavyhin,

Please let us know if you need any further information from us. If you are good with the answer to your questions/issue, we will go ahead and close this issue.

psavyhin, 2021-01-25:
@leegeth Yes, thanks, I'm closing it.

closed 2021-01-25 by psavyhin