aws/amazon-freertos #2552: POSIX time.h not including expected structures.

osloapps-max, 2020-10-12:
**Describe the bug**
A clear and concise description of what the bug is.

**System information**
- Which hardware board or part numbers?
- IDE used
- Operating System [Windows|Linux|MacOS]
- Version of FreeRTOS (run ``git describe --tags`` to find it)
- Project [Custom Application|MQTT|Shadow|Greengrass|OTA|Device Defender|TCP Echo|Sub pub Demo|Simple TCP Server]
- If your project is a Custom Application, please add the relevant code snippet in the section `Code to reproduce the bug`. 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots or console output**
If appropriate, please cut-and-paste the console output to explain the issue. If possible, include the call stack. 

**To reproduce**
Steps to reproduce the behavior:

1. ...
2. ...

**Code to reproduce the bug**
The code should be wrapped in the cpp tag in order to be displayed clearly. For example:

```cpp
printf("Hello World")
```

If the code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

**Additional context**
Add any other context about the problem here.

Thank you!

dachalco, 2020-10-13:
Hi @osloapps-max 

Thanks for bringing this to our attention. I've queued this to the appropriate internal team and they should provide a response once this issue is addressed.

dachalco, 2020-12-02:
Hi @osloapps-max 

Recently merged PR #2797 which adds `struct tm` to our `time.h`. Please re-open the issue as needed.

closed 2020-12-02 by dachalco