aws/amazon-freertos #2566: common-io: GPIO interrupt level can cause device hang-up in case of test failure

osloapps-max, 2020-10-14:
**Describe the bug**
If the GPIO interrupt level test cases (low and/or high level) fail due to an error in the "trigger GPIO", the test cases could cause an ISR dead-loop depending on the device in question. 

The reason is due to the test assuming we can "disable" the trigger by setting the trigger pin to the opposite value. If this is what fails in the first place then the interrupt could keep firing as the level never change. 

**System information**
- Which hardware board or part numbers?
   Texas Instruments CC3220SF
- IDE used
  None
- Operating System [Windows|Linux|MacOS]
 Windows
- Version of FreeRTOS (run ``git describe --tags`` to find it)
- Project [Custom Application|MQTT|Shadow|Greengrass|OTA|Device Defender|TCP Echo|Sub pub Demo|Simple TCP Server]
  aws_tests
- If your project is a Custom Application, please add the relevant code snippet in the section `Code to reproduce the bug`. 

**Expected behavior**
The test fails nicely allowing for the rest of the test suit to continue.

I realize it might not be a single good way to do this but disabling the interrupt from the ISR instead of changing trigger level (inside prvGpioInterruptCallback() @ 564) is a possible workaround. 

In case the iot_gpio_ioctl() API is considered "non ISR safe" then we most likely need to specify a ISR safe API to allow the user to disable interrupts as this particular "failure" is similar to that of having an external sensor where you can't control the trigger level.

Thank you!

pvyawaha, 2020-10-15:
Hello, 

Thank you for reporting this , we are looking into this issue.

lundinc2, 2020-11-05:
Sorry for the delay, PR is now out for review https://github.com/FreeRTOS/corePKCS11/pull/26

yourslab, 2021-02-06:
The above PR has been merged so I will close this issue now.

closed 2021-02-06 by yourslab