aws/amazon-freertos #2569: Issue running multiple demos

nickfaughey, 2020-10-14:
What's the best approach for running two (or more) of the provided demos simultaneously? I've tried simply repeating `iot_demo_runner.c`'s `Iot_CreateDetachedThread()` with another demo's entry function (OTA and MQTT) like so:

```c
void DEMO_RUNNER_RunDemos( void )
{
    /* These demos are shared with the C SDK and perform their own initialization and cleanup. */

    static demoContext_t demoContext =
    {
        .networkTypes                = democonfigNETWORK_TYPES,
        .demoFunction                = DEMO_entryFUNCTION,
        .networkConnectedCallback    = DEMO_networkConnectedCallback,
        .networkDisconnectedCallback = DEMO_networkDisconnectedCallback
    };

    Iot_CreateDetachedThread( runDemoTask,
                              &demoContext,
                              democonfigDEMO_PRIORITY,
                              democonfigDEMO_STACKSIZE );

    static demoContext_t demoContext2 =
    {
        .networkTypes                = democonfigNETWORK_TYPES,
        .demoFunction                = DEMO_entryFUNCTION_2,
        .networkConnectedCallback    = DEMO_networkConnectedCallback,
        .networkDisconnectedCallback = DEMO_networkDisconnectedCallback
    };

    Iot_CreateDetachedThread( runDemoTask,
                              &demoContext2,
                              democonfigDEMO_PRIORITY,
                              democonfigDEMO_STACKSIZE );

}
```

Where `DEMO_entryFUNCTION` is defined as the OTA demo's `vStartOTAUpdateDemoTask` and `DEMO_entryFUNCTION_2` is defined as the MQTT demo's `RunMqttDemo`. They play nicely together until MQTT publishes, when it gets stuck in a loop waiting for a semaphore (likely in use by the OTA demo) to free up:

```
499 42047 [iot_thread] [DEBUG][THREAD][42047] Locking mutex 0x20074448.
500 42054 [iot_thread] [DEBUG][THREAD][42054] Unlocking mutex 0x20074448.
501 75985 [iot_thread] [WARN ][THREAD][75985] Timeout waiting on semaphore 0x2007436c.
502 75993 [iot_thread] [DEBUG][THREAD][75993] Locking mutex 0x20074448.
503 76000 [iot_thread] [DEBUG][THREAD][76000] Unlocking mutex 0x20074448.
504 76486 [iot_thread] [WARN ][THREAD][76486] Timeout waiting on semaphore 0x2007436c.
505 76494 [iot_thread] [DEBUG][THREAD][76494] Locking mutex 0x20074448.
506 76501 [iot_thread] [DEBUG][THREAD][76501] Unlocking mutex 0x20074448.
507 101461 [iot_thread] [WARN ][THREAD][101461] Timeout waiting on semaphore 0x2007436c.
```

The advice in `iot_demo_runner.h` says to do it differently:

```c
/*  By design only one demo can be selected at a time. If
 *  you choose more than one only the first choice will be run. If you require
 *  more than one demo to run at the same time you can create a new combined
 *  demo and run both the required demo tasks from your demo entry function as
 *  2 separate tasks
 *
 *  The define to choose the demo to run is in aws_demo_config.h included above.
 */
```

But it isn't quite clear how I'd go about doing this. Does this mean I should use a single demo entry function and spawn new freertos tasks with `xTaskCreate()` in that function for each demo? If so, how will I pass them the internal IoT parameters normally given directly to the entry function?

```c
int RunMqttDemo( bool awsIotMqttMode,
                 const char * pIdentifier,
                 void * pNetworkServerInfo,
                 void * pNetworkCredentialInfo,
                 const IotNetworkInterface_t * pNetworkInterface )
```

pvyawaha, 2020-10-14:
Hello, 

The OTA demo creates MQTT connection and this connection handle is passed when initializing the OTA Agent. The demo then starts OTA Agent task which handles all the OTA operations using this connection. This connection can be shared for other MQTT operations required in the application. 

For example we can modify and use OTA demo for performing MQTT operations using the same connection. Please refer to the example I have created here https://github.com/aws/amazon-freertos/blob/mchp/otaapp/demos/ota/aws_iot_ota_update_demo.c

In this example the OTA Agent is initialized and the demo publishes to an example topic `otamqttdemo/topic1` the message `Hello OTA + MQTT`. 

Please let me know if this helps or you have more questions.

nickfaughey, 2020-10-16:
Thanks, this is a decent workaround for now. Since in general the OTA functionality is long-running, I'll just use its own MQTT connection for other publish operations.

closed 2020-10-16 by nickfaughey

Yoimer, 2021-02-26:
@pvyawaha, @nickfaughey Hello. Is there a way to implement this in the latest AWS FreeRTOS ESP32 version?