aws/amazon-freertos #2578: common-io: GPIO interrupt test should set callback before enabling interrupt

osloapps-max, 2020-10-15:
**Describe the bug**

The common-io GPIO interrupt test currently enable the GPIO interrupt before providing a callback. This could leave the driver in a state where a continuous "stream" of interrupts fire and are partly processed by the device specific layer in case the trigger remains true. 

One example of this is lthe AFQP_IotGpioInterrupt test case where a level based triggers could freeze the device in a "ISR loop" before the callback that should terminate it is provided.

**System information**

Which hardware board or part numbers?
Texas Instruments CC3220SF
IDE used
None
Operating System [Windows|Linux|MacOS]
Windows
Version of FreeRTOS (run git describe --tags to find it)
Project [Custom Application|MQTT|Shadow|Greengrass|OTA|Device Defender|TCP Echo|Sub pub Demo|Simple TCP Server]
aws_tests
If your project is a Custom Application, please add the relevant code snippet in the section Code to reproduce the bug.

**Expected behavior**

The test should setup callbacks prior to enabling the interrupt. The AFQP_IotGpioInterrupt test case is one example of a test case that currently enable the interrupt and then sets up the callback.

Thank you!

lundinc2, 2021-02-22:
Apologies for the delay, work on this will start shortly

lundinc2, 2021-03-11:
Merged update. Thanks for bringing this to our attention!

closed 2021-03-11 by lundinc2